Freetz-NG/freetz-ng #3: 3490-devel-15222M startup kernel warning

WileC, 2019-02-25:
```
Feb 25 20:33:54 fritz kern.warn kernel: [  183.060000][0]------------[ cut here ]------------
Feb 25 20:33:54 fritz kern.warn kernel: [  183.060000][0]WARNING: at /home/jpluschke/FBox/GIT-GU/GU_IQ17p2_NL1_vr9/RELEASE_NETWORK_SOURCE_vr9_build/ar7/dsld/kernel/kdsld.c:1915 0x815300a0 kdsld_net_xmit+0x480/0x560 [kdsldmod]()
Feb 25 20:33:54 fritz kern.debug kernel: [  183.060000][0]Modules linked in: ifx_ppa_mini_qos(P) ifx_ppa_mini_sessions ifxmips_ppa_hal_vr9_e5 krtp(PO) ltq_eth_oam_handler sch_fq_codel sch_llq sch_tbf userman_mod(PO) kdsldmod(PO) cprocfsmod(PO) capi_codec(P) isdn_fbox_f
Feb 25 20:33:54 fritz kern.debug kernel: [  183.060000][0]CPU: 0 PID: 0 Comm: swapper/0 Tainted: P           O 3.10.107 #1
Feb 25 20:33:54 fritz kern.debug kernel: [  183.060000][0]Stack : 00000016 8004b364 00000000 00000000 00000000 00000000 8090c0fa 00000040
Feb 25 20:33:54 fritz kern.debug kernel: [  183.060000][0]	  00000000 80910000 80723ad0 00000000 00000000 00000000 807e8bc0 807e8907
Feb 25 20:33:54 fritz kern.debug kernel: [  183.060000][0]	  80723ad0 00000000 00000000 8090b890 00000000 8ed3c860 8cd85238 8004c778
Feb 25 20:33:54 fritz kern.debug kernel: [  183.060000][0]	  80920000 800485f0 00000000 00000000 807276f4 807d774c 807d774c 807e8bc0
Feb 25 20:33:54 fritz kern.debug kernel: [  183.060000][0]	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 807d76c0
Feb 25 20:33:54 fritz kern.debug kernel: [  183.060000][0]	  ...
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]Classified pointer on stack:
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]8090c0fa textbuf.29243+0x2/0x3e0
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]80723ad0 __func__.4157+0xa5ce0/0x132950
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]807e8bc0 init_task+0x1d0/0x490
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]807e8907 init_uts_ns+0xc7/0x1a0
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]8090b890 buf.15869+0x0/0x18
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]8ed3c860 [slab: type:ip6_dst_cache size:224 start:0x8ed3c860+0x0 allocated by:0x804f85e4 dst_alloc+0x64/0x1a0]
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]800485f0 print_tainted+0x30/0x100
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]807276f4 __func__.4157+0xa9904/0x132950
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]807d774c init_thread_union+0x374c/0x4000
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]807d774c init_thread_union+0x374c/0x4000
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]807d76c0 init_thread_union+0x36c0/0x4000
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]80048878 warn_slowpath_common+0x78/0xc0
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]Call Trace:
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]36c0: [<80022900>] 0x80022900 show_stack+0x80/0xa0
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]37a8: [<80048878>] 0x80048878 warn_slowpath_common+0x78/0xc0
Feb 25 20:33:54 fritz kern.err kernel: [  183.060000][0]37d0: [<80048958>] 0x80048958 warn_slowpath_null+0x18/0x40
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]37f0: [<815300a0>] 0x815300a0 kdsld_net_xmit+0x480/0x560 [kdsldmod]
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3828: [<804f13fc>] 0x804f13fc dev_hard_start_xmit+0x35c/0x620
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3870: [<80527270>] 0x80527270 sch_direct_xmit+0x170/0x300
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]38a0: [<804f1978>] 0x804f1978 dev_queue_xmit+0x2b8/0x700
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]38d8: [<805af50c>] 0x805af50c ip6_finish_output2+0x22c/0x6c0
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3918: [<805cb594>] 0x805cb594 ndisc_send_skb+0x414/0x500
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]39a8: [<805cdf9c>] 0x805cdf9c ndisc_send_redirect+0x31c/0x440
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3a78: [<805b29c8>] 0x805b29c8 ip6_forward+0xbc8/0xd60
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3ae0: [<804ede34>] 0x804ede34 __netif_receive_skb_core+0x274/0xa80
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3b38: [<804eeb00>] 0x804eeb00 netif_receive_skb+0x40/0xe0
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3b68: [<805fbba8>] 0x805fbba8 br_handle_frame+0x288/0x380
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3ba0: [<804edd6c>] 0x804edd6c __netif_receive_skb_core+0x1ac/0xa80
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3bf8: [<804eeb00>] 0x804eeb00 netif_receive_skb+0x40/0xe0
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3c28: [<81320e10>] 0x81320e10 process_backlog_e5+0xb70/0x1040 [ifxmips_ppa_datapath_vr9_e5]
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3ca0: [<804efc30>] 0x804efc30 net_rx_action+0x190/0x260
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3ce0: [<800539e4>] 0x800539e4 __do_softirq+0x184/0x300
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3d40: [<80053c68>] 0x80053c68 do_softirq.part.1+0x68/0xa0
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3d58: [<80054034>] 0x80054034 irq_exit+0xb4/0x100
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3d70: [<80014fd0>] 0x80014fd0 hw4_irqdispatch+0xd0/0x220
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3da0: [<8000248c>] 0x8000248c ret_from_irq+0x0/0x4
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3e70: [<8001edd4>] 0x8001edd4 __pastwait+0x0/0xc
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3e70: [<8001edcc>] 0x8001edcc r4k_wait_irqoff+0x2c/0x34
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3e88: [<80094060>] 0x80094060 cpu_startup_entry+0x180/0x1e0
Feb 25 20:33:54 fritz kern.err kernel: [  183.070000][0]3eb8: [<808b89b8>] 0x808b89b8 start_kernel+0x3d8/0x3f4
Feb 25 20:33:54 fritz kern.debug kernel: [  183.070000][0]
Feb 25 20:33:54 fritz kern.warn kernel: [  183.070000][0]---[ end trace e89d9aa87e55137c ]---
```

fda77, 2019-02-26:
Das sagt mir nichts. Kann alles sein. Ich rate mal drauf los:
Hast du replace kernel aktiviert? Irgend Kernelmodule nachinstalliert? Evtl einen Removepatch der ein Kernelmodul löscht?

WileC, 2019-02-27:
nichts von all dem genannten. Anbei mal meine .config
[config.txt](https://github.com/freetz-ng/freetz-ng/files/2911988/config.txt)

fda77, 2019-03-01:
> Und ist das Problem neu oder schon mit Freetz wie es im Dezember 2018 war?

> nichts von all dem genannten

Dann gäbe es das Problem gar nicht :) 
Es ist nämlich die Frage wo zu suchen ist, ob ich da was kaputt gemacht habe oder es noch nie funktionieret

WileC, 2019-03-01:
Also ich kanns dir leider nur so posten, wie es sich mit dem aktuellen Build verhält :(

WileC, 2019-03-01:
So sieht der StartUp mit einem **Freetz 15027** aus:

```
Jan  1 01:00:54 fritz syslog.info syslogd started: BusyBox v1.27.2
Jan  1 01:00:54 fritz kern.notice kernel: klogd started: BusyBox v1.27.2
Jan  1 01:00:54 fritz kern.notice kernel: [    0.000000] Linux version 3.10.107 (jpluschke@GU_IQ17p2_NL1_vr9) (gcc version 5.4.0 (Buildroot 2016.05-g4d7c863) ) #1 SMP Wed Aug 29 19:46:45 CEST 2018
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] [env_init] 0x81019370[0]
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] [env_init] 0x810193b0[1]
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] [env_init] 0x810194b0[2]
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] [env_init] switch to ram location
Jan  1 01:00:54 fritz kern.err kernel: [    0.000000] [init_avm_kernel_config] AVM Kernel Config (ptr 808a8000)
Jan  1 01:00:54 fritz kern.err kernel: [    0.000000] [init_avm_kernel_config] AVM Kernel Config: module memory entry
Jan  1 01:00:54 fritz kern.err kernel: [    0.000000] [init_avm_kernel_config] AVM Kernel Config: version info entry
Jan  1 01:00:54 fritz kern.err kernel: [    0.000000] [init_avm_kernel_config] AVM Kernel Config: device-tree for subrev 0 found
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] bootconsole [early0] enabled
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] CPU0 revision is: 00019556 (MIPS 34Kc)
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] Determined physical RAM map:
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000]  memory: 10000000 @ 00000000 (usable)
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Jan  1 01:00:54 fritz kern.debug kernel: [    0.000000] Zone ranges:
Jan  1 01:00:54 fritz kern.debug kernel: [    0.000000]   Normal   [mem 0x00000000-0x0fffffff]
Jan  1 01:00:54 fritz kern.debug kernel: [    0.000000] Movable zone start for each node
Jan  1 01:00:54 fritz kern.debug kernel: [    0.000000] Early memory node ranges
Jan  1 01:00:54 fritz kern.debug kernel: [    0.000000]   node   0: [mem 0x00000000-0x0fffffff]
Jan  1 01:00:54 fritz kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Jan  1 01:00:54 fritz kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 8089d960, node_mem_map 81001ce0
Jan  1 01:00:54 fritz kern.debug kernel: [    0.000000]   Normal zone: 512 pages used for memmap
Jan  1 01:00:54 fritz kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Jan  1 01:00:54 fritz kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Jan  1 01:00:54 fritz kern.err kernel: [    0.000000] [module-alloc] use 0x69b000 bytes at 0x81203000
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] Detected 1 available secondary CPU(s)
Jan  1 01:00:54 fritz kern.debug kernel: [    0.000000] [avm_cache_set_coherency]: setting cache coherency for HWRevision=212 to write back allocate 
Jan  1 01:00:54 fritz kern.debug kernel: [    0.000000] Primary instruction cache 32kB, 4-way, VIPT, linesize 32 bytes.
Jan  1 01:00:54 fritz kern.debug kernel: [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] PERCPU: Embedded 7 pages/cpu @818a2000 s7328 r8192 d13152 u32768
Jan  1 01:00:54 fritz kern.debug kernel: [    0.000000] pcpu-alloc: s7328 r8192 d13152 u32768 alloc=8*4096
Jan  1 01:00:54 fritz kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 
Jan  1 01:00:54 fritz kern.debug kernel: [    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
Jan  1 01:00:54 fritz kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n8r nor_size=0MB sflash_size=1024KB nand_size=512MB ethaddr=C8:0E:14:18:1F:0C 
Jan  1 01:00:54 fritz kern.err kernel: [    0.000000] [NAND] nand_size = 0x20000000
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] Writing ErrCtl register=00040920
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] Readback ErrCtl register=00040920
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] Memory: 242784k/262144k available (6291k kernel code, 19360k reserved, 2628k data, 288k init, 0k highmem)
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] Hierarchical RCU implementation.
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] NR_IRQS:181
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] SRSConf0 0x3feffbfe: SRS1: 0xffffffff, SRS2 0xffffffff, SRS3 0xffffffff
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] Lantiq ICU driver, version 3.0.1, (c) 2001-2011 Lantiq Deutschland GmbH
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] Console: colour dummy device 80x25
Jan  1 01:00:54 fritz kern.info kernel: [    0.000000] console [ttyS0] enabled, bootconsole disabled
Jan  1 01:00:54 fritz kern.info kernel: [    0.020000] Calibrating delay loop... 331.77 BogoMIPS (lpj=1658880)
Jan  1 01:00:54 fritz kern.info kernel: [    0.080000] pid_max: default: 32768 minimum: 301
Jan  1 01:00:54 fritz kern.info kernel: [    0.080000] Security Framework initialized
[... cut ...]
Mar  1 10:25:09 fritz kern.info kernel: [  137.020000][0]kdsld: IPv6: internet: gu 2003:c1:37ff:1296:ca0e:14ff:fe18:1f10 remote ::
Mar  1 10:25:23 fritz daemon.info chronyd[4645]: chronyd version 3.2 starting (+CMDMON +NTP +REFCLOCK +RTC +PRIVDROP -SCFILTER -SECHASH -SIGND +ASYNCDNS +IPV6 -DEBUG)
Mar  1 10:25:23 fritz daemon.warn chronyd[4645]: commandkey directive is no longer supported
Mar  1 10:25:25 fritz kern.info kernel: [  153.070000][0]kdsld: IPv6: internet: gu 2003:c1:37ff:1296:ca0e:14ff:fe18:1f10 remote ::
Mar  1 10:25:28 fritz daemon.info chronyd[4645]: Selected source 194.25.134.196
```

fda77, 2019-03-01:
Dann check doch einfach einen alten Stand aus und prüf es
Ich kann mich nämlich nicht erinner dass ich in dem Bereich was geändert hab

closed 2019-03-01 by fda77