Freetz-NG/freetz-ng #6: AVM front page inaccessible when removing DE language

jpsollie, 2019-11-29:
When removing DE language (I only wanted ENG support) via FREETZ_REMOVE_LANGUAGE_de on FritzOs 7.12 international, you get past the login prompt, but that's about it: it doesn't load the status page, and immediately returns to the login prompt. including the de language fixes the issue.

fda77, 2019-11-29:
What is the current selected language?
Maybe it helps to
 - flash FOS with all languages
 - select desired language in AVM webif
 - flash FOS only with selected language

The remove patch is quiet simple: https://github.com/Freetz-NG/freetz-ng/blob/master/patches/scripts/110-remove_languages.sh

jpsollie, 2019-11-29:
> What is the current selected language?
ENG
>Maybe it helps to... 
Just not removing the German language fixes it. 
On my mobile phone, /var/www/all/login.lua gives a parse error at line 231, but this error also dissapears when not removing German. 
flash FOS with all languages - > I created an identical image with DE not removed (and ENG also included) , and it worked again. ENG was still the default language, no German @ front page.

fda77, 2019-12-01:
The differen language removes where added generic, I have no international device for test.
It seems AVM has untranslated strings in the german db. So i disabled removing german, its now only an unselectable comment for information.

closed 2019-12-01 by fda77