Freetz-NG/freetz-ng #7: dd not found when updating

jpsollie, 2019-11-29:
when installing freetz-ng with 7.12 international, I get an error message on installation that dd is not found.
The issue is here:
/var/install.sh (line 537):
flashtool_dd=`find . -name dd -xdev 2>/dev/null`
changing it to:
flashtool_dd=`find / -name dd -xdev 2>/dev/null`
fixes the issue.
this is both in the original and modified firmware source

fda77, 2019-11-29:
Does this happen with *installed* Freetz-NG?
Or want you flash Freetz-NG with installed Freetz(er13)?
In the 2nd case, please enable to flash Freetz-NG for the 1st time
1. menuconfig
2. Firmware packaging
3. Compatibility for forks with buggy flash

```
  │ Some Freetz forks have a buggy flash method. If you have                                                                                                                                                                             │
  │ installed such one, you can not flash images by AVM or                                                                                                                                                                               │
  │ modified with Freetz-NG.                                                                                                                                                                                                             │
  │ Enable this option to be compatible and allow to flash                                                                                                                                                                               │
  │ Freetz-NG for the 1st time.                                                                                                                                                                                                          │
  │ If you plan to flash by the bootloader, this option isn't needed.                                                                                                                                                                    │
```

Freetz fixed this 2 month ago: https://github.com/Freetz/freetz/commit/17872f5c45f9627455072f0018d9d6dde364fe7f
Related change from february in Freetz-NG https://github.com/freetz-ng/freetz-ng/commit/0866172de7

jpsollie, 2019-11-29:
Indeed, was moving from freetz to freetz-NG because it seemed more powerful. As this apparently is not a bug, I'll close the issue

closed 2019-11-29 by jpsollie

WileC, 2020-02-10:
könnte man das eigentlich Autmatisieren, dass je nach Update-Art oder installierter Original-Firmware / Freetz / Freetz-NG das Update-Skript dementsprechend richtig durchläuft? Ohne zuvor die Option im Menuconfig auszuwählen?

fda77, 2020-02-10:
Klar.
Ich finde es aber nicht sinnvoll wenn **immer** von **Freetz-NG** ein **AVM** Script verändert wird, damit ein kaputtes **Freetz(er13)**-Image geflascht werden kann.
Deshalb nur optional, für die die es brauchen.

Bei er13 ist es ja mittlerweile auch gefixt, dh mit jedem aktuellen freetz* kann man ein aktuelles freetz* quer-flashen