Freetz-NG/freetz-ng #8: [7490] Warning at build

harryboo, 2019-12-07:
Hi,

i get this warning at build for the 7490:

WARNING: no blkid tool installed, automatic mounting may not work

No matter what I choose, the message stays.

What should i do now?

fda77, 2019-12-09:
This warning was added not a long tim ago. Sadly you cutted the installed packages above . It could show if you have installed fstype or e2fsprogs or util-linux or blkid-busybox-applet? Does mounting of usb sticks etc works?
Have you selected **Remove e2fsprogs**?
In case you dont want to use blkid of avm's e2fsprogs, you have to check if the tool you want to use supports the arguments avm uses in the mount scripts!

harryboo, 2019-12-09:
The config was made from scratch. I dont have removed e2fsprogs.
Here is the config.

[config.txt](https://github.com/Freetz-NG/freetz-ng/files/3940600/config.txt)

fda77, 2019-12-09:
The FREETZ_AVM_HAS_E2FSPROGS symbol was missing. I checked for e2fsck, but avm does not include it always, sometime only blkid. Due to this, the warning was shown and the "Remove e2fsprogs" was not visible. Should be fixed now with d11f2229089f24dd9c7f2a1b5c93f5a37ae15331

closed 2019-12-09 by fda77