Freetz-NG/freetz-ng #10: [ENHANCEMENT] Missing serial modul

harryboo, 2020-01-18:
Can the kernal modul for the cheap CH340/CH341 implemented?

[Patch for this feature can be found here...](https://www.ip-phone-forum.de/threads/freetz-erkennt-usb-rs232-nicht.239864/)

fda77, 2020-01-18:
You want to add a patch which adds a kernel module not yet in the kernel?
This needs to be done for every kernel avm/freetz uses!

Quick overview:
[CODE]
ls -ld make/linux/patches/*/*/
make/linux/patches/2.6.13.1/04.30/
make/linux/patches/2.6.13.1/04.33/
make/linux/patches/2.6.13.1/04.40/
make/linux/patches/2.6.13.1/04.49/
make/linux/patches/2.6.13.1/04.57/
make/linux/patches/2.6.13.1/04.67/
make/linux/patches/2.6.13.1/04.70/
make/linux/patches/2.6.13.1/04.76/
make/linux/patches/2.6.13.1/04.80/
make/linux/patches/2.6.13.1/04.87/
make/linux/patches/2.6.13.1/5140_04.67/
make/linux/patches/2.6.13.1/7140_04.33/
make/linux/patches/2.6.13.1/r7203/
make/linux/patches/2.6.19.2/7270_04.86/
make/linux/patches/2.6.28.10/7340_06.06/
make/linux/patches/2.6.28.10/7390_05.21/
make/linux/patches/2.6.28.10/7390_06.01/
make/linux/patches/2.6.28.10/7390_06.30/
make/linux/patches/2.6.28.10/7390_06.51/
make/linux/patches/2.6.28.10/7390_06.80/
make/linux/patches/2.6.32.41/7270_05.21/
make/linux/patches/2.6.32.41/7270_05.51/
make/linux/patches/2.6.32.60/7270_06.05/
make/linux/patches/2.6.32.61/3490_06.31/
make/linux/patches/2.6.32.61/4020_06.50/
make/linux/patches/2.6.32.61/6810_06.21/
make/linux/patches/2.6.32.61/7272_06.20/
make/linux/patches/2.6.32.61/7320_06.30/
make/linux/patches/2.6.32.61/7330_06.50/
make/linux/patches/2.6.32.61/7360_06.20/
make/linux/patches/2.6.32.61/7490_06.30/
make/linux/patches/2.6.39.3.x86/6490_06.87/
make/linux/patches/2.6.39.3.x86/6490_07.02/
make/linux/patches/2.6.39.3.x86/6490_07.10/
make/linux/patches/2.6.39.3.x86/6590_06.87/
make/linux/patches/2.6.39.3.x86/6590_07.02/
make/linux/patches/3.10.104/7583_07.10/
make/linux/patches/3.10.104/7590_07.01/
make/linux/patches/3.10.107/5491_07.01/
make/linux/patches/3.10.107/7490_07.01/
make/linux/patches/3.10.107/7490_07.11/
make/linux/patches/3.10.12/7580_06.53/
make/linux/patches/3.10.12/7580_06.81/
make/linux/patches/3.10.12/7590_06.92/
make/linux/patches/3.10.73/3390_06.51/
make/linux/patches/3.10.73/7490_06.51/
make/linux/patches/3.10.73/7490_06.60/
make/linux/patches/3.10.73/7490_06.83/
make/linux/patches/3.10.73/7490_06.93/
make/linux/patches/4.4.60/4020_07.01/
make/linux/patches/4.4.60/4040_07.01/
make/linux/patches/4.4.60/7530_07.02/
[/CODE]
Maybe it should only be added to the kernel you need, and not old 7170 kernel?

fda77, 2020-01-19:
I understood now... ch341 ist available in "newer" kernel, only 2.6.13.1 needs a patch.
See https://github.com/Freetz-NG/freezu-ng/commit/e62343476e15b5d407ee35912465da3c8b39ea52

closed 2020-01-19 by fda77