Freetz-NG/freetz-ng #12: build mit net-snmp für die 6590 bricht mit fehlermeldung ab.

Conan179, 2020-02-23:
Hallo erst mal.
Ich wollte für meine 6590 ein freetz-ng image bauen.
Das minimal Image, ohne änderung geht ohne probleme.

Jetzt habe ich net-snmp hinzugefügt.
Jetzt kommt beim build immer diese Fehler Meldung:

/home/freetz/6590/toolchain/build/i686_gcc-5.5.0_uClibc-1.0.14-nptl/i686-linux-uclibc/lib/gcc/i686-linux-uclibc/5.5.0/../../../../i686-linux-uclibc/bin/ld: cannot find -lperl
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:617: libnetsnmpagent.la] Error 1
make[2]: Leaving directory '/home/freetz/6590/source/target-i686_gcc-5.5.0_uClibc-1.0.14-nptl/netsnmp-5.7.1/agent'
make[1]: *** [Makefile:652: subdirs] Error 1
make[1]: Leaving directory '/home/freetz/6590/source/target-i686_gcc-5.5.0_uClibc-1.0.14-nptl/netsnmp-5.7.1'

ERROR: Build failed.
make: *** [make/netsnmp/netsnmp.mk:139: source/target-i686_gcc-5.5.0_uClibc-1.0.14-nptl/netsnmp-5.7.1/agent/.libs/snmpd] Error 1

Google suche hat nichts gefunden, was das behebet. Ich hoffe ich mache das so jetzt richtig.
Gruss Conan

fda77, 2020-02-24:
Hast du perl und libstring-crc32-perl installiert?
Evtl brauch man die auf dem Host, bin aber nicht sicher!
Siehe https://freetz-ng.github.io/freetz-ng/wiki/10_Beginner/install.de#installation-der-ben%C3%B6tigten-pakete-ubuntu

Conan179, 2020-02-24:
Ich nutzt ein vorkonfigurierte VM um freetz zu bauen.
und hab nach geschaut:
root@freetz-linux:~# apt install perl
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
perl ist schon die neueste Version (5.28.1-6build1).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
root@freetz-linux:~# apt install libstring-crc32-perl
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
libstring-crc32-perl ist schon die neueste Version (1.7-1build1).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
root@freetz-linux:~#

Beides ist innstalliert

fda77, 2020-02-24:
Ich vermute mal dass du das "perl" Problem lösen konntest, da du hier eine andere Fehlermeldung gepostet hast: https://github.com/Freetz/freetz/issues/267

Conan179, 2020-02-25:
Nein, das problem konnt ich wirklich nicht lösen, ich bekomme beim freetz nur eine andere fehlermeldung.

fda77, 2020-02-25:
Bei beiden Freetzes kann netsnmp für x86 (also docsis fritzboxen) nicht compiliert werden. Fehler
```
checking how to printf a size_t... configure: error: underlying type of size_t not known
```
Die beiden geposteten Fehlermeldungen von dir sind was anderes

Conan179, 2020-02-25:
Beide fehlermeldunen kommen aber nur wen ich net-snmp hinzufüge.
Ein reines Image ohne änderungen wird ohen probleme gebaut.

fda77, 2020-02-25:
Der Commit von heute sollte helfen.
Am besten noch ein:
```make netsnmp-dirclean```

closed 2020-02-25 by fda77

Conan179, 2020-02-25:
PERFEKT! Es läuft durch. Ich danke dir für deine Hilfe :)