Freetz-NG/freetz-ng #15: Dvb-c repeater Alien Firmware findet kein wlan

frozen1900, 2020-04-16:
Ich habe gerade das alien Image für den dvb-c repeater erstellt und hochgeladen. Leider findet er jetzt aber kein anderes WLAN mehr. Weder im Monitor sieht er andere WLANs noch im Assistenten

fda77, 2020-04-16:
Keine Ahnung, ich hab kein Gerät zum testen da.
Schau doch mal was in den ganzen Logs, Syslog, dmesg, eventsdump usw steht
Betrifft es 2,4 >und< 5ghz?

Der 1750 und dvbc sollten im Bereich Wlan soweit gleich sein:
https://boxmatrix.info/wiki/FRITZ!WLAN_Repeater_DVB-C
https://boxmatrix.info/wiki/FRITZ!WLAN_Repeater_1750E
Siehe "Hardware", boxmatrix ist aber dafür bekannt öfter ungenau zu sein!

frozen1900, 2020-04-16:
crash.log:
```
1970-01-01 01:00:18(1) [Segmentation fault] meshd([678]) CRASHED at 00000000 (?) accessing 00000000
SIGNO 11 ERRNO 0 CODE 1
```

AVM Log:

> wlan-modul konnte nicht korrekt initialisiert werden. bitte fritz!repeater neu starten #0102

dmesg:
```
[   82.439528] avm_ath_ext_detach_ah:111
[   82.439578] band steering terminated  for direct attach hardware 
[   82.439589] ieee80211_bsteering_detach: Band steering terminated
[   82.439608] acfg_detach Netlink socket released
[   82.439813] ath_net80211_dfs_clist_update: called, cmd=1, nollist=  (null), nentries=0
[   82.445382] SPECTRAL : Module removed (spectral = 81c1c000)
[   82.445414] GreenAP on wifi0: Detached
[   82.450950] ath_detach: remove global_ic[0]..gloabl_ic ptr:8126b2e0
[   82.463518] Removing athdebug proc file
[   82.463560] ath_dev: driver unloaded
[   82.475714] ath_rate_atheros: driver unloaded
[   82.486524] ath_hal: driver unloaded
[   82.487771] release_bug_debug_table: warning: 'ath_hal' not found! (driver maybe bugfree)
[   82.499158] avm_ath_ext_detach_ieee:86
[   82.516793] ath_spectral: driver unloaded
[   82.518064] release_bug_debug_table: warning: 'ath_spectral' not found! (driver maybe bugfree)
[   82.527677] ath_dfs: driver unloaded
[   82.528947] release_bug_debug_table: warning: 'ath_dfs' not found! (driver maybe bugfree)
[   82.550607] release_bug_debug_table: warning: 'asf' not found! (driver maybe bugfree)
[   82.561216] release_bug_debug_table: warning: 'mem_manager' not found! (driver maybe bugfree)
[   82.570563] Net trace device 'HW (2.4 + 5 GHz, wifi0)' with minor 129 was not deregistered properly.
[   82.571284] avm_net_trace: destroy udev device avm_net_trace128
[   82.571298] avm_net_trace_work: ERROR must be reset to 0
[   82.572292] avm_net_trace: destroy udev device avm_net_trace129
[   82.572308] avm_net_trace_work: ERROR must be reset to 0
[   82.625026] [wlan_config] Given config is:
[   82.625047] [wlan_config]   hw_interface=0 chip_type=8 (scorpion) offload=1 (non)
[   82.625059] [wlan_config]   hw_interface=1 chip_type=11 (peregrine) offload=1 (non)
[   82.628539] avm_net_trace: New net trace device 'WLAN Management Traffic' registered with minor 128.
[   82.628880] avm_net_trace: udev device avm_net_trace128 created
[   82.632291] [wlan_config] HWRevision now 205
[   82.632325] [wlan_config] HWSubRevision now 1
[   82.632366] [wlan_config] maca now =HIDDEN-IP6=
[   82.652660] [wlan_eeprom] EEPROM1: Awaiting calibration data via char dev
[   82.652748] [wlan_eeprom] New valid EEPROM1 loaded
[   82.652759] [wlan_eeprom] EEPROM #1, type "AR93xx/AR95xx":
[   82.652787] [wlan_eeprom] Customer data="DVBC_Repeater_CAL1_V"
[   82.652827] [wlan_eeprom] MinCCAPwr thresh set to 2GHz:-70,-70,-70 5GHz:-1,0,0
[   82.652836] [wlan_eeprom] Build with ART2 4.4
[   82.652903] [wlan_eeprom] EEPROM2: Awaiting calibration data via char dev
[   82.652965] [wlan_eeprom] New valid EEPROM2 loaded
[   82.652974] [wlan_eeprom] EEPROM #2, type "QCA98xx":
[   82.652999] [wlan_eeprom] Customer data="DVBC_Repeater_CAL2_V"
[   82.653008] [wlan_eeprom] patching regDmnExt to enable ext chan 144 for fcc
```

frozen1900, 2020-04-16:
crash.log:
```
1970-01-01 01:00:18(1) [Segmentation fault] meshd([677]) CRASHED at 00000000 (?) accessing 00000000
SIGNO 11 ERRNO 0 CODE 1
Version: 07.12
No bugmsg
ze: 00000000 at: 7fe9cfe2 v0: 00000000 v1: 559db090
a0: 774cc8e4 a1: 00000001 a2: 7fe9d0e4 a3: 5596e73d
t0: 00000001 t1: 321312ed t2: 00000001 t3: 6d657368
t4: 00000000 t5: 00002000 t6: fffffffc t7: 00000000
s0: 774e0d00 s1: 7fe9d0e4 s2: 0093b024 s3: 0093b040
s4: ffffffff s5: 0093b040 s6: 0093b024 s7: 0048ab0a
t8: 00000001 t9: 00000000 k0: 7fe9cefa k1: 00000000
gp: 559db090 sp: 7fe9d020 fp: 0047bdb0 ra: 5596e76b
FA 00000000
PC 00000000
PC Code: <illegal function pointer>
RA 5596e76b std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count(std::__shared_count<(__gnu_cxx::_Lock_policy)2> const&)+0x2e (/sbin/meshd at 0006476b)
RA Code: 6d01 ea40 653a <1003> 9a61 4b01 da61 6444 e8a0
[bt] *5596e76a std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count(std::__shared_count<(__gnu_cxx::_Lock_policy)2> const&)+0x2e (/sbin/meshd at 0006473d/0x6476b)
                        Code: 6d01 ea40 653a <1003> 9a61 4b01 da61 6444 e8a0
[bt] *77cfb9ce [77cfb9ab] <avm::Aicmd::Aicmd(std::shared_ptr<avmipc> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xe>+0x24 (//lib/libmesh_shared.so.0 at 000629ab/0x629cf)
                        Code: 675c f012 9af8 <ef40> 653f 4047 9608 67b1 4a05
[bt] *77d38844 [77d38641] <avm::daemon::Mainloop::Mainloop()+0x10>+0x204 (//lib/libmesh_shared.so.0 at 0009f641/0x9f845)
                        Code: 950b 6782 6722 <ef40> 653f 9604 659e f011 98b0
[bt] *77d389d4 [77d389a1] <avm::daemon::Mainloop::get()+0x10>+0x34 (//lib/libmesh_shared.so.0 at 0009f9a1/0x9f9d5)
                        Code: f3b0 98a4 9407 <ed40> 653d 9604 659e f651 98b8
[bt] *5595fe82 [5595fca5] <main+0x10>+0x1de (/sbin/meshd at 00055ca5/0x55e83)
                        Code: 1023 f193 994c <ea40> 653a 6d00 9604 6782 f272
[bt]  77707d1c __uClibc_main+0x1fc (//lib/libc.so.1 at 00063b20/0x63d1c)
                        Code: 27a30018 ac438c20 8f828af4 <0320f809> 8c460000 1000001c 8fbc0010 8f908a04 8e1900c8
[bt] finished.
1970-01-01 01:00:18(1) [Segmentation fault] meshd([678]) CRASHED at 00000000 (?) accessing 00000000
SIGNO 11 ERRNO 0 CODE 1
Version: 07.12
No bugmsg
ze: 00000000 at: 7f9c3c22 v0: 00000000 v1: 55829090
a0: 773988e4 a1: 00000001 a2: 7f9c3d24 a3: 557bc73d
t0: 00000001 t1: 96566fdb t2: 00000001 t3: 6d657368
t4: 00000000 t5: 00002000 t6: fffffffc t7: 00000000
s0: 773ac480 s1: 7f9c3d24 s2: 00584024 s3: 00584040
s4: ffffffff s5: 00584040 s6: 00584024 s7: 0048ab0a
t8: 00000001 t9: 00000000 k0: 7f9c3b3a k1: 00000000
gp: 55829090 sp: 7f9c3c60 fp: 0047bdb0 ra: 557bc76b
FA 00000000
PC 00000000
PC Code: <illegal function pointer>
RA 557bc76b std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count(std::__shared_count<(__gnu_cxx::_Lock_policy)2> const&)+0x2e (/sbin/meshd at 0006476b)
RA Code: 6d01 ea40 653a <1003> 9a61 4b01 da61 6444 e8a0
[bt] *557bc76a std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count(std::__shared_count<(__gnu_cxx::_Lock_policy)2> const&)+0x2e (/sbin/meshd at 0006473d/0x6476b)
                        Code: 6d01 ea40 653a <1003> 9a61 4b01 da61 6444 e8a0
[bt] *77bc79ce [77bc79ab] <avm::Aicmd::Aicmd(std::shared_ptr<avmipc> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xe>+0x24 (//lib/libmesh_shared.so.0 at 000629ab/0x629cf)
                        Code: 675c f012 9af8 <ef40> 653f 4047 9608 67b1 4a05
[bt] *77c04844 [77c04641] <avm::daemon::Mainloop::Mainloop()+0x10>+0x204 (//lib/libmesh_shared.so.0 at 0009f641/0x9f845)
                        Code: 950b 6782 6722 <ef40> 653f 9604 659e f011 98b0
[bt] *77c049d4 [77c049a1] <avm::daemon::Mainloop::get()+0x10>+0x34 (//lib/libmesh_shared.so.0 at 0009f9a1/0x9f9d5)
                        Code: f3b0 98a4 9407 <ed40> 653d 9604 659e f651 98b8
[bt] *557ade82 [557adca5] <main+0x10>+0x1de (/sbin/meshd at 00055ca5/0x55e83)
                        Code: 1023 f193 994c <ea40> 653a 6d00 9604 6782 f272
[bt]  775d3d1c __uClibc_main+0x1fc (//lib/libc.so.1 at 00063b20/0x63d1c)
                        Code: 27a30018 ac438c20 8f828af4 <0320f809> 8c460000 1000001c 8fbc0010 8f908a04 8e1900c8
[bt] finished.
```

frozen1900, 2020-04-16:
Betroffen sind 2,4GHz und 5GHz. Nachdem ich vorhin mit dem WLAN vom Repeater verbunden war funktioniert dies aktuell auch nicht mehr

fda77, 2020-04-17:
Die meshd-segfaults kann man vermutlich ignorieren, sowas hab ich auch auf anderen Geräten gesehen. Kann gut sein dass die "normal" sind und man die ohne Freetz nur nicht sehen kann

> Nachdem ich vorhin mit dem WLAN vom Repeater verbunden war funktioniert dies aktuell auch nicht mehr

Ich versteh den Satz nicht. Also hat es kurz mal funktioniert? Und plötzlich nicht mehr?

Vielleicht braucht man die Dateien athwlan.bin und utf.bin aus  lib/firmware/AR9888/hw.2/
Diese haben jedenfalls nicht die gleiche Größe.

Zum testen: Den Patch unten anwenden (patch -p0 < alien.patch.txt), ins menuconfig + speichern und make

[alien.patch.txt](https://github.com/Freetz-NG/freetz-ng/files/4494208/alien.patch.txt)

fda77, 2020-04-17:
Was ganz anderes: Hast du mal ein Downgrade der Firmware gemacht, ohne die avm-Konfiguration zurückzusetzen?
Dadurch kommt beim Wlan gern was durcheinander, zb macht Wlan immer autokanal, egal was als Kanal eingestellt ist

frozen1900, 2020-04-17:
Beim downgrade gab es keine Probleme. Beim Upgrade gibt es die Probleme mit uns ohne reset.
Den Patch werde ich am Montag Mal testen und gebe Rückmeldung

frozen1900, 2020-04-20:
Patch eingespielt, leider findet der Repeater immer noch kein WLAN und sendet auch kein WLAN. Habe danach nochmal auf Werkseinstellungen zurückgesetzt. Auch dadurch leider keine Besserung

fda77, 2020-04-20:
Komisch an den Logs oben ist dass sie "DVBC_Repeater_CAL1_V" und "DVBC_Repeater_CAL2_V" enthalten. Wenn man ein Alien geflasht hat, sollte nichts mehr von DVBC zu finden sein, da die Firmware zu über 99% die 1750 Firmware ist.
Es gibt nur minimale Änderungen: https://github.com/Freetz-NG/freetz-ng/blob/master/patches/scripts/100-1759_1750.sh
Sind die Logs mit dvb firmware?
Hab keine Idee wie man ohne Hardware oder remote herausfinden könnte wodurch das Wlan Problem verursacht wird

frozen1900, 2020-04-22:
Die Logs waren immer mit der alien firmware
wir können gerne gemeinsam mal remote drauf gucken. muss dafür aber mal in den nächsten tagen die VM auf den PC kopieren

fda77, 2020-04-25:
Danke, das zähle ich aber auch zu remote und würde nicht helfen.
Es gab Berichte, dass eine *unmodifizierte* 1750 firmware (mit ein paar Tricks) auf den dvb-c aufspielbar ist, und damit dann alles bis auf dvb (weil das webif, usb & dvb fehlt) funktioniert. Wlan ging dann wohl auch.
Siehe zB:  https://www.ip-phone-forum.de/threads/299123/page-37#post-2330066
Dh eigentlich sollte ein Alien funktionieren! Die Änderungen sind minimal, siehe https://github.com/Freetz-NG/freetz-ng/blob/master/patches/scripts/100-1759_1750.sh
Da dein Wlan mit dvb-v firmware geht kann es nciht kaputt sein. Also stimmt was am alien nicht. Hast du mal mit minimalen änderungen in Freetz probiert, also höchsten dorpbear und syslog?

fda77, 2020-04-29:
Besorg von AVM mal den Sourcecode von
 - 1750e 134.07.01
 - DVB-c 133.07.01
Meld dich wieder wenn es diese hier gibt: http://osp.avm.de/

Und häng ein Log vom "Step 3" von make an!

frozen1900, 2020-06-29:
Hey fda77. Der Sourcecode sind mittlerweile Online. Ich habe nochmal versucht ein aktuelles Image mit 7.1.12 (und der aktuellesten Laborversion) zu erstellen, beide haben das gleiche problem = kein Wlan.

`STEP 3: PACK/SIGN
  checking for left over version-control-system files
  integrate freetz info file into image
packing var.tar
  checking signature key files
    adding public signature key file
creating filesystem image (SquashFS4-xz)
  SquashFS block size: 64 kB (65536 bytes)
merging kernel image
  kernel image size: 13.7 MB, max 14.9 MB, free 1.2 MB (1206016 bytes)
adding checksum to kernel.image
packing images/DVB-c-Alien1750e_07.19.all-rev79031_labor_freetz-ng-16862-ec7d7efd2_20200629-104324.image
  packed image file size: 14.8 MB (15503360 bytes)
signing packed .image file
  signed image file size: 14.8 MB (15503360 bytes)
source firmware: 1750E_en-de-es-fr-it-pl 134.07.19-BETA rev79031 {ALL} [PSQ19] (29.05.2020 11:41:09) 
  source image file size: 14.4 MB (15124480 bytes)
done.

FINISHED
`

`STEP 3: PACK/SIGN
  checking for left over version-control-system files
  integrate freetz info file into image
packing var.tar
  checking signature key files
    adding public signature key file
creating filesystem image (SquashFS4-xz)
  SquashFS block size: 64 kB (65536 bytes)
merging kernel image
  kernel image size: 12.0 MB, max 14.9 MB, free 2.9 MB (2992896 bytes)
adding checksum to kernel.image
packing images/DVB-c-Alien1750e_07.12.all_freetz-ng-16862-ec7d7efd2_20200629-104423.image
  packed image file size: 13.4 MB (14069760 bytes)
signing packed .image file
  signed image file size: 13.4 MB (14069760 bytes)
source firmware: 1750E_en-de-es-fr-it-pl 134.07.12 rev70775 {ALL} [MESH18 NL2] (12.08.2019 11:22:24) 
  source image file size: 13.4 MB (14039040 bytes)
done.

FINISHED`

kamasudo, 2020-06-29:
Hab schon gesehen, muss noch eingebaut werden

fda77, 2020-06-30:
Hab schon gesehen dass es die Sourcen gibt. Die müssten dann noch eingebaut werden.
Und jemand müsste noch die vom 1750 und dvbc vergleichen ob/welche Unterschiede es gibt
Du baust also selbst und bist nciht der gleiche der im DEB mit einem fertigen Image die gleichen Problem schilderte?

frozen1900, 2020-07-01:
Ich habe bisher immer selbst gebaut. Den Thread im DEB habe ich gelesen, weiß allerdings auch nicht was der User da meinte.

Wenn du mir sagst wo ich ungefähr suchen kann werde ich mich am Freitag mal hinsetzen und die beiden Sourcecodes vergleichen.

fda77, 2020-07-01:
Alles :D
Am betsen nimmst du zuerst die beiden 7.0x Quellpakete. Darin ist eine release-kernel Datei, die den Kernel+.config enthält.
Zuerst mal die .configs vergleichen, aber vorsicht AVM arbeitet hier oft sehr unsauber
 - mal sind symbole aktiviert obwohl sie in der firmware selbst aus sind
 - mal sind symbole deativiert obwohl sie in der firmware selbst enhalten sind
 - mal gibt es in der .config symbole die es gar nicht im Quelltext gibt.
Also schonmal alle Glaskugeln bereitstellen...
Und dann die kompletten Kernel-Quellverzeichnisse selbst, da können Unterschiede durch Patches von avm sind
Falls es bei allem dem keine Unterschiede bezüglich Wlan gibt, müsste das Problem in den nur binären Kernelmodulen oder den Scripten sein

fda77, 2020-07-04:
Bitte mal mit der aktuellsten Revision ausprobieren

frozen1900, 2020-07-06:
weiterhin bekomme ich folgenden fehler mit 7.12 bzw beta 7.19 images

`WLAN-Modul konnte nicht korrekt initialisiert werden. Bitte FRITZ!Repeater neu starten! (#0102)`

Leider hatte ich am Wochenende nicht die Zeit mir die Kernel Sources anzusehen

fda77, 2020-07-06:
Schade die Änderung an Busybox nicht auch hier hilft
Mach doch zum Vergleichen einfach ein ```diff -Naur dir1/ dir2/ > ausgabe.txt```
Evtl auch noch die Konfiguration der Busybox von beiden vergleichen. Diese sind nicht im "busybox" archv, sondern "gcc"

frozen1900, 2020-07-06:
Ich habe den kompletten Inhalt verglichen, das waren die einzigen Dateien die bei mir im diff unterschiedlich waren

fda77, 2020-07-07:
Ich meinte den Inhalt der Dateien, also WAS sich in diesen geändert hat :}
Es wurden nochmal einige Busybox Applets zusätzlich aktiviert, die AVM auch aktiviert hat.
Vielleicht hilfts ja, könntest nochmal probieren ob es nun läuft

frozen1900, 2020-07-08:
Funktioniert leider weiterhin nicht. Mit beiden Versionen bekomme ich kein WLAN (Empfang und Senden)

fda77, 2020-07-09:
Aber DVB funktioniert, sowohl mit Alien als auch mit der normalen Firmware? Wegen https://github.com/Freetz-NG/freetz-ng/issues/23

frozen1900, 2020-07-09:
DVB-C habe ich nie genutzt. Habe ich hier auch nicht liegen, sonst hätte ich das kurz testen können. 
Der Kauf ein "Fehlkauf". Ich hab den Karton vom 1750e bekommen (und diesen auch bestellt) aber der dvb-c lag im Karton.
Da damals aber noch die Ansage war "die sind eh gleich" habe ich den dvb-c repeater behalten

fda77, 2020-07-09:
Die sind auch fast gleich, bis auf den hub mit den Tunern. Deshalb versteh ich auch nicht so ganz warum es mittlerweile diese Problem mit Wlan/DVB gibt. Ich hatte selbst mal einen 1750 und DVB, das war aber noch FritzOS 6.23/6.30. Auf beiden lief Wlan ohne Probleme

fda77, 2020-07-14:
Unterschiedliche Dateien der FOS 7.01 Sourcen 1750 > DVBC
```
Dateien 1750_07_01/linux/drivers/char/avm_new/avm_dist_event/avm_event_gen_debug.c und 1759_07_01/linux/drivers/char/avm_new/avm_dist_event/avm_event_gen_debug.c sind verschieden.
Dateien 1750_07_01/linux/drivers/char/avm_new/avm_dist_event/avm_event_gen_endian.h und 1759_07_01/linux/drivers/char/avm_new/avm_dist_event/avm_event_gen_endian.h sind verschieden.
Dateien 1750_07_01/linux/drivers/char/avm_new/avm_dist_event/avm_event_gen_enum_range.c und 1759_07_01/linux/drivers/char/avm_new/avm_dist_event/avm_event_gen_enum_range.c sind verschieden.
Dateien 1750_07_01/linux/drivers/char/avm_new/avm_dist_event/avm_event_gen_enum_range.h und 1759_07_01/linux/drivers/char/avm_new/avm_dist_event/avm_event_gen_enum_range.h sind verschieden.
Dateien 1750_07_01/linux/drivers/char/avm_new/avm_dist_event/avm_event_gen_flow.dot und 1759_07_01/linux/drivers/char/avm_new/avm_dist_event/avm_event_gen_flow.dot sind verschieden.
Dateien 1750_07_01/linux/drivers/char/avm_new/avm_dist_event/avm_event_gen_types.h und 1759_07_01/linux/drivers/char/avm_new/avm_dist_event/avm_event_gen_types.h sind verschieden.
Dateien 1750_07_01/linux/drivers/char/avm_new/avm_dist_event/avm_event_perl_convert.pm und 1759_07_01/linux/drivers/char/avm_new/avm_dist_event/avm_event_perl_convert.pm sind verschieden.
Dateien 1750_07_01/linux/drivers/char/avm_new/avm_event.h und 1759_07_01/linux/drivers/char/avm_new/avm_event.h sind verschieden.
Dateien 1750_07_01/linux/include/linux/avm_event.h und 1759_07_01/linux/include/linux/avm_event.h sind verschieden.
Dateien 1750_07_01/linux/include/uapi/linux/avm_event.h und 1759_07_01/linux/include/uapi/linux/avm_event.h sind verschieden.
Dateien 1750_07_01/linux/.kernelvariables und 1759_07_01/linux/.kernelvariables sind verschieden.
```

Dateien im Filesystem die "event" im Namen enthalten
```
  9944 14. Mär 2019  build/original/filesystem/bin/avmipc_send_event
    21 14. Jul 19:59 build/original/filesystem/lib/libavm_event.so -> libavm_event.so.1.0.0
    21 14. Jul 19:59 build/original/filesystem/lib/libavm_event.so.1 -> libavm_event.so.1.0.0
 10212 14. Mär 2019  build/original/filesystem/lib/libavm_event.so.1.0.0
  5768 14. Mär 2019  build/original/filesystem/sbin/eventadd
  5752 14. Mär 2019  build/original/filesystem/sbin/eventsdump
  1471 14. Mär 2019  build/original/filesystem/usr/www/avme/js/event_triggered_data.js
  1471 14. Mär 2019  build/original/filesystem/usr/www/avm/js/event_triggered_data.js
```

Diese könnte man zum Testen für das Alien aus der original Firmware mittels ```patches/scripts/100-1759_1750.sh``` übernehmen wie der Patch in diesem Beispiel: https://github.com/Freetz-NG/freetz-ng/issues/15#issuecomment-615353168

fda77, 2020-07-25:
Bitte die neue Option "Busybox applets" > "keep avm binaries" aus https://github.com/Freetz-NG/freetz-ng/commit/a34135fcc88303373986232d9c97a5dc289b46ec aktivieren und testen, vielleicht hilfts

frozen1900, 2020-08-03:
hab gerade wieder version 7.12 und 7.19 mit der option kompiliert und hochgeladen. leider findet das gerät immer noch kein einziges wlan und sendet auch kein signal raus

fda77, 2020-08-03:
Danke! Wie es aussieht werden durch die Busybox Option "enable NOEXEC/NOFORK performance optimizations" Applets bevorzugt, auch wenn es Binaries gibt. Das hat AVM nicht an Freetz als default

gstammw, 2020-12-23:
Ich habe das Problem unter dem aktuellen Freetz-NG vom 22.12.2020, mit dem ich das Alien-Image für den DVB-C auf Basis des 7.21-er 1750E-Image (FRITZ.Box_WLAN_Repeater_1750E-07.21) gebaut habe, ebenfalls.
Mein Image "DVB-c-Alien1750e_07.21.all_freetz-ng-17595-34903d42f_20201222-110448.image" erzeugt auf dem DVB-C-Repeater folgende Fehlermeldung:
`**WLAN-Modul konnte nicht korrekt initialisiert werden. Bitte FRITZ!Repeater neu starten! (#0102)**`
WLAN funktioniert nicht, ich kann weder nach anderen Netzen scannen noch lokal einen AP hochfahren (Modus LAN-Brücke).

Die Build-Config ist default, d.h. unter "Busybox Applets" ist "keep binaries" bereits gesetzt.

Output von dmesg anbei:
`root@fritz:/var/mod/root# dmesg
[    0.588437] yaffs: yaffs Installing.
[    0.588811] io scheduler noop registered
[    0.592745] io scheduler deadline registered (default)
[    0.601579] ttyS0 at MMIO 0x18020000 (irq = 19, base_baud = 2500000) is a PORT_16550A
[    0.610075] [loadcontrol] set auto - scale=1
[    0.614372] [avm] configured: watchdog event debug
[    0.619253] AVM_WATCHDOG: Watchdog Driver for AR7 Hardware (Version 1.0)
[    0.626386] Register push button event to receive the set_factory_kernel event
[    0.634201] AVM Simple Profiling enabled Version 3.0
[    0.639199] [simple-profiling]:4 performance counters implemented, OLD_34K
[    0.646499] avm_net_trace: Up and running.
[    0.657441] loop: module loaded
[    0.660925] Registering ATH-flash-driver...
[    0.665177] [ath_flash] Use 3 byte addressing
[    0.692569] 6 avm_squashfs partitions found on MTD device ath-nor
[    0.698678] [TFFS3_Register_Panic_CB] registering panic callback for mtd ath-nor
[    0.706158] Creating 6 MTD partitions on "ath-nor":
[    0.711093] 0x00000019dc00-0x000000f00000 : "rootfs"
[    0.717146] [rootfs] use mtd0 (rootfs) as root
[    0.721647] mtdsplit: no squashfs found in "rootfs"
[    0.726525] 0x000000020000-0x00000019dc00 : "kernel"
[    0.732585] 0x000000000000-0x000000020000 : "urlader"
[    0.738602] 0x000000f00000-0x000000f80000 : "tffs (1)"
[    0.744859] 0x000000f80000-0x000001000000 : "tffs (2)"
[    0.751013] 0x000000000000-0x000001000000 : "reserved-kernel"
[    0.757801] Atheros on-chip NAND FLash Controller Driver, Version 1.0 (c) 2014 AVM GmbH, 2010 Atheros Communications, Ltd.
[    0.769061] tun: Universal TUN/TAP device driver, 1.6
[    0.774121] tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com>
[    0.780603] i2c /dev entries driver
[    0.784091] AVM PA for Linux Linux version 4.4.60 (jenkins@edxbuildxjplxpsq19p1xscrpnx1750E) (gcc version 5.5.0 (Buildroot 2018.11.4-gf81a09a5a) ) #1 Tue Nov 17 16:55:04 CET 2020
                (late init)
[    0.803238] gre: GRE over IPv4 demultiplexor driver
[    0.808212] NET: Registered protocol family 10
[    0.813565] avm_pa: try to activate hw accelaration for pid 2 (ipv6) called from 0x802f27ec
[    0.822095] NET: Registered protocol family 17
[    0.826579] Bridge broadcast ratelimiter registered
[    0.831461] bridge: automatic filtering via arp/ip/ip6tables has been deprecated. Update your scripts to load br_netfilter if you need this.
[    0.844243] Bridge firewalling registered
[    0.848319] l2tp_core: L2TP core driver, V2.0
[    0.852662] l2tp_ip: L2TP IP encapsulation support (L2TPv3)
[    0.858310] l2tp_netlink: L2TP netlink interface
[    0.862999] l2tp_eth: L2TP ethernet pseudowire support (L2TPv3)
[    0.868944] l2tp_ip6: L2TP IP encapsulation support for IPv6 (L2TPv3)
[    0.875477] 8021q: 802.1Q VLAN Support v1.8
[    0.880621] [TFFS3_Init] Called.
[    0.883818] [TFFS3_Init] No storage module registered, trying legacy fallback
[    0.891057] [TFFS3_CACHE_Configure] Setting up caching for backend legacy
[    0.897909] [TFFS3-CACHE] Caching module for TFFS 3.x
[    0.903020] [TFFS3_LGCY_Setup] using mtd3(tffs (1)), mtd4(tffs (2))
[    0.909358] [TFFS3_LGCY_Setup] mtd "tffs (1)": segment value 4
[    0.915252] [TFFS3_LGCY_Setup] mtd "tffs (2)": segment value 3
[    0.921143] [TFFS3_LGCY_Setup] Using segment 4 (avail: 4 + 3)
[    0.926949] [TFFS3_LGCY_Setup] mtd3 size=0x80000
[    0.932339] TFFS: tiny flash file system driver. GPL (c) AVM Berlin (Version 3.0)
[    0.939902] Adam2 environment variables API installed.
[    0.946169] {avmnet_cfg_netinit}
[    0.949408] [avmnet] No config found for HWRev 205, HWSubRev 1, Profile-ID 0, trying base config for HWSubRev
[    0.959404] [avmnet] No config found for HWRev 205, HWSubRev 1, trying base config for HWRev
[    3.395563] [avmnet] [avmnet_ar8033_powerdown] Called for module ar8033
[    3.402255] [avmnet_set_macaddr] Setup Mac Addr for Device(eth0): c0:25:06:f3:40:f8
[    3.414978] VFS: Mounted root (squashfs filesystem) readonly on device 31:0.
[... 325 lines cut ...]
[  159.006192] acfg_detach Netlink socket released
[  159.006382] ath_net80211_dfs_clist_update: called, cmd=1, nollist=00000000, nentries=0
[  159.011950] SPECTRAL : Module removed (spectral = 835d8000)
[  159.011979] GreenAP on wifi0: Detached
[  159.017512] ath_detach: remove global_ic[0]..gloabl_ic ptr:812825f0
[  159.048706] Removing athdebug proc file
[  159.048747] ath_dev: driver unloaded
[  159.051328] ath_rate_atheros: driver unloaded
[  159.065814] ath_hal: driver unloaded
[  159.067018] release_bug_debug_table: warning: 'ath_hal' not found! (driver maybe bugfree)
[  159.069563] avm_ath_ext_detach_ieee:86
[  159.078607] ath_spectral: driver unloaded
[... cut ...]
[  159.085230] release_bug_debug_table: warning: 'mem_manager' not found! (driver maybe bugfree)
[  159.085303] Net trace device 'HW (2.4 GHz, wifi0)' with minor 129 was not deregistered properly.
[  159.086012] avm_net_trace: destroy udev device avm_net_trace128
[  159.086023] avm_net_trace_work: ERROR must be reset to 0
[  159.086200] avm_net_trace: destroy udev device avm_net_trace129
[  159.086211] avm_net_trace_work: ERROR must be reset to 0
[  159.191843] [avm_connect] (idle): received <wlan_error> state 1
[  159.192736] AVM_WATCHDOG_release(hdl=11 wland): success
`

fda77, 2020-12-24:
@gstammw Du hast dein Anliegen hier und in mindestens 2 Foren gepostet. Wie im IPPF zu lesen könnte ein falscher device-tree die Ursache sein. Um das zu testen, würde ich einfach mal den kompletten Kernel mit dem aus einer DVB-C Firmware ersetzen. Geeignet sind die 7.0-stable und 7.08-inhaus, da diese auch Version 4.4.60 haben. Kann auch sein dass dadurch manche Kernelmodule erst gar nicht geladen werden

fda77, 2021-06-10:
Der dvbc im Gegensatz zum 1750 hat in manchen fw usb im Kernel, sonst vermutlich als Modul: https://github.com/Freetz-NG/freetz-ng/commit/82e7c7d82433d33c1d12bd7f208d196e029a97da

baribalbear, 2021-08-06:
Hi @fda77.  I have the same issue: DVB-c-Alien1750E_07.27.all_freetz-ng-18484-4ae9ecb5e_20210805-193049.image freshly build can't initialize the WLAN module. How can we move forward from here? Could you guide how to replace the kernel as you wrote above? I flashed just the kernel.image from DVB-c-Alien1750E_07.27.all_freetz-ng-18484-4ae9ecb5e_20210805-193049.image via PS scripts.

baribalbear, 2021-08-06:
I updated the alien patch with both recommendations from this thread. Please find it and support logs attached. I don't see now errors in logs that wifi can't be initialized but it still doesn't work. It looks like it is active for the short time after restart and then down again. Anything more we could do to test further? 
[meshsupport FRITZ.WLAN Repeater DVB-C 133.07.27_06.08.21_1744.txt.zip](https://github.com/Freetz-NG/freetz-ng/files/6946576/meshsupport.FRITZ.WLAN.Repeater.DVB-C.133.07.27_06.08.21_1744.txt.zip)
[support FRITZ.WLAN Repeater DVB-C 133.07.27_06.08.21_1743.txt.zip](https://github.com/Freetz-NG/freetz-ng/files/6946577/support.FRITZ.WLAN.Repeater.DVB-C.133.07.27_06.08.21_1743.txt.zip)
[meshsupport FRITZ.WLAN Repeater DVB-C 133.07.27_06.08.21_1744.txt.zip](https://github.com/Freetz-NG/freetz-ng/files/6946580/meshsupport.FRITZ.WLAN.Repeater.DVB-C.133.07.27_06.08.21_1744.txt.zip)
[support FRITZ.WLAN Repeater DVB-C 133.07.27_06.08.21_1743.txt.zip](https://github.com/Freetz-NG/freetz-ng/files/6946581/support.FRITZ.WLAN.Repeater.DVB-C.133.07.27_06.08.21_1743.txt.zip)

baribalbear, 2021-08-06:
[FRITZ.WLAN_Repeater_DVB-C_133.07.27-freetz-ng-18484M-4ae9ecb5e_2021-08-06_1734_support.tgz.zip](https://github.com/Freetz-NG/freetz-ng/files/6946583/FRITZ.WLAN_Repeater_DVB-C_133.07.27-freetz-ng-18484M-4ae9ecb5e_2021-08-06_1734_support.tgz.zip)

fda77, 2021-08-07:
In source-files-FRITZ.Box_WLAN_Repeater_DVB_C-07.01.tar.gz there are for 1750 & dvbc dtsi files. Diff:
```
--- Fritz_Box_HW206.dtsi
+++ Fritz_Box_HW205.dtsi
@@ -1,9 +1,10 @@
 / {
-       model = "AVM FritzRepeater 1750E";
+       model = "AVM FritzRepeater DVB-C";

        aliases {
                serial0 = "/ahb/apb/uart@18020000";
                pci0 = &pci_0;
+               ehci_0 = &ehci0;
                eth0 = "/ahb/ethernet@19000000";
                eth1 = "/ahb/ethernet@1a000000";
        };
@@ -31,70 +32,95 @@
                        value = <100>;
                };
                gpio_avm_led_fsi1 {
-                       value = <101>;
+                       value = <104>;
                };
                gpio_avm_led_fsi2 {
-                       value = <102>;
+                       value = <103>;
                };
                gpio_avm_led_fsi3 {
-                       value = <103>;
+                       value = <102>;
                };
                gpio_avm_led_fsi4 {
-                       value = <104>;
-               };
-               gpio_avm_led_lan {
-                       value = <13>;
+                       value = <101>;
                };
-               gpio_avm_led_wlan {
+               gpio_avm_led_tv {
                        value = <105>;
                };
-               shift_register_size {
-                       value = <8>;
-                       param = <AVM_DEF_HW_PARAM_GPIO_IN_ACTIVE_HIGH>;
-                       function = <0x00000000>;
-               };
                gpio_avm_led_power {
                        value = <106>;
-                       default = <1>;
+               };
+               gpio_avm_led_wlan {
+                       value = <107>;
+               };
+               gpio_avm_em28174_reset {
+                       value = <108>;
+               };
+               shift_register_size {
+                       value = <16>;
                };
                gpio_avm_button_wlan {
                        value = <4>;
-                       param = <AVM_DEF_HW_PARAM_GPIO_IN_ACTIVE_HIGH>;
+                       param = <AVM_DEF_HW_PARAM_GPIO_IN_ACTIVE_LOW>;
                };
                gpio_avm_uart_rx {
                        value = <9>;
-                       param = <AVM_DEF_HW_PARAM_GPIO_IN_ACTIVE_HIGH>;
-                       function = <0x00000008>;
+                       //function = <0x00000008>;
[... 91 lines cut ...]
                                      <0x180c0000 0x1000>,
                                      <0x10000000 0x02000000>,
-                                     <0 1>;
+                                       <0 1>;
                                reg-names = "cfg_base","ctrl_base","crp_base",
                                                "mem_base","io_base";
                                interrupt-parent = <&ip2_int>;
                                interrupts = <0>;
                                status = "okay";
                        };
+
+               i2c0: i2c@18018000 {
+                       status = "okay";
                };
+       };

                spi_0: spi@1f000000 {
                        status = "okay";
```

So it seems you have to replace kernel (or extend/replace the device tree)! I dont know if RK works at all, better you test it first without "alien"
If it works, you could select alien+rk and unpack the kernel, then patch the 1750-dtsi or replace it with the DVBC-Version.
Luckily the 7.27 source still contain dtsi for both devices, but you have to check first if the file its still okay.
Maybe then DVB could also be used with "em28174" module, which is a hauppauge dual-hd tuner
For that all a serial console is very usefull!!! I attached to my 1200 shortly to make openwrt working. No soldering needed, just some tape

baribalbear, 2021-08-08:
Hi @fda77,

Many thx for the hint. I created the patch from above and try to build now only 1750E w/o alien. Could you please tell what is "RK"? BTW replacing the module dtsi in kernel in the original firmware image I should use fw* tools?

Regards,
Eugene

fda77, 2021-08-08:
Just unpack the REPLACE KERNEL, patch, (eg copy the 1 dts(i) over the other) and then compile, see make help.
Dts is not a module, but something like bios settings of devices

baribalbear, 2021-08-08:
Ok, I did the following:
"git pull"
"make clean" 
"make menuconfig" (ticked 1750E, 7.25+, international, didn't tick Alien options at all)
applied dtsi patch (see attached)
make
[dtsi_patch.txt](https://github.com/Freetz-NG/freetz-ng/files/6950499/dtsi_patch.txt)

After flashing WLAN doesn't work, factory reset didn't help either, got the log message "wlan-modul konnte nicht korrekt initialisiert werden. bitte fritz!repeater neu starten #102".

Then I applied the alien patch (see attached).
[alien.patch.txt](https://github.com/Freetz-NG/freetz-ng/files/6950502/alien.patch.txt)
"make menuconfig" (ticked 1750E, 7.25+, international, ticked Alien options)
make

After flashing WLAN doesn't work, factory reset didn't help either, however I don't see messages like "wlan-modul konnte nicht korrekt initialisiert werden. bitte fritz!repeater neu starten #102".

Anything I did wrong?

fda77, 2021-08-08:
Ich antworte übrigens meist in der Sprache in der gefragt wurde...
Wie gesagt, oben das Diff ist von 07.0x, wenn du 7.25 gleich testen willst, vergleich die dts aus den Sourcen und schau ob die vom Repeater angepasst werden müssen. Daraus ein diff und irgendwo unter make/linux/patches/ kopieren. Nach make kernel -clean + -unpacked sollte die Datei dann erscheinen. Oder einfach wie vergeschlagen zuerst 7.0x testen! Ich glaub das war noch eine ganz andere Kernelversion, 2.6x vielleicht
Aus dem alien-patch ist doch alles überflüssig, ausser 100-1759_1750.sh.
Ansonsten auf Fehler achten, in serialler Console, Syslog, dmesg usw. lsusb etc könnten auch helfen, geladene Module usw

fda77, 2021-08-08:
Vielleicht reicht es auch aus die HWrev in der .kernelvariables im Hautpverzeichnis des Kernels zu ändern

baribalbear, 2021-08-08:
@fda77 

Unfortunately I can't communicate in German, even English is not my native language but at least I can communicate in English. 

I changed ADD_FILE_PID = Fritz_Box_HW205 inside the 1750E sources .kernelvariables. "make clean" and rebuilt the firmware - no luck still. (

Here https://www.ip-phone-forum.de/threads/freetz-f%C3%BCr-avm-dvb-c-repeater-verf%C3%BCgbar.307070/page-2 there was a proposal this issue is tied to avm_kernel_config which is used by 1750E and should be extracted from the DVB-C kernel due to avm_kernel_config is not part of sources provided by AVM. I am not that strong in building/extracting/packaging kernels etc. so understand all this is not so easy for me. Still trying to understand how to do it. Also I couldn't find original image and sources for 1750E 7.01 so far to try your another advice. (

fda77, 2021-08-08:
Okay :)
I not sure if Replace Kernel works on these devices at all due to AKC, so test first with non alien, den with alien 7.0x (as i wrote above). Copying avm_kernel_config could also help and not change the kernel. Ask @PeterPawn here or at IPPF, he did initial support as avm forgot a tool to create ^^
Its really a silly decision of avm to delete all older images+sources

baribalbear, 2021-08-10:
@fda77,

Could you please tell how to build 1750E 7.27 kernel using the extracted avm_kernel_config_area.S from the DVB-C? It looks like FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_0=y for 1750E and the whole avm_kernel_config replacement part is skipped? Can I add it manually to ./source/kernel/ref-qca955x-1750_07.27/linux-4.4/arch/mips/kernel/? Will it be used during make kernel-precompiled phase?

fda77, 2021-08-10:
Length of 0 bytes is for "failure" of the config-area tool. Initial developer by PeterPawn, extended by er13 and now noone feels responsible to fix it for newer kernel... you have to do it yourself!
You could change the size in the config it in config/.img/*, but the tool still cant handle this kernel

To only make changes on the kernel, do it in build/original/, this dir is just copied to modified

PeterPawn, 2021-08-10:
> but the tool still cant handle this kernel

Maybe this is true for the implementation by @er13 - but I've updated my project yesterday (https://github.com/PeterPawn/YourFritz/commit/fbd0495bf4f55af84966f9166852743cbc9f0f79) and as shown [here](https://www.ip-phone-forum.de/threads/freetz-f%C3%BCr-avm-dvb-c-repeater-verf%C3%BCgbar.307070/page-3#post-2440340), I was able to extract the `avm_kernel_config` area from AVM's kernel for 1750E repeater's firmware version 07.27.

fda77, 2021-08-10:
To generate FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_* in config/.img/ *  `avm_kernel_config.extract` is used. If i remember correctly only er13's version has this. The version in freetz fails for
```
grep FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_0 config/.img/* -l
config/.img/0450--07_0X.in
config/.img/0450--07_1X.in
config/.img/0540--07_0X.in
config/.img/0540--07_1X.in
config/.img/0546--07_0X.in
config/.img/0546--07_1X.in
config/.img/1240--07_0X.in
config/.img/1240--07_1X.in
config/.img/1750--07_0X.in
config/.img/1750--07_1X.in
config/.img/1750--07_25.in
config/.img/1750--07_2X.in
config/.img/1759--07_0X.in
config/.img/1759--07_1X.in
config/.img/2400.in
config/.img/4020--07_0X.in
config/.img/4060.in
config/.img/5530.in
config/.img/6000.in
config/.img/6591.in
config/.img/6660.in
config/.img/6820_V1--07_0X.in
config/.img/6820_V1--07_1X.in
config/.img/6820_V1--07_25.in
config/.img/6820_V2.in
config/.img/6820_V3.in
config/.img/6890--07_25.in
config/.img/6890--07_2X.in
config/.img/7539.in
config/.img/7560--07_25.in
config/.img/7580--DE--07_25.in
config/.img/7580--DE--07_2X.in
config/.img/7583--07_25.in
config/.img/7583--07_2X.in
config/.img/7584.in
config/.img/7590--07_25.in
config/.img/7590--07_2X.in
config/.img/7599.in
```

The change could be "since kernel version 4" - maybe the tool could be added a 2nd time...
But then the size itself is still missing

```
grep KERNEL config/.img/7590*
config/.img/7590--07_25.in:config FREETZ_AVM_PROP_KERNEL_4_9_198
config/.img/7590--07_25.in:config FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_0
config/.img/7590--07_2X.in:config FREETZ_AVM_PROP_KERNEL_4_9_198
config/.img/7590--07_2X.in:config FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_0
config/.img/7590--DE--06_8X.in:config FREETZ_AVM_PROP_KERNEL_3_10_12
config/.img/7590--DE--06_8X.in:config FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_96
config/.img/7590--DE--06_9X.in:config FREETZ_AVM_PROP_KERNEL_3_10_12
config/.img/7590--DE--06_9X.in:config FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_160
config/.img/7590--DE--07_0X.in:config FREETZ_AVM_PROP_KERNEL_3_10_104
config/.img/7590--DE--07_0X.in:config FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_160
config/.img/7590--DE--07_1X.in:config FREETZ_AVM_PROP_KERNEL_3_10_104
config/.img/7590--DE--07_1X.in:config FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_160
config/.img/7590--EN--06_8X.in:config FREETZ_AVM_PROP_KERNEL_3_10_12
config/.img/7590--EN--06_8X.in:config FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_96
config/.img/7590--EN--07_0X.in:config FREETZ_AVM_PROP_KERNEL_3_10_104
config/.img/7590--EN--07_0X.in:config FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_160
config/.img/7590--EN--07_1X.in:config FREETZ_AVM_PROP_KERNEL_3_10_104
config/.img/7590--EN--07_1X.in:config FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_160
```
```
grep KERNEL config/.img/1750*
config/.img/1750--06_2X.in:config FREETZ_AVM_PROP_KERNEL_2_6_32_61
config/.img/1750--06_5X.in:config FREETZ_AVM_PROP_KERNEL_2_6_32_61
config/.img/1750--07_0X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/1750--07_0X.in:config FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_0
config/.img/1750--07_1X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/1750--07_1X.in:config FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_0
config/.img/1750--07_25.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/1750--07_25.in:config FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_0
config/.img/1750--07_2X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/1750--07_2X.in:config FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_0
```
absent of FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_* = no config-area at all (=kernel v2?)

PeterPawn, 2021-08-11:
Jetzt mache ich mal in Deutsch weiter, damit es nicht ermeut Verständnisprobleme beim Lesen gibt.

Für das Extrahieren des Konfigurationsbereichs eines Kernels muß man dessen Größe gar nicht kennen ... hier reicht es tatsächlich, den größten bekannten Wert zu verwenden, denn beim Generieren der Assembler-Dateien werden ja ohnehin nur die vorhandenen Daten ausgegeben - was da noch an Müll dahinter kommt, interessiert gar nicht. Ein "quick hack", wie man die Größe dennoch aus dem (passenden) Loader-Skript für den Kernel (`vmlinux.lds.S`) auslesen KÖNNTE,, wäre das hier: https://github.com/PeterPawn/YourFritz/commit/2a18581a481f45fa0166334094fea4d349fb5de9 (zwei Stunden älter als Dein Kommentar oben) ... aber man braucht es - wie geschrieben - gar nicht.

Jedoch gibt es noch andere, markante Unterschiede zwischen den verschiedenen Versionen, die von AVM benutzt werden - und deren Behandlung regelt @er13 vermutlich anders (ich habe mir das seit der "Übernahme" nie wieder richtig angesehen), als ich das tat (und vermutlich auch anders, als ich es weiterhin tun würde).

Mein Ziel war es nämlich, so weit wie möglich eine 1:1-Kopie des AVM-Bereichs zu erstellen ... das ist nun mal die Krux bei der **ersten** Implementierung, solange man noch nicht weiß, ob es überhaupt funktioniert. Da ist es nur natürlich, wenn man so dicht wie möglich an das Original will - nur so kann man bei einem Mißerfolg (einigermaßen) ausschließen, daß es an solchen Unterschieden gescheitert ist ... ansonsten müßte man das gesamte Prinzip beim Mißlingen in Frage stellen.

Die Unterschiede bestehen bei AVM einerseits in den Werten für die Anzeige des letzten Eintrags in dem Array mit den Pointern auf die Strukturen - hier wird von AVM ein `enum`-Typ verwendet (in `drivers/char/avm_new/include/uapi/avm/enh/fw_info.h`):

```
  9 // Ordering in these tags is important and can not be changed
 10 // as it is dependent upon in other persistent paths of the fw
 11 enum _avm_kernel_config_tags {
 12         avm_kernel_config_tags_undef,
 13         avm_kernel_config_tags_module_memory,
 14         avm_kernel_config_tags_version_info,
 15         avm_kernel_config_tags_hw_config,
 16         avm_kernel_config_tags_cache_config,
 17
 18         /* subrev müssen aufeinander folgen */
 19         avm_kernel_config_tags_device_tree_subrev_0,
 20         avm_kernel_config_tags_device_tree_subrev_1,
 21         avm_kernel_config_tags_device_tree_subrev_2,
[...]
273         avm_kernel_config_tags_device_tree_subrev_254,
274         avm_kernel_config_tags_device_tree_subrev_255,
275         avm_kernel_config_tags_device_tree_subrev_last =
276                 avm_kernel_config_tags_device_tree_subrev_255,
277
278         avm_kernel_config_tags_avmnet,
279         avm_kernel_config_tags_urlader_env,
280         avm_kernel_config_tags_last
281 };
282
283 #define avm_subrev_max                                    \
284         (avm_kernel_config_tags_device_tree_subrev_last - \
285          avm_kernel_config_tags_device_tree_subrev_0 + 1)
286
287 #endif
```
und die Anzahl der Einträge für diesen (und damit auch der Wert für den letzten) hängt von der möglichen Anzahl von `HWSubrevision`s ab, für die AVM hier Werte reserviert. Das paßt zwar (m.W. immer) zu den jeweiligen Kernel-Quellen für ein Modell, aber es ist natürlich gleichzeitig ein Punkt, der die Verwendung eines "gemeinsamen" (sprich: vorbereiteten) "host tools" für mehr als ein Modell verbietet ... außer man verwendet gar nicht mehr die AVM-Strukturen (aus der `fw_info.h`), was dann andere Konsequenzen hat. 

Zum Beispiel die Notwendigkeit erneuter Änderungen, wenn AVM an diesen Strukturen weitere Änderungen vornimmt (daß das passiert, dazu komme ich gleich noch), auch wenn diese Änderungen ansonsten gar keine Auswirkungen auf die Verarbeitung durch das Programm hätten ... daß es Änderungen gibt, die auch bei Nutzung der Daten aus der `fw_info.h` Änderungen am Programm erfordern, ist dadurch natürlich nicht ausgeschlossen.

Aber das Problem könnte man für die Tags wohl tatsächlich noch einigermaßen zuverlässig lösen, denn alle Einträge außer dem letzten (mit Tag `avm_kernel_config_tags_last`) enthalten ja auch einen Pointer auf irgendwelche anderen Daten und dessen Fehlen (bzw. dessen Wert NULL) kann den letzten Eintrag auch identifizieren.

Aber genau so eine Änderung, die Auswirkungen auf den bisher verwendeten Code hat, wurde durch AVM an einer weiteren Stelle vorgenommen ... so sah früher bei der 06.83 für die 7490 die o.a. Datenstruktur noch so aus:

```
  4 enum _avm_kernel_config_tags {
  5         avm_kernel_config_tags_undef,
  6         avm_kernel_config_tags_modulememory,
  7         avm_kernel_config_tags_version_info,
  8         avm_kernel_config_tags_hw_config,
  9         avm_kernel_config_tags_cache_config,
 10         avm_kernel_config_tags_device_tree_subrev_0,  /* subrev müssen aufeinander folgen */
 11         avm_kernel_config_tags_device_tree_subrev_1,
 12         avm_kernel_config_tags_device_tree_subrev_2,
 13         avm_kernel_config_tags_device_tree_subrev_3,
 14         avm_kernel_config_tags_device_tree_subrev_4,
 15         avm_kernel_config_tags_device_tree_subrev_5,
 16         avm_kernel_config_tags_device_tree_subrev_6,
 17         avm_kernel_config_tags_device_tree_subrev_7,
 18         avm_kernel_config_tags_device_tree_subrev_8,
 19         avm_kernel_config_tags_device_tree_subrev_9,  /* subrev müssen aufeinander folgen */
[... cut ...]  machen, wie ich meine Implementierung so an die unterschiedlichen Gegebenheiten bei AVM anpassen kann, daß man mit einem Quelltext alle Versionen bearbeiten kann. 

Das war ja auch mal mein ursprüngliches Ziel - nicht umsonst gibt es bei meiner Implementierung eine Funktion, die erst einmal versucht, die Reihenfolge bei der Speicherung (BE vs. LE) in den vorliegenden DATEN zu ermitteln (und damit auch erst zur Laufzeit), anstatt sich nur fix auf die Byte-Order des jeweiligen Modells zu stürzen.

fda77, 2021-08-11:
Auf deutsch ist auf jeden Fall einfacher!

x64 Support brauche ich nicht unbedingt, ohne funktioniert es halt nicht auf zb aarch64:

```
 grep HOST_CFLAGS_FORCE_32BIT_CODE tools/make -r
tools/make/Makefile.in:HOST_CFLAGS_FORCE_32BIT_CODE:=$(if $(filter-out 32,$(HOST_BITNESS)),-m32)
tools/make/yourfritz-akc-host/yourfritz-akc-host.mk:            BITNESS="$(HOST_CFLAGS_FORCE_32BIT_CODE)" \
tools/make/pseudo-host/pseudo-host.mk:          --cflags="-Wno-cast-function-type -Wno-nonnull-compare -fcommon $(if $(BIARCH_BUILD_SYSTEM),$(HOST_CFLAGS_FORCE_32BIT_CODE))" \
tools/make/pseudo-host/pseudo-host.mk:          CFLAGS="$(TOOLS_CFLAGS) $(HOST_CFLAGS_FORCE_32BIT_CODE)" \
tools/make/fakeroot-host/fakeroot-host.mk:              CFLAGS="$(TOOLS_CFLAGS) $(HOST_CFLAGS_FORCE_32BIT_CODE)" \
tools/make/yourfritz-akc4-host/yourfritz-akc4-host.mk:          BITNESS="$(HOST_CFLAGS_FORCE_32BIT_CODE)" \
tools/make/dtc-host/dtc-host.mk:                BITNESS="$(HOST_CFLAGS_FORCE_32BIT_CODE)"
```

Es sieht so aus als ob es nur Kernel v4 betrifft:
```
grep PROP_KERNEL_[2345] `grep FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_0 config/.img/* -l`
config/.img/0450--07_0X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/0450--07_1X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/0540--07_0X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/0540--07_1X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/0546--07_0X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/0546--07_1X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/1240--07_0X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/1240--07_1X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/1750--07_0X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/1750--07_1X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/1750--07_25.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/1750--07_2X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/1759--07_0X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/1759--07_1X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/2400.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/4020--07_0X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/4060.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/6000.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/6820_V1--07_0X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/6820_V1--07_1X.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/6820_V1--07_25.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/6820_V2.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/6820_V3.in:config FREETZ_AVM_PROP_KERNEL_4_4_60
config/.img/6890--07_25.in:config FREETZ_AVM_PROP_KERNEL_4_9_198
config/.img/6890--07_2X.in:config FREETZ_AVM_PROP_KERNEL_4_9_198
config/.img/7539.in:config FREETZ_AVM_PROP_KERNEL_4_1_52
config/.img/7560--07_25.in:config FREETZ_AVM_PROP_KERNEL_4_9_198
config/.img/7580--DE--07_25.in:config FREETZ_AVM_PROP_KERNEL_4_9_198
config/.img/7580--DE--07_2X.in:config FREETZ_AVM_PROP_KERNEL_4_9_198
config/.img/7583--07_25.in:config FREETZ_AVM_PROP_KERNEL_4_9_198
config/.img/7583--07_2X.in:config FREETZ_AVM_PROP_KERNEL_4_9_198
config/.img/7584.in:config FREETZ_AVM_PROP_KERNEL_4_9_198
config/.img/7590--07_25.in:config FREETZ_AVM_PROP_KERNEL_4_9_198
config/.img/7590--07_2X.in:config FREETZ_AVM_PROP_KERNEL_4_9_198
config/.img/7599.in:config FREETZ_AVM_PROP_KERNEL_4_9_198
```
Die Liste ist übrigens nicht vollständig, zb 6660 hat 4.9 .175/.250/.199 aber FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_0 ist in der 6660.in!

Somit könnte man das ganze AKC für kernel 3 ignorieren und so lassen wie es ist - es funktioniert ja. Man muss nichts reparieren was nicht kaputt ist. Es gibt höchstens neue Fehler. Wie genau das mit den verschiedenen 7490 Versionen gelöst ist weiss ich nicht, ich sehe das als Blackbox an, solang herauskommt was ich erwarte ist es okay.
Und für Kernel v4 was ganz neues implementieren. Bei der Frage wie genau will ich gar nicht mitreden da ich keine Ahnung davon hab... Mir ist also egal ob host-tool oder wie auch immer. Da aktuell `FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_0` RK nicht auswählen lässt, kann man auf keinen Fall was kaputt machen.

> und da man das Tool auch gar nicht benötigt, solange man kein "Replace kernel" machen kan

oder halt einen Alien der es erfordert, falls dieser ohne replace kernel funktionieren soll. Das ist aber nicht unbedingt nötig und man kann einfach automatisch RK auswählen

> nicht umsonst gibt es bei meiner Implementierung eine Funktion, die erst einmal versucht, die Reihenfolge bei der Speicherung (BE vs. LE) in den vorliegenden DATEN zu ermitteln

In Freetz gibt es die Symbole `FREETZ_AVM_PROP_ARCH_BE` und `FREETZ_TARGET_ARCH_BE` mit denen man den Test evlt übergehen könnte

Ich hab übrigens kein AVM-Gerät mehr das mit Kernel v4 läuft und kann in dem Bereich nichts mehr testen. Die 7490 als ISDN Bridge hat noch v3

PeterPawn, 2021-08-12:
Vielleicht ist es ohnehin besser, auf "Replace kernel" zu verzichten - man weiß ja auch nie, was bei AVM in den Quellen noch alles fehlt. Außerdem ist das "Einarbeiten" der AVM-Quellen (gerade auch mit dem Freetz-Ansatz, das mit einer "Stammversion" und entsprechenden Patches für die Abweichungen bei anderen Modellen zu regeln) ja auch eine ziemliche Fleißarbeit und oft erst lange nach dem Release neuer Versionen machbar, wenn AVM die Quellen dann nachgereicht hat - da ist oft schon das nächste Release im Anflug.

Es gibt nur sehr wenige Kernel-Options, die tatsächlich einen Rebuild erfordern (NFS wäre m.W. so eine, weil das an allen möglichen Stellen dann seine Hooks einbauen muß) und wenn man nur zusätzliche LKM braucht, muß man den Kernel ja noch lange nicht ersetzen. Die anderen Optionen bzw. Pakete, die einen tieferen Eingriff in den Kernel benötigen, funktionieren dann halt nicht für die neuen Modelle - der "Protest" wird sich in Grenzen halten.

Entscheidender sind sicherlich andere Pakete, die auch mit dem originalen Kernel funktionieren ... ich bin nicht mal sicher, ob der neueste 7490-Kernel noch die Patches für die Traps bei Verwendung von TUN/TAP-Devices braucht (https://github.com/PeterPawn/YourFritz/tree/main/patch_kernel) und die GRX-Modelle brauchen den wohl schon länger nicht mehr - aber alles auch ohne jegliches "Wissen" meinerseits, da ich das nicht wirklich verfolge/teste und andere sicherlich auch nicht, solange das irgendwo "integriert" ist. Da es auch kein Problem ist, wenn das Zeug nichts zum Patchen findet, wird das vermutlich bis zum jüngsten Tag so implementiert bleiben.

fda77, 2021-08-12:
Die delta-Patches müssen nicht unbedingt sein, er13 wollte aber immer Platz auf den Mirrors und Bandbreite sparen. Anfang wurden sie von Freetz noch bei AVM direkt heruntergalden, die Pakete waren aber sehr gross und wenn Herr AVM nichts zu tun hat wurden auch schon immer zufällig Datein umbenannt, gelöscht oder unter gleichem Dateinamen ersetzt... das war nichts.
Ich hab letzte Zeit aber fast keine Deltas mehr eingefügt, da es für neues eh keine RK gibt. Allerdings kommen mittlerweile Quellen relativ(tm) flott von avm, ich würde mal max 2 Wochen schätzen. Es gibt aber vermutlich wesentlich mehr Leute die mailen, da es sich langsam herumspricht dass man wirklich jedes mal bei avm herumbetteln muss. Auch bei den Leuten die auf Serielle Treiber stehen (aber leider noch nicht dass es sinnvoll wäre Änderungen an der Kernelconfig zu PRen damit man es beim nächsten mal nicht wieder machen muss)
Für 7490 7.2x wurden die YF Patches benötigt, ich hatte sie deaktiviert und es gab Bechwerden, nach Aktivieren ging Openvpn wieder. Für 7.25 sind die Patches auch noch aktiv, ohne Beschwerden - halt vielleicht überflüssig. Ich glaube aber nicht dass AVM da viel geändet hat.
Die 7590 braucht die Patches seit Kernel v4 nicht mehr https://github.com/Freetz-NG/freetz-ng/blame/master/config/avm/features.in#L137

fda77, 2021-08-13:
@PeterPawn, eine Antwort auf https://github.com/Freetz-NG/freetz-ng/issues/335#issuecomment-898672975
"Einfach" die config-area aus dem vorhandenen Kernel extrahieren sollte am wenigsten Aufwand sein. Bei Freetz wird daran ja nie etwas geändert. Damit könnten wohl die Alien als auch replace-kernel mit kernel v4 funktionieren! Beim Auspacken für build/original/ fürde diese dann erstellt. (Dadurch könnte man auch manuell an der Datei herumexperimentieren)
Bei Alien gibt es schon eine Funktion einer 2. ("tk"/ursprünglich telekom) Firmware aus der man den blob verwenden könnte mitauszupacken.
Ist es für den DVB-c okay den Kernel der 7.0x-release als Qulle für 1750-7.25+ zu verwenden? Die neustes 7.08 ist inhaus oder labor.
Hast du die dd(?)-Befehle zum extrahieren und hinzufügen parat? Ein Script in yourfritz-host wäre auch eine Möglichkeit

PeterPawn, 2021-08-13:
Laß uns (also "meinen Tester" und mich) mal erst (im IPPF) feststellen, ob das überhaupt (wie erwartet) funktioniert mit der 1750E-Firmware auf dem DVB-C-Repeater - und das dann auch mit vollem WLAN, worum es sich hier ja dreht.

Wenn das klappen sollte (ich bin auch nicht 100% sicher, das derzeitige Problem mit der Initialisierung des LAN-Ports verstehe ich noch gar nicht), dann kann man hinterher noch einmal in Ruhe die notwendigen Änderungen ansehen ... und wenn sich dabei auch noch die RK-Option für andere, ähnliche Modelle (das mit dem Konfigurationsbereich zieht AVM bei allen Architekturen, egal ob ARM oder MIPS - bei den Pumas bin ich nicht sicher und müßte selbst erst nachsehen - konsequent durch) ergeben sollte, dann soll mir das auch recht sein.

fda77, 2021-08-14:
Ich denke die tk firmware könnte euch beim Testen helfen und mache Fehlerquelle ausschliessen, zb (ent)packen. Die Freetz-Sachen sind beim Starten des Kernels ja noch egal.
Script für dvbc Alien: https://github.com/Freetz-NG/freetz-ng/blob/master/patches/cond/100-aliens/scripts/100-1759_1750.sh 
Variablen:
```
TK_DIR="${DIR}/.tk/original"
FIRMWARE_TK_DIR="${TK_DIR}/${FIRMWARE_SUBDIR}"
FILESYSTEM_TK_DIR="${TK_DIR}/${FILESYSTEM_SUBDIR}"
KERNEL_TK_DIR="${TK_DIR}/${KERNEL_SUBDIR}"
VARTAR_TK_DIR="${KERNEL_TK_DIR}/${VARTAR_SUBDIR}"
```

PeterPawn, 2021-08-14:
Verstehe mich bitte nicht falsch ... aber ich wüßte gar nicht, was ich mit dem Skript anfangen soll und die Kommandos zum Entpacken und Packen, sowie zum Zerlegen eines Images, habe ich (meist) im Kopf bzw. verifiziere ich üblicherweise erst noch einmal (bei einem neuen Modell), indem ich sie Schritt für Schritt ausführe (wie so etwas aussieht, kannst Du Dir im parallelen IPPF-Thread ansehen - wobei dort natürlich nur die funktionierenden Kommandos und ihre Ergebnisse gezeigt werden und nicht die "Irrtümer", denen man zwischendrin zwangsläufig auch immer wieder unterliegt) und dabei jedes Zwischenergebnis noch einmal prüfe.

Außerdem macht Freetz (und seine Forks) vieles aus meiner Sicht unnötig kompliziert ... ein Beispiel wäre hier die Suche nach der Stelle, wo man bei einem kombinierten Image ansetzen muß, um das in den Kernel und das Dateisystem (und ggf. auch in die NMI-Vector-Tabelle) zu zerlegen. Wenn ich mich nicht irre, verwendet Freetz an der Stelle immer noch das "Schweizer Taschenmesser" - was aber auch erst mal installiert (oder kompiliert) werden müßte. Das geht - zumindest in den Fällen, die ich benötige - aber mit "Standard-Tools" (u.U. sogar mit der Beschränkung auf solche, die zum POSIX-Standard gehören) mindestens genauso gut ... zumindest bin ich bisher noch auf keine unüberwindlichen Probleme gestoßen. Anderes wieder, wie das "Einbauen" eines NMI-Vectors wieder an der richtigen Stelle im Firmware-Image, wenn der bei der AVM-Firmware vorhanden war/ist, ist m.W. in Freetz (und Forks) gar nicht realisiert - "es geht ja auch ohne" ist die Begründung.

Aber wie ich schon an anderer Stelle schrieb, mag es zwar beim Nachbauen von Lösungen denkbar sein, wenn man versucht, soviele "Sonderfälle" wie möglich unter ein gemeinsames Dach zu quetschen (z.B. beim Packen von Firmware-Images für die verschiedensten Modelle) oder sich auf ein "funktioniert doch" zurück zu ziehen, aber das ist etwas, was man bei "Erstbesteigungen" tunlichst vermeiden sollte - denn jede zusätzliche Abweichung (z.B. weil irgendein unerwarteter "Pfad" in so einem "Super-Skript" genommen wird) kann zu einem Mißerfolg führen und wenn man sich dann dieser Effekte nicht bewußt ist, marschiert man in die falsche Richtung weiter - will heißen: man zieht die falschen Schlußfolgerungen und muß schon sehr viel Glück haben, wenn man nicht noch viel mehr Zeit verschwendet hat, bevor man wieder ein paar Schritte zurück macht und dann im zweiten (oder dritten, etc.) Anlauf doch noch über die Stelle stolpert, wo es zuvor eben NICHT wie erwartet lief, ohne daß man es bemerkte.

Es verbietet sich also bei solchen "Erkundungen im Neuland" von selbst, auf "Altbewährtes" zurückzugreifen, solange das nicht tatsächlich nur etwas "sehr Kleines" mit einer einzigen Aufgabe ist (wo dann auch nichts "unbemerkt" schief gehen kann) - womit wir wieder bei der Theorie "One tool for one job." wären ... auf diese These berufe ich mich ja nicht ohne Grund (und entsprechende Erfahrungen) immer wieder aufs Neue und dazu gehört auch das Bestreben, eine Aufgabe so weit es möglich ist, mit Standard-Tools zu lösen und auf die Verwendung von "Exoten" zu verzichten - auch wenn das nicht in jedem Fall funktionieren wird, manchmal auch nur aus Performance-Gründen - wie z.B. hier: https://github.com/PeterPawn/YourFritz/blob/main/scriptlib/functions/yf_base64.function, denn das dauert für größere Datenmengen mal so richtig.

---

Um noch mal auf das Skript oben zurück zu kommen, wenn ich mir seinen Inhalt ansehe - das mag zwar alles irgendwann(!) mal "nett" sein als Änderung, aber ich sehe für nicht eine davon (bisher) eine Notwendigkeit, weil es - sofern tatsächlich der Kernel gepatcht werden muß - wohl nur dann Sinn macht (wenn überhaupt), das Image wieder "komplett" zusammen zu packen, wenn man das über Freetz installieren wollte. Irgendwelche "Cross-Updates" (wo von einer DVB-C-Version auf einen Alien (basierend auf der 1750E) gewechselt wird oder umgekehrt) werden ohnehin nicht funktionieren - nur in so einem Fall macht es aber überhaupt Sinn, sich mit `CONFIG_INSTALL_TYPE` zu befassen, da der "in der Firmware" steht (und nicht im Gerät) und solange es sich um eine Firmware für dasselbe Gerät handelt, ist der Wert ohnehin schon derselbe.

Die automatische Suche nach anderen Dateien bei AVM (u.a. nach Online-Updates) macht auch fast keinen Sinn (sei es wg. notwendiger Änderungen am Kernel oder am Dateisystem) und wenn ich eine 1750E-Firmware als Alien auf einem DVB-C-Repeater betreibe, wird auch die Oberfläche weiterhin wie die eines 1750E aussehen ... denn die Dateien zur Unterstützung der anderen Features (von USB bis DVB-C) sind in der Alien-Firmware gar nicht vorhanden. Also entfällt auch `CONFIG_LINEAR_TV` (die Zeile ist ja schon nur noch ein Kommentar, macht aber ohnehin keinen Sinn) und auch die Änderung von `CONFIG_VERSION_MAJOR` dürfte (meine Ansicht) mehr zerstören, als hilfreich zu sein. 

Denn mit diesem Wert werden eben auch einige URLs gebildet und wenn die dann bei AVM auf eine Version zeigen, die es eigentlich gar nicht gibt, wird u.a. die Online-Hilfe (ohne jede Notwendigkeit) nicht funktionieren. Wenn da eine Firmware eines 1750E auf dem DVB-C-Repeater läuft, dann IST das auch eine 1750E-Firmware - egal, was auf dem Gerät selbst aufgedruckt ist. Irgendwelche Mimikry-Versuche sind da eher kontraproduktiv - zumindest in diesem Fall (und nach meiner Erfahrung auch in vielen anderen).

[... cut ...] t - von solchen Bestrebungen wäre mir zumindest nichts bekannt und wenn es sie doch geben sollte, würde ich immer noch für jede einzelne Änderung nach dem Grund (der ja die Auswirkungen berücksichtigen sollte) fragen.

Als Fazit würde ich also praktisch ALLE Änderungen aus dem oben verlinkten Skript für vollkommen unnötig (ja, ggf. sogar für kontraproduktiv) halten in diesem speziellen Kontext (1750E-Firmware als Alien auf einem DVB-C-Repeater) - und nur dafür ist das Skript ja offenbar gemacht.

fda77, 2021-08-14:
Ihr müsst es ja nicht benutzen wenn es dadurch nicht einfacher wird. Dateianhänge im IPPF sind übrigens problematisch da diese hinter einer Login-wall sind.
Wie das genau mir den Alien-Scripten ist weiss ich gar nicht mehr, zuletzt hatte ich Alien selbst genutzt um auf eine 7320 6.5x zu bekommen. Die hat aber mittlerweile Openwrt und damit kernel 5.4, wpa3 usw
[image]

Ich hatte selbst ca 2015 einen dvbc und 1750, und für mich sahen beide gleich aus, ausser dass der dvbc noch einen internen usb und daran einen em28174 dual-tuner hat. Das war da aber nicht im svn-Repo Mit neuerem Kernel v4 könnte sogar video4linux (oder wie das genau hiess) funktionieren. 
[image]
[image]

Eigentlich überrascht es mich dass Wlan beim Alien nicht mehr funktioniert. Bei Betrachtung von https://github.com/Freetz-NG/freetz-ng/issues/15#issuecomment-894713744 könnte dies problematisch sein, der Rest sind wohl nur die Button+Led
```
                gpio_avm_extphy1_reset {
                        value = <11>;
-                       param = <AVM_DEF_HW_PARAM_GPIO_OUT_ACTIVE_HIGH>;
+                       param = <AVM_DEF_HW_PARAM_GPIO_OUT_ACTIVE_LOW>;
```

PeterPawn, 2021-08-14:
> sfk ist übrigens Teil der Hosttool

Es ist mir ja auch egal, was Freetz ((und Forks) da nutzt - ich wollte nur ein Beispiel bringen, daß nicht alles aus Freetz so ist, wie ich es (häufig) brauche (oder gebraucht habe).

Ich weiß nicht, warum man hier überhaupt auf die Idee kam (und nicht nur hier, sondern auch im IPPF-Thread - ich bin ja dort eigentlich eingestiegen und das schon vor Monaten, als jemand durch Änderungen an der Hardware-Beschreibung sein Problem lösen wollte - mit einem (noch deutchen) Kommentar, daß geänderte DTS-Files nichts helfen.

Auch hier bin ich eher darauf angesprungen, wie man solche FDT-Änderungen erreichen kann - ohne zunächst noch einmal selbst zu hinterfragen, ob solche Änderungen überhaupt zielführend oder notwendig wären. Schaut man sich aber die FDT-Dateien in der 07.27 für den 1750E an, sind das für den DVB-C-Repeater und den 1750E andere Strukturen - das gesamte `hui`-Zeugs ist wohl erst später (ich würde vermuten, um das bei den neueren Repeater-Modellen weiter zu vereinheitlichen und zu vereinfachen) hinzugekommen.

Und schaut man dann noch genauer in die Boot-Protokolle einer 07.27 auf dem DVB-C-Repeater, stellt sich heraus, daß es wohl gar nicht an der Hardware-Initialisierung scheitert (egal, welches fehlende PCI-Gerät da vermeintlich vermeldet wird), sondern irgendwann später. Und da wohl bei der Konfiguration der WLAN-Komponenten (durch `/sbin/wlanconfig`, wenn ich mich richtig erinnere) jetzt die Kombination "HWRevision 205 + HWSubrevision 2" verwendet wird (die es beim originalen 1750E gar nicht gäbe), vermute ich mal, daß das Problem eher dort irgendwo zu finden sein könnte. 

Aber ich warte jetzt auf passende (und aktuelle) Protokolle für diese Konstellation - und (um ganz sicher zu gehen) das auch mit einer "selbst erstellten" Firmware, die wieder die NMI-Vector-Tabelle enthält. Das mag überflüssig sein - aber wie schon erwähnt: Bei solchen Unternehmungen tritt man mit der Maxime an, die Unterschiede so klein wie möglich zu halten und der Einbau der NMI-Vektor-Tabelle sind drei (oder vier, wenn man dessen Extraktion mitzählen wollte) zusätzliche Kommandos.

So ganz weit im Hinterkopf habe ich auch noch die Idee als zusätzliche Fehlerquelle, daß irgendjemand zwar die neue Firmware als Alien über den Bootloader installiert hat, dabei aber total vergessen wurde, einfach mal das Gerät auch noch auf die Werkseinstellungen zu setzen, falls irgendein alter Inhalt in der `wlan.cfg`, der vom DVB-C-Repeater stammt, diese Probleme bereitet. 

Da m.W. bisher von systematischer(!) Suche nach dem Problem nichts zu sehen war, muß man wohl alles noch einmal von Beginn an überprüfen - ich ärgere mich schon genug über die verschwendete Zeit bei den (vermutlich unnötigen) Versuchen, die Hardware-Beschreibung irgendwie zu ändern. So weit, wie das alles schon funktioniert, kann es fast kein Layout-Problem mehr sein - dafür funktioniert viel zu viel mit der Alien-Firmware, wie ich jetzt erst gesehen habe.

fda77, 2021-08-14:
Werkseinstellungen sind ein gute Idee. Wobei ich eher nicht glaube dass es hilft. Am Alien funktioniert alles, bis auf Wlan (und DVB weil es das in der FW nicht gibt). Wenn ich mich richtig erinnere!! funktionierte Wlan bei älteren FOS noch als Alien.
Bei solchen Problemen wäre eine serielle Console sehr hilfreich

PhoenixG36, 2021-09-24:
Ich habe einen DVB-C und 1750E Repeater da und die können beide zur Fehlersuche genutzt werden.
Aktuell habe ich eine Alien Firmware auf den DVB-C gespielt und natürlich kein Wlan, auch ein zurücksetzten bringt nichts.
Was mir aber passiert ist, dass ich zuerst die Alien Firmware auf den ganz normalen 1750E gespielt habe und diese hat dort funktioniert. Dieser meldet dann einen DVB-C Repeater mit 7.27 und strahlt das Standard WLAN mit dem Namen XX DVB-C aus.

Ich kann auch eine Serielle Konsole öffnen und die Ausgaben zur Verfügung stellen. Leider habe ich es auf Anhieb nicht hinbekommen etwas vollständig lesbares zu bekommen. Allerdings habe ich auch einen alten TTL Adapter, der nicht so richtig gut läuft und auf Notebook ist schon Windows 11. Aber vielleicht lag es ja auch nur an den COM Settings. Dazu wäre eine Anleitung super, um zu wissen das alles passt. Am besten mit dem dazu getesteten Terminal Tool. Putty und Tera Term waren nicht erfolgreich. Settings 115200-8-N-1

[4 more replies]

closed 2022-05-02 by fda77