Freetz-NG/freetz-ng #20: [remove patch] only a warning at 'remove webdav'

harryboo, 2020-05-24:
There is a warning if the patch 'remove webdav' is selected.

attached a log from the creation of a 7390 image:
[image]

fda77, 2020-05-24:
Before https://github.com/Freetz-NG/freetz-ng/commit/a6b39568401cceecc66526faaef043abe1829815
these "warnings" (not errors) where only visible with verbose 3.
But verbose 3 never really worked for "build", only the "script" part.
For eg "configure" verbose above 2 was ignored, and nothing was displayed, like verbose 0.
So nothing to worry, lower verbose does not show the warnings

fda77, 2020-07-05:
Messages are shown with verbos=2 only, should be okay

closed 2020-07-05 by fda77