Freetz-NG/freetz-ng #23: DVB-C capability on DVB-C Repeater not working

AleXSR700, 2020-06-15:
Hi everyone,
I have been trying to build a Freetz-NG version of my AVM DVB-C Repeater firmware. The build process seems to work and I do get a build. Both for the 133.07.01 and the 133.07.08 Inhaus firmware.

However, even though the only modification was to add OpenVPN, the DVB-C functionality no longer seems to work. No stations are being found.

Is this a known issue and/or is there a solution for this?

Thank you
Alex

fda77, 2020-06-15:
I dont know if this is a problem. And i have no hardware to test.
Are you using a self compiled image or from some forums?
There was another report for DVB-C problems  but this was related to an image of priskat: https://github.com/Freetz-NG/freetz-ng/issues/15

AleXSR700, 2020-06-15:
I am compiling them myself.
I used freetz-linux, cloned your repository, made a menuconfig and then started the make process.

In the menuconfig I left everything at "beginner" level as I just wanted to add OpenVPN. So I selected the stable 133.07.01, added OpenVPN including all subfeatures except LZ4 and then started the make.

I'd be happy to test an image built by someone else, but I haven't run across any pre-built images.

fda77, 2020-06-16:
A self build image is always better, as you know what was included!
I cant help you with DVB. You have to investigate the cause of the problem, chek eg syslog, /tmp/ or /var/logs/, during boot and while using DVB

AleXSR700, 2020-06-16:
By the way, we can communicate in German also, if you'd like.

Anyway, is there something I can look for? I just set up a brand new Ubuntu 20.04 LTS system and did a clean new build. Unfortunately the problem remains the same. 
Is there something I can do to prevent the build to remove anything at all? Kind of just let is build without removing anything and just doing kind of a clean rebuild or something?
I would like to see if there is a fundamental flaw or if some packet is being removed or updated that should not be removed/updated.

fda77, 2020-06-16:
Wäre schon einfacher...
Schau dir halt alle möglichen Logs an, ob irgendwo ein Fehler auftritt
Mit Freetz wird so einiges verändert (was aber eigentlich nichts am DVB behindern sollte.
Kannst auch mal verschieden busybox versionen versuchen. Evtl hat avm auch ein neuen applet aufgenommen, das in freetz fehlt, vergleich die avm und freetz versionen doch mal
Usw

AleXSR700, 2020-06-16:
Wie kann ich denn avm und freetz miteinander vergleichen?

Hier mal mein neuester Build inklusive allem, was irgendwie bei dem Suchbegriff "dvb" im menuconfig auftauchte.
[DVB-c_07.01.all_freetz-ng-16845-5840162d7_20200616-182526.zip](https://github.com/Freetz-NG/freetz-ng/files/4787927/DVB-c_07.01.all_freetz-ng-16845-5840162d7_20200616-182526.zip)

fda77, 2020-06-17:
Die andene dvb tools sind für Sticks usw
Beide busybox aufrufen und die Ausgabe der Applets vergleichen
Logs angeschaut??

AleXSR700, 2020-06-17:
Ich hänge hier mal die Logs nach gestartetem DVB-C Sendersuchlauf an. Ich habe nichts gesehen, das mir hilft. Siehst du etwas?

Das ist jetzt für mich auch etwas neues Gebiet. Meinst du, eine eigene und dann eine AVM Firmware installieren und dort dann die Busybox abfragen?
Also so:
`BusyBox v1.27.2 (2020-06-16 18:08:29 CEST) built-in shell (ash)

root@fritz:/var/mod/root# busybox
BusyBox v1.27.2 (2020-06-16 18:08:29 CEST) multi-call binary.
BusyBox is copyrighted by many authors between 1998-2015.
Licensed under GPLv2. See source distribution for detailed
copyright notices.

Usage: busybox [function [arguments]...]
   or: busybox --list
   or: function [arguments]...

        BusyBox is a multi-call binary that combines many common Unix
        utilities into a single executable.  The shell in this build
        is configured to run built-in utilities without $PATH search.
        You don't need to install a link to busybox for each utility.
        To run external program, use full path (/sbin/ip instead of ip).

Currently defined functions:
        [, [[, addgroup, adduser, arp, arping, ash, awk, base64, basename,
        brctl, bunzip2, bzcat, bzip2, cat, chgrp, chmod, chown, chroot, cmp,
        cp, crond, crontab, cryptpw, cut, date, dd, delgroup, deluser, df,
        dirname, dmesg, dnsdomainname, du, echo, egrep, env, ether-wake, expr,
        false, fgconsole, fgrep, find, flock, free, fstrim, ftpget, ftpput,
        fuser, getopt, grep, groups, gunzip, gzip, halt, head, hostname, httpd,
        id, ifconfig, ifdown, ifup, inetd, init, insmod, iostat, ip, ipaddr,
        iplink, ipneigh, iproute, iprule, iptunnel, kill, killall, killall5,
        klogd, ln, logger, login, logname, logread, ls, lsmod, makedevs,
        md5sum, mkdir, mkfifo, mknod, mkpasswd, mkswap, modprobe, more, mount,
        mpstat, mv, nbd-client, nc, netstat, nice, nohup, nslookup, passwd,
        pidof, ping, ping6, pivot_root, poweroff, printenv, printf, ps, pstree,
        pwd, pwdx, readlink, realpath, reboot, renice, reset, rm, rmdir, rmmod,
        route, sed, seq, setconsole, setserial, sh, sha3sum, sleep, smemcap,
        sort, start-stop-daemon, stat, stty, stun-ip, swapoff, swapon,
        switch_root, sync, sysctl, syslogd, tail, tar, taskset, tee, test,
        tftp, time, top, touch, tr, traceroute, true, tty, umount, uname, uniq,
        unxz, unzip, uptime, uudecode, vconfig, vi, wc, which, whois, xargs,
        xz, xzcat, yes, zcat
root@fritz:/var/mod/root# ERROR: modpatch: Target file build/modified/filesystem
/usr/www/all/menus/menu_data.lua does not exist
-sh: ERROR:: not found
`

Ich habe den menu_data.lua does not exist Fehler auch eben bei einem anderen Neubau gesehen. Woran liegt das denn?

Die AVM Firmware installiere ich dann hoffentlich heute Abend, wenn ich wieder zu Hause bin. Sofern du sagst, dass du genau den Output brauchst.
Wenn ich vorher noch einen Befehl auf dem Eigenbau ausführen soll, dann bitte Bescheid geben.

Danke dir!

[FRITZ.WLAN_Repeater_DVB-C_133.07.01-freetz-ng-16845-5840162d7_2020-06-17_0614_support.tgz.zip](https://github.com/Freetz-NG/freetz-ng/files/4790314/FRITZ.WLAN_Repeater_DVB-C_133.07.01-freetz-ng-16845-5840162d7_2020-06-17_0614_support.tgz.zip)

fda77, 2020-06-17:
Von "sh: ERROR:: not found" müsste man den Kontext wissen, wird halt irgend ein Befehl nicht gefunden
"ERROR: modpatch" Irgendein Patch passt nicht zum Repeater. Welcher es ist steht drüber.
Busybox: Kopier doch die avm-Binary einfach auf das Gerät
Ob das alles mit dvb was zu tun hat ist fraglich, musst selbst suchen

fda77, 2020-07-02:
Evtl ist dir Ursache auch was mit udev/TTYs, wie evtl bei den LTE Modellen. Siehe
 - https://github.com/Freetz-NG/freetz-ng/issues/25
 - https://www.ip-phone-forum.de/threads/307287
Kannst du ja mal auf deinem Gerät anschauen, die Tuner sind am internen USB angebunden

fda77, 2020-07-04:
Bitte mal mit der aktuellsten Revision ausprobieren

closed 2022-05-02 by fda77