Freetz-NG/freetz-ng #24: Fehler bei build dropbear-2020.79 für 7590 Labor 07.19-79495

michael0815000, 2020-06-20:
Hallo,

beim make bekomme ich folgenden Fehler:

dbrandom.c: In function 'process_getrandom':
dbrandom.c:174:8: warning: implicit declaration of function 'getrandom' [-Wimplicit-function-declaration]
  ret = getrandom(buf, sizeof(buf), GRND_NONBLOCK);
        ^
dbrandom.c:174:36: **error: 'GRND_NONBLOCK' undeclared** (first use in this function)
  ret = getrandom(buf, sizeof(buf), GRND_NONBLOCK);
                                    ^
dbrandom.c:174:36: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [Makefile:153: dbrandom.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/freetz/7590/source/target-mips_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.9/dropbear-2020.79'

ERROR: Build failed.
make: *** [make/dropbear/dropbear.mk:77: source/target-mips_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.9/dropbear-2020.79/dropbearmulti] Error 1

Mit dropbear 2019.78 hatte ich das Problem nicht.

Gruß
michael0815000

fda77, 2020-06-21:
War doch schon behoben, als das hier erstellt wurde: https://github.com/Freetz-NG/freetz-ng/commit/9561989854e8fe8605af1921abbdddd84d3923a7 ;-)
Evtl noch ein ```make dropbear-dirclean``` um es anzuwenden

closed 2020-06-21 by fda77

michael0815000, 2020-06-21:
Super, vielen Dank dafür!
Da habe ich gestern `git pull` wohl einen Moment zu früh ausgeführt :-)
Ging heute dann auch ohne `make dropbear-dirclean`.

fda77, 2020-06-26:
Mit https://github.com/Freetz-NG/freetz-ng/commit/93b7bbee73609572c857f1bcd2cdeaa32bcadc9c sollte man https://github.com/Freetz-NG/freetz-ng/commit/9561989854e8fe8605af1921abbdddd84d3923a7 nicht mehr brauchen