Freetz-NG/freetz-ng #25: FB 6890 no LTE

WileC, 2020-07-01:
Same in original Freetz:

[](https://github.com/Freetz/freetz/issues/136)

fda77, 2020-07-01:
Keine Ahnung, musst du suchen woran das liegt. Ich hab keine 6890
Du musst halt die logs anschauen, busybox applets prüfen, Kernelmodule usw...
Da es bereits seit über 1 Jahr eine Issue und die dazu mehr Infos infos enthält gibt, schliesse ich das hier -> https://github.com/Freetz/freetz/issues/136

closed 2020-07-01 by fda77

fda77, 2020-07-02:
@WileC & @PeterPawn
```grep -i tty patches/ -r -l```
Davon kommt nur ```patches/scripts/102-dont-change-tty-settings.sh``` in Betracht, da die anderen conditional sind. Dieser verweist auf ```patches/scripts/102-add_inittab.sh```. 
Freetzmount kann nicht die Ursache sein, da dies für "neuere" FOS ab 6.5 nicht angepasst wurde und deshalb nicht mehr auswählbar ist (stattdessen udevmount)
Die ```make/mod/files/root/etc/device.table``` sollte auch noch angeschaut werden

PeterPawn, 2020-07-02:
Ich sag's gleich deutlich an ... die (gemeinsame) Fehlersuche macht nur dann Sinn, wenn alle an derselben Stelle schreiben - und im Moment wäre das für mich das IPPF: https://www.ip-phone-forum.de/threads/fb-6890-kein-lte-mehr-nach-freetz.307287/

Ggf. kann man das auch in einer Issue im GitHub behandeln ... aber bitte nicht obendrein noch in einer, die bereits "closed" ist, sondern in der hier in comment 2 auch verlinkten.

Das ist ansonsten kein "Miteinander" mehr (und für die Leute ohne Gerät geht es nicht ohne die mit der Hardware, für die mit der Hardware und ohne Schimmer, wo sie in Freetz suchen sollen, nicht ohne diejenigen, die Freetz (etwas) kennen) - am Ende kann niemand die ganzen verstreuten Stellen unter Beobachtung halten, egal wie sehr das verlinkt wird (als "mentioned"). Wer nicht ohnehin schon die Benachrichtigungen in seinem Dashboard aktiviert hat, kriegt davon letztlich gar nichts mit.

fda77, 2020-07-02:
Ich wollte ja verhindern dass hier auch noch was dazu steht indem ich auf die andere Issue verwiesen hab.
Dass es jetzt im IPPF darüber weiter geht ist halt Pech, Ich kenne meine Passwort von meinem alten, scheinbar nicht gelöschten Account nicht mehr und die Mailaddresse hatte ich auf einen Wegwerfmailer geändert

PeterPawn, 2020-07-02:
Aber in der anderen Issue kannst Du ja auch schreiben ... wenn das in Kombination aus GitHub und einem BBS geklärt wird (weil letzteres bessere Möglichkeiten bietet), ist es ja auch in Ordnung - nur nicht noch in zwei verschiedenen Issues, das macht (wie Du selbst in comment 2 festgehalten hast) keinen Sinn.

fda77, 2020-07-04:
Funktioniert nun, von koronth getestet

fda77, 2020-07-13:
@WileC 
Du öffnest dieses Issue, damit du nachdem es hier behoben wurde das über 1 Jahr alte freetz-er13 Issue https://github.com/Freetz/freetz/issues/136 mit https://github.com/Freetz/freetz/pull/303 durch https://github.com/Freetz/freetz/commit/b58853fdbdf38a8a0cc7720c50b4d85b8a54dace beheben kannst?
Merkst du das eigentlich selbst nicht?

PeterPawn, 2020-07-13:
Was ist da jetzt das Problem?

In https://github.com/Freetz/freetz/commit/b58853fdbdf38a8a0cc7720c50b4d85b8a54dace werden genau drei Einstellungen für die BusyBox geändert - hat das irgendwo auch nur entfernte Ähnlichkeiten mit einem Patch in Freetz-NG, den Du durch diesen Pull-Request nicht ausreichend gewürdigt siehst?

Solange Du selbst keine PRs für das ursprüngliche Repo "einreichst", müssen das halt andere erledigen und hier ging es für die Besitzer einer passenden FRITZ!Box halt darum, das Problem gefixt zu sehen.

Ich sag's schon mal prophylaktisch an, daß auch ich Änderungen am "dropbear"-Paket in Freetz vorgenommen habe (https://github.com/PeterPawn/YourFreetz/tree/bump-dropbear), die - das liegt nun mal in der Natur der Sache - sich nicht wirklich von dem unterscheiden, was Du da (wie ich mittlerweile gesehen habe) in Freetz-NG machst ... und trotzdem ist jede einzelne Zeile davon (es kommen noch weitere Änderungen, daher gibt's noch keinen PR dazu von mir) von mir selbst geschrieben (bzw. von Tools, die ich selbst aufgerufen habe) und so werde auch ich da keine "Referenz" auf Freetz-NG angeben. Ich werde aber mal vorsichtshalber die Reihenfolge der Zeilen in meinen Patches (spez. in der "localoptions.h") noch ändern, damit mir nicht jemand am Ende erzählt, ich hätte irgendwo "abgeschrieben".

Angesichts der Tatsache, daß @WileC auch im IPPF-Thread dabei war und sich auch der Commit-Text wohl eher auf die dort diskutierten Schlußfolgerungen für die Konfiguration bezieht, als auf irgendwelche Patches in Freetz-NG, sehe ich absolut nichts im PR von @WileC, was auch nur den Anschein des "Klauens" in einem anderen Projekt erwecken würde. An der Diskussion im IPPF-Thread kann/darf jeder teilnehmen und wenn Du für Dich entschieden hast, das nicht zu tun, ist das Deine Sache.

Obendrein gibt's im dem PR auch noch den Verweis, woher die Info/Idee kommt ... mehr geht fast nicht.

So sehe ich das jedenfalls ... und da verwundert mich dann die Frage

> Merkst du das eigentlich selbst nicht?

schon etwas.

fda77, 2020-07-13:
Also ich finde das von @WileC schon ziemlich dreist. Und dies ist auch nicht der erste Zwischenfall mit ihm (Stichwort "Zufälle")
Mir ging es darum dass er Freetz-er13 nutzt, wo er sei Problem nach fast 1,5 Jahren nicht gelöst bekommen hat.
Deshalb macht er dann dieses Issue hier auf, um danach den Fix in Freetz zu nutzen... soe ein verhalten nennt man hier "jemanden verarschen"

PeterPawn, 2020-07-13:
Sehe ich anders ... die Lösung des Problems erfolgte nämlich gar nicht in dieser Issue - Du erinnerst Dich sicherlich, daß Du selbst hier mit Hinweis auf den älteren Eintrag (https://github.com/Freetz/freetz/issues/136) geschlossen hast und ich denke auch, daß ein guter Teil der Fehlersuche und -analyse im IPPF stattgefunden hat (auch Deine Commits referenzieren ja den Thread dort).

Wenn man sich nicht hinstellen und absichtlich Unterschiede einbauen will (wie ich das jetzt gerade mal für die "localoptions.h" im Dropbear-Paket getan habe), sind viele Änderungen schon deshalb sehr ähnlich, weil es eben eine gemeinsame Basis gibt und mal ganz ehrlich ... an dem von Dir monierten Patch kann man eigentlich gar nichts anders machen, wenn man die Optionen in sortierter Reihenfolge auflisten will (was ja auch sinnvoll ist, falls man mal einen Eintrag in der Liste sucht).

Meine Frage war ja, welcher Patch in Freetz-NG denn derjenige sein soll, der so überwältigende Ähnlichkeit mit diesem PR in Freetz/freetz hat. Und selbst wenn es so einen geben sollte ... das sind exakt die drei Optionen, die im IPPF-Thread angeführt sind und da die ihre Namen nun mal von der BusyBox kriegen, KANN man den Patch eigentlich gar nicht anders machen.

Was erwartest Du denn jetzt? Daß die Benutzer von "Freetz/freetz" künftig darauf verzichten, den Fehler (und ich bleibe dabei, daß es einer ist, der auch in Deinem Fork zu finden war und das auch noch nach den Änderungen, die Du zur Nutzung der verschiedenen BusyBox-Versionen eingebaut hast) zu beheben bzw. beheben zu lassen, weil Du der Erste warst, der den Patch dafür umgesetzt hat in seinem Fork?

fda77, 2020-07-13:
Es ging mir doch gar nicht um die Angabe des Autors! Das ist doch okay.
Bei Freetzs gibts schon ewig die Vereinbarung dass man mit "by" Commits kennzeichnet wenn man was von jemand anderem (fast) unverändert übernimmt. Ein "thx" wenn es auf Grund einer Idee committed wurde.

Sondern dass WileC öfter unangenehm auffällt. Zb die Sache mit get_ip & ipv6 support.
Das kann man in Iusses hier und bei Freetz nachlesen, ich bin jetzt zu faul die zu suchen.
Ich hatte die Option get_ip hinzugefügt durch ein sed auf ifconfig. WilleC übernahm das sed unverändert, verpasste aber der Option einen anderen Namen (Schöpfungshöhe annähernd 0). Nach seinen Angaben hat er "zufällig" exakt das gleich sed wie ich genommen. Im Anschluss hat er sich hier dann beschwert dass in den 2 Freetzen die Option unterschiedlich heisst, und ich sie doch ändern solle!
Da muss ich halt annehmen dass er es selbst "nicht merkt" wie dreist er drauf ist, daher finde ich die Frage berechtigt