Freetz-NG/freetz-ng #26: make menuconfig: "ERROR: unmet direct dependencies detected for FREETZ_BUSYBOX___V131_LSOF" seit 8fc9def5d

michael0815000, 2020-07-05:
Habe gerade git pull ausgeführt und bekomme jetzt einen Fehler:

freetz@freetz-linux:~/7590$ **git pull**
remote: Enumerating objects: 14, done.
remote: Counting objects: 100% (14/14), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 10 (delta 8), reused 10 (delta 8), pack-reused 0
Unpacking objects: 100% (10/10), 1.73 KiB | 104.00 KiB/s, done.
From https://github.com/Freetz-NG/freetz-ng
   b40cea63a..8fc9def5d  master     -> origin/master
Updating b40cea63a..8fc9def5d
Fast-forward
 make/busybox/Config.in | 84 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++---------------
 1 file changed, 69 insertions(+), 15 deletions(-)

freetz@freetz-linux:~/7590$ **make menuconfig**
WARNING: The program unar was not found in path.
WARNING: The program unar was not found in path.

**ERROR: unmet direct dependencies detected for FREETZ_BUSYBOX___V131_LSOF**
  Depends on [n]: FREETZ_REPLACE_BUSYBOX [=y] && FREETZ_BUSYBOX__VERSION_V131 [=y] && !FREETZ_PACKAGE_LSOF [=y]
  Selected by [y]:
  - FREETZ_BUSYBOX_LSOF [=y] && FREETZ_REPLACE_BUSYBOX [=y] && FREETZ_BUSYBOX__VERSION_V131 [=y]
make: *** [Makefile:382: menuconfig] Error 1

Hatte gestern noch ein neues Image bauen können.

fda77, 2020-07-05:
Hat sich ja seit gestern was getan ^^

closed 2020-07-05 by fda77

michael0815000, 2020-07-05:
VIELEN DANK! :-)

fda77, 2020-07-05:
Falls du nicht irgend was besonderes vom package lsof brauchst, kannst du es auch abwählen und das Busybox Applet nehmen (wird dann automatisch ausgewählt)