Freetz-NG/freetz-ng #27: dnsdist package

benleb, 2020-07-05:
hey cuma, first, thanks for your great work here! 🤝 

it would be really cool to have [dnsdist](https://dnsdist.org/) available as package :) I tried to implement it but... For someone who knows what to do, its probably much less work :D

Thanks!

fda77, 2020-07-05:
Sorry, no time.  Whats wrong with dnsmasq? The main feature of dnsdist "balance DNS queries to several backend servers" i'm using with with dnsmasq since a long time

benleb, 2020-07-05:
nothing is wrong with it, but its "consumer-grade" dns... dnsdist is an "isp grade" dns with much much much more flexibility (lua scripting)

fda77, 2020-07-05:
Are all required libs in freetz, like these for lua scripting? Could be much work

benleb, 2020-07-05:
I am not 100% sure but afaik just libedit is missing

fda77, 2020-07-05:
From https://dnsdist.org/install.html
```
Installing from Source

In order to compile dnsdist, a modern compiler with C++ 2011 support (like GCC 4.8+ or clang 3.5+) and GNU make are required. dnsdist depends on the following libraries:

    Boost
    Lua 5.1+ or LuaJit
    Editline (libedit)
    libsodium (optional)
    protobuf (optional)
    re2 (optional)
```
Freetz has the lua and a "protobuf-c", not sure if this is ok. The other things need to be added, if you need the features provided by these libs
So, start with libedit and boost :D

fda77, 2020-07-05:
I dont know what ragel is, but max need it too:
```
To compile from git, these additional dependencies are required:

    GNU Autoconf
    GNU Automake
    Ragel
```

Best would be you post your yet done work  to IPPF, and hopefully there someone could assist you

closed 2020-07-05 by fda77