Freetz-NG/freetz-ng #28: Multi Boot

webfalter, 2020-07-07:
Vor Monaten habe ich ein 7590-Labor-Freetz-Image auf die Box gespielt. Dieses gab danach beim starten immer eine Rebootschleife. Zugriff und Wechsel auf die andere Partition bekam ich durch Direktzugriff über Lan. Seitdem läuft diese wieder.
Wie kann ich jetzt die nichtstartende Partition mit einer laufenden Firmware wieder flashen?

fda77, 2020-07-07:
Ganz normal über das Webinterface. Es wird normal in die nicht-aktive Partition geflasht
Falls du tools/push_firmware über den Bootloader nutzt, kannst du auch als Parameter angeben in welche Parition du flashen möchest

webfalter, 2020-07-08:
Danke!!! hat geklappt.
Habe mit der 7.20 wieder das Problem, lag also nicht an der Labor.
Weiss ja jetzt wie ich wieder zurück komme.

michael0815000, 2020-07-08:
Habe leider auch den Bootloop mit 7.19 und 7.20 und gehe dann immer wieder auf die 7.12 in der anderen Partition zurück.
cat /proc/avm/log_sd/panic zeigt bei mir am Ende 

`<0>[  138.734953][2]Kernel panic - not syncing: ar7wdt_hw_reboot: init sequence hangs !`
`<0>[  138.734953][2]`
`<3>[  138.742734][2]set_reboot_status: Soft-Reboot(PANIC)  - PANIC(1)SUM(1)UP(138)UTC(138)FW(07.19-79768)HW(226)HWS(4)BV(1.3578)`

Hat evtl. damit zu tun, dass der Start zu lange dauert und der Watchdog nach 120 Sek. einen Timeout bekomt, was dann zum Reboot führt.
Nur: Wie kann man rausfinden, wo es hakt beim Start...

fda77, 2020-07-08:
"beim starten immer eine Rebootschleife" ist nicht wirklich eine Fehlerbeschreibung!
* Schau zuerst mal ob in deiner Freetz konfiguration wie zb in rc.custom komische dinge drin sind, wie warten auf ne internetverbindung oder irgend ein watchdog script
* Ansonsten kann ein avm daemon abstürzen, und der avm watchdog rebootet. Im IPPF beschrieb peterpawn irgendwo wie man den watchdog ausschaltet und schaut warum er das macht
* Kann auch sein dass was in freetz kaputt ist, bei nur 1 Meldung eher unwahrscheinlich

fda77, 2020-07-08:
@michael0815000: Am besten wäre ein serielle Console um zu sehen wie weit die Box kommt und was als letztes geschah
Ich hab mal nach "not syncing: ar7wdt_hw_reboot: init sequence hangs !" gesucht
https://www.ip-phone-forum.de/threads/302494/
Da hatte jemand in der rc.custom ein script, das beim starten mit wget etwas herunterladen sollte

webfalter, 2020-07-08:
Hallo @fda77 ich stecke da nicht so drin, wenn ich eine konkrete FM gesehen hätte, dann wäre sie auch hier gelandet.

- Die  rc.custom ist leer.
- Callmonitor habe ich mal raus gelassen, Reboot besteht trotzdem

Bin gern bereit weiter mit zutesten, brauche blos paar Hinweise von Euch Experten in Sachen Freetz.

fda77, 2020-07-08:
Bei Problemen mit einem Minimalimage testen, also ohne remove patches und wenigen paketen, nur was man unbedingt braucht, wie zb dropbear und syslog

michael0815000, 2020-07-09:
Nach "ar7wdt_hw_reboot: init sequence hangs !" hatte ich auch bereits gesucht und auch den von Dir erwähnten Thread gefunden.
Demgemäß hatte ich auch bereits die rc.custom überprüft. Das ist bei mir leider nicht die Ursache.
Der Callmonitor ist nicht im Image.
Ich habe jetzt noch einmal ein ganz frisches Image gebaut und dann den Watchdog deaktiviert mit
`echo "disable" > /dev/watchdog` 

Dadurch läuft die Box jetzt immerhin (bis zum nächsten unbeobachteten Restart :-)
Das Log schaue ich später mal durch.
Die erste Fehlermeldung, welche bei nicht vorhandenem Zeit-Sync einen Zeitstempel vom 31.12.2019 00:03:27 hatte war dann:

`Kategorie: User`
`Programm: Telefon`
`Nachricht: avmipc_send_kpi exited with stastu scode 255.`

Melde mich später erneut.

webfalter, 2020-07-09:
Minimalimage gleiches Verhalten. 
Hier https://www.ip-phone-forum.de/threads/freetz-f%C3%BCr-7-20-schon-verf%C3%BCgbar.307326/#post-2377608 scheint es ja auch noch einen zugeben.

fda77, 2020-07-09:
@webfalter: Schön dass jemand dieses Problem schon mit einer Labor hatte. Wenn man aber weder die Ursache sucht noch es überhaupt meldet, wird sich daran höchstwahrscheinlich nichts ändern
@michael0815000 Geht bei dir die DNS Auflösung auf *der Box*? Einfach mit "ping" testen. Evtl passt das zu einer anderen Fehlermeldung "Internet geht nicht"
Was chrony so treibt kann man sich mit *chronyc* anschauen, zb "chronyc sources". Glaube an der Zeitsynchronisationhat avm was verändert, ich kann die Option für den NTP Server nicht mehr auf der Weboberfläche finden, scheinbar werden einfach zufällige Zeitserver genommen
Das avmipc_send_kpi sagt mir nichst

fda77, 2020-07-09:
@webfalter Schau mal ob das gegen deinen "Bootloop" hilft.
Der Patches ist nicht als Dauerlösung, sondern zur Ursachensuche gedacht!
@michael0815000 das avmipc_send_kpi hab ich auf der 7490 mit neuer labor auch, scheint aber nicht zu stören

webfalter, 2020-07-09:
@fda77 Danke, dein Patch funktioniert erstmal und die Box startet wie normal.
Was hat das für Auswirkungen, weil Du schreibst es dient nicht als Dauerlösung ?

michael0815000, 2020-07-09:
Habe auch den [add disable-avm-watchdog patch](https://github.com/Freetz-NG/freetz-ng/commit/eb2f355e758b2ec2d0cb8a5f8917902f2ba79691) mal installiert und damit bleibt die Box auch bei mir oben - im Gegensatz zu meinem Versuch heute Morgen (`echo "disable" > /dev/watchdog`).

Das mit den Auswirkungen der Watchdog-Deaktivierung würde mich auch interessieren.

DNS Auflösung geht bei mir:
`root@fritz:/var/mod/root# ping www.heise.de`
`PING www.heise.de (193.99.144.85): 56 data bytes`
`64 bytes from 193.99.144.85: seq=0 ttl=249 time=8.758 ms`

`root@fritz:/var/mod/root# chronyc sources`
`210 Number of sources = 4`
`MS Name/IP address         Stratum Poll Reach LastRx Last sample`
`===============================================================================`
`^- ntp1.sul.t-online.de          2  10    37   486  -3282us[  -37ms] +/-   38ms`
`^- customer-85-204-133-12.i>     2  10    37   474  -3534us[-3534us] +/-   37ms`
`^- stz-bg.com                    2  10    37   475  +4318us[+4318us] +/-  112ms`
 `^* tock.ntp.infomaniak.ch        1  10    37   475   -574ns[  -34ms] +/-   12ms`

> @webfalter: Schön dass jemand dieses Problem schon mit einer Labor hatte. Wenn man aber weder die Ursache sucht noch es überhaupt meldet, wird sich daran höchstwahrscheinlich nichts ändern

Ich hatte mich übrigens am 04.06. bereits gemeldet: [7590: Bootloop nach Update von freetz 154.07.12 nach freetz-ng Labor 154.07.19-78839](https://www.digital-eliteboard.com/threads/7590-bootloop-nach-update-von-freetz-154-07-12-nach-freetz-ng-labor-154-07-19-78839.490237/) :-)

Hatte allerdings auch nicht die Zeit intensiver zu suchen und es war ja auch nur Labor und ich wollte nicht aufdringlich sein...

An welchen Stellen könnte ich denn sonst noch nach der Ursache suchen?
Mit Userids (hatte viele doppelte unter 1000, Benutzer tor u.a.) kann es nichts zu tun haben, oder?

fda77, 2020-07-09:
Mit der Uhrzeit sieht gut aus, war dann noch vor vorhandener Internetverbindung.
Der Patch macht nicht viel anderes, nur halt so früh wie es für Freetz möglich ist.
Zusätzlich noch das "init-done", aber nur dafür dass im Syslog steht wieviel Zeit noch übrige war.
Wieviele Sekunden waren es bei dir noch??
Ich hab auf der 7490 noch 175Sekunden von vermutlich 300, hat also etwas über 2 Minuten gebraucht bis zum Patch. Es soll aber auch Geräte geben bei denen es nur 120 maximal sind.
Das könnte auch schon der "Fehler" sein!
Wenn nicht: Im Syslog steht welche daemons überwacht wurden, schau ob die alle laufen
Im DEB lese ich nicht mehr, die Fehlerberichte dort sind so gut wie nie zu gebrauchen, und auch oft von Images mit Änderungen drin

michael0815000, 2020-07-09:
Sorry, dass ich frage, aber wonach muss ich im Syslog suchen, um die von Dir beschriebenen Zeitangaben zu finden? Bin mir unsicher. Habe folgendes im Syslog gefunden:
`Jan  1 01:00:59 fritz kern.crit kernel: [   58.061037][2]AVM_WATCHDOG: System Init Ueberwachung abgeschlossen (78 s noch verfuegbar)`
`Jan  1 01:00:59 fritz kern.crit kernel: [   58.061188][2]AVM_WATCHDOG_disable()`

"Disabling avm watchdog ... " konnte ich z.B. nicht finden.

Und die daemons erkenne ich an z.B.
`Jan  1 01:00:59 fritz user.notice FREETZMOD: Starting Inetd ... done.`

`root@fritz:/var/log# ps|grep inetd`
 `3351 root      1360 S    inetd`     ?

fda77, 2020-07-09:
Einfach alles nach "watchdog" durchsuchen, und die Zeilen nach dem disable(). Von einer 7490:
```
    1.720000] [avm] configured: watchdog event debug 
    1.720000] AVM_WATCHDOG: Watchdog Driver for AR7 Hardware (Version 1.0)
   15.890000] AVM_WATCHDOG: System Init Ueberwachung 240 Sekunden
   37.880000][0]AVM_WATCHDOG_release(hdl=6 avmnexusd): success
   44.660000][1]AVM_WATCHDOG_release(hdl=6 avmnexusd): success
   54.550000][0]AVM_WATCHDOG_release(hdl=14 kpid): success
   65.810000][1]AVM_WATCHDOG: System Init Ueberwachung abgeschlossen (190670 ms noch verfuegbar)
   65.830000][1]AVM_WATCHDOG_disable()
   65.830000][1]registered appls:
   65.830000][1]  hdl=2, supervisor, disabled.
   65.830000][1]  hdl=3, run_clock, disabled.
   65.830000][1]  hdl=4, boxnotifyd, disabled.
   65.830000][1]  hdl=5, untrustedd, disabled.
   65.830000][1]  hdl=6, avmnexusd, disabled.
   65.830000][1]  hdl=7, l2tpv3d, disabled.
   65.830000][1]  hdl=8, dsl_monitor, disabled.
   65.830000][1]  hdl=9, ctlmgr, disabled.
   65.830000][1]  hdl=10, multid, disabled.
   65.830000][1]  hdl=11, pcpd, disabled.
   65.830000][1]  hdl=12, wland, disabled.
   65.830000][1]  hdl=13, deviceinfod, disabled.
   65.830000][1]  hdl=14, pbd, disabled.
   65.830000][1]  hdl=15, telefon, disabled.
   65.830000][1]  hdl=16, upnpd, disabled.
   65.830000][1]  hdl=17, mobiled, disabled.
   65.830000][1]  hdl=18, meshd, disabled.
   79.640000][0]AVM_WATCHDOG: System Init Ueberwachung abgeschlossen (176840 ms noch verfuegbar)
nach Ablauf der 5 Minuten manuell ausgeführt:
  527.830000][1]AVM_WATCHDOG: System Init Ueberwachung abgeschlossen (4294695946 ms noch verfuegbar)
```
Es wird für jedes init-done ne Meldung ausgegeben, also das von freetz und das ursprüngliche von avm
Wie viele Sekunden sind noch bei dem anderen "System Init Ueberwachung" übrig?
Das "disabling" kommt nur in der freetz-logdatei als Erinnerung dass man es aktiviert hat, nicht im syslog

michael0815000, 2020-07-10:
`root@fritz:/var/log# grep -i watchdog *`
`7590.log:Jan  1 01:00:59 fritz kern.err kernel: [   42.566962][3]AVM_WATCHDOG_release(hdl=6 avmnexusd): success`
`7590.log:Jan  1 01:00:59 fritz kern.err kernel: [   53.519257][1]AVM_WATCHDOG_release(hdl=7 avmnexusd): success`
`7590.log:Jan  1 01:00:59 fritz kern.crit kernel: [   58.061037][2]AVM_WATCHDOG: System Init Ueberwachung abgeschlossen`
`(78 s noch verfuegbar)`
`7590.log:Jan  1 01:00:59 fritz kern.crit kernel: [   58.061188][2]AVM_WATCHDOG_disable()`
`7590.log:Jan  1 01:00:59 fritz kern.err kernel: [   58.061388][0][grx:watchdog] stop on cpu0`
`7590.log:Jan  1 01:00:59 fritz kern.err kernel: [   58.061410][1][grx:watchdog] stop on cpu1`
`7590.log:Jan  1 01:00:59 fritz kern.err kernel: [   58.061423][2][grx:watchdog] stop on cpu2`
`7590.log:Jan  1 01:00:59 fritz kern.err kernel: [   58.061446][3][grx:watchdog] stop on cpu3`

Ich habe also nur eine Zeile mit "System Init Ueberwachung abgeschlossen".

Eventuell ist auch noch folgendes interessant, oder ist das eine Folge der Deaktivierung des Watchdog:

`root@fritz:/var/log# cat dpm_manager.txt`
`[dpm_manager main][FORCE]AVM dpm_manager v2.0`
`[dpm_manager main][ERR]watchdog_register() failed, starting without watchdog`
`[dpm_manager plcservice][ERR]csock_bwrite(0x77c5b760,,15) failed with -5`
`[dpm_manager plcservice][ERR]csock_bwrite(0x77c5b760,,15) failed with -5`
`[dpm_manager plcservice][ERR]csock_bwrite(0x77c5b760,,15) failed with -5`
`[dpm_manager plcservice][ERR]csock_bwrite(0x77c5b760,,15) failed with -5`
`[dpm_manager plcservice][ERR]csock_bwrite(0x77c5b760,,15) failed with -5`
`[dpm_manager plcservice][ERR]csock_bwrite(0x77c5b760,,15) failed with -5`
`[dpm_manager plcservice][ERR]csock_bwrite(0x77c5b760,,15) failed with -5`

fda77, 2020-07-10:
Kann schon sein dass die 7490 und 7590 sich da unterscheiden, sind immerhin verschiedene CPUs und verschieden Kernelversionen.
Ich war auch davon überrascht, dass bei mir die "System Init Ueberwachung abgeschlossen". mehrfach kommen, scheint behoben

plcservice .. keine Ahnung. Probier doch mal aus was passier wenn du den watchdog angeschaltet lässt, und den Removepatch ""Remove PLCD (powerline)" aktivierst. Der ist recht überschaubar:
```
rm_files \
  "${FILESYSTEM_MOD_DIR}/lib/systemd/system/plcd.service" \
  "${FILESYSTEM_MOD_DIR}/etc/init.d/E50-plcd" \
  "${FILESYSTEM_MOD_DIR}/sbin/plcd" \
  "${FILESYSTEM_MOD_DIR}/usr/sbin/plcd" \
```
Mich hatten bei diesem damon die ständigen Broadcasts genervt...

michael0815000, 2020-07-10:
Hat leider nicht geholfen, die Reboots kamen zurück.

michael0815000, 2020-07-18:
Ich habe jetzt festgestellt, dass ich auch einen Reboot habe, sobald ich mit der MyFRITZ!-App eine Heimnetzverbindung aufbaue.
Die Reihenfolge der Logeinträge ist von unten nach oben chronologisch:

`2020-07-15    23:09:07    Information    fritz.box    kern    kernel    [ 935.485247][2]Kernel bug detected[#1]:`
`2020-07-15    23:09:07    Error    fritz.box    kern    kernel    [ 935.485074][2]BUG() at function 'esp_init_authenc' line: 771 file: /GU/KERNEL_grx5_build/linux/net/ipv4/esp4.c`
`2020-07-15    23:09:07    Error    fritz.box    kern    kernel    [ 935.483869][2]set_reboot_status: Soft-Reboot(KCRASH) - KCRASH(1)SUM(1)UP(934)UTC(1594847347)FW(07.20)HW(226)HWS(4)BV(1.3578)`
`2020-07-15    23:09:07    Error    fritz.box    kern    kernel    [ 935.483776][2]Code(0x80bd7974): 0x8eb50040 0x2c430001 <0x00030336> 0x9447000e 0x8e660000`
`2020-07-15    23:09:07    Error    fritz.box    kern    kernel    [ 935.483731][2][kernel-trap] pc=0x80bd797c(0x80bd797c esp_init_state+0x220/0x474) addr=0x9c09d300 task=vpnd pid=2234 ra=0x80bd7978(0x80bd7978 esp_init_state+0x21c/0x474)`

fda77, 2020-07-18:
Also falls du den master branch nutzt, sind kein kernel oder module für 7.20 vefügbar. Daher vermute(!) ich dass dieser Fehler von avm so gewollt ist... Ich kenn mich da aber nicht so gut aus
Es gibt diese "BUG_ON" auch die einen Reboots bei Verbindungsaufbau über OpenVPN auslösen, das mit PeterPawns yf_patchkernel verhindert wird
Dieser "vpnd" ist von avm, und nicht von dir auf die Box gepackt?
Das "line: 771 file: /GU/KERNEL_grx5_build/linux/net/ipv4/esp4.c" kannst du im Source nachschauen, das liegt bei Freetz entpackt unter source/kernel/....

michael0815000, 2020-07-27:
Das hört sich gut an. Bin vor dem Urlaub zur Sicherheit wieder auf 7.12 zurück, werde danach aber testen. Bin sehr gespannt :-)

fda77, 2020-07-27:
Hier hat das schon jemand freundlicherweise getestet: https://github.com/Freetz-NG/freetz-ng/issues/34#issuecomment-663858885
Was du ausprobieren könntest: Die Option "keep avm binaries" von https://github.com/Freetz-NG/freetz-ng/commit/a34135fcc88303373986232d9c97a5dc289b46ec aktivieren und den watchdog-patch wie im anderen Issue beschrieben deaktivieren

ChriBer, 2020-07-31:
Hab gerade mal kurz getestet mit "keep avm binaries" und watchdog aktiv... gibt bei mir leider trotzdem ständig reboots.

JokerGermany, 2020-11-09:
Ich würde gerne mithelfen die Quelle des Problems zu finden, da ich keine Version nutzen möchte, bei der der AVM-Watchdog deaktiviert ist...
https://gist.github.com/JokerGermany/6d1b97c0fd5c7b46fd9a6110fe9ce89f

https://www.ip-phone-forum.de/threads/avm-watchdog-kann-man-den-guten-gewissen-deaktivieren.308425/

fda77, 2020-11-11:
https://www.ip-phone-forum.de/threads/308425/#post-2395125

fda77, 2020-12-30:
Aktueller Status: Es wird nicht mehr der watchdog deaktiviert, sondern der untrustedd entfernt: https://github.com/Freetz-NG/freetz-ng/commit/66e419a52ce354c0696bc773c77a31a3b3fe9c34
@berndy2001: Mein ippf account ist wieder mal gesperrt... Das ist immer ein Krampf mit diesen möchtegern-Sittenwächtern

Conan179, 2020-12-30:
> Das ist immer ein Krampf mit diesen möchtegern-Sittenwächtern

wars der novize?

fda77, 2020-12-30:
Keine Ahnung, da steht nur wegen "verstössen". PMs sind nicht lesbar
Ich wurde vor 1 Monat schon wegen "fäkalsprache" verwarnt (hab vergessen von wem) weil ich "verarschen" benutzt hab. Da ich stets  freundlich und hilfsbereit bin und dazu heut Zeit hatte, haben ich einiges Beiträge die auch dieses Unwort enthalten mal gemeldet. Vielleicht 10, können aber auch 50 gewesen sein. Aber ich glaube weniger als 100. Mehr als 200 zeigt die Forensuche eh nicht an. Es ist übrigens voll nervig dass man zwischen jeder Meldung ne Zeit lang warten muss!
Wie daraus nun ein Regelverstoss konstruiert wird weiss ich nicht. Halt typische IPPF wie man es kennt

Conan179, 2020-12-30:
ok ein letzes offtopivc, kannst du mir mal eine email von dir zukommen lassen? will dir mal schicken.

fda77, 2020-12-30:
Ich markiere die als offtopic, damit man später nichts überflüssiges lesen muss. Aber kann man ja trotzdem noch ausklappen. Find ich ganz praktisch.
@PeterPawn Sorry dass ich nicht im ippf antworten konnte, siehe letzten Posts

fda77, 2020-12-30:
Mail ist wie meine Name hier ohne Zahlen beim deutschen gmx. Aber bitte nichts was hierunter Iusses oder Disukussion passt! (dies lösche ich gleich wieder)

PrinzVonBillAir, 2021-01-11:
Bei mir ist es so. dass wenn ich bei meiner gefreezten 7590 mit aktueller FINAL Firmware den AVM WD nicht deaktiviere, die Box alle 1-2 Minuten, nachdem sie komplett durchgebootet hat und die für kurze Zeit funktionsfähig ist, rebootet. Wenn ich den WD mittels freetz option deaktiviere läuft sie tadellos.

fda77, 2021-01-12:
Eigentlich braucht man den `disable-watchdog` nicht mehr wenn `remove-untrusedd` ausgewählt ist. Und der wird automatisch bei der 7590 ausgewählt. Meine 7590 läuft so
Der AVM-Watchdog schreibt ins syslog wenn ein Reboot ausgelöst wird, evtl erfährt man dadurch weshalb

PrinzVonBillAir, 2021-01-12:
Das war bei mir standardmäßig / automatisch nicht ausgewählt (habe extra nochmal neu ausgecheckt). Mit remove-untrusedd funktioniert es nun auch, wenn der WD nicht deaktiviert wird.

fda77, 2021-01-12:
Ich glaub der Watchdog ist 2 oder 4 Minuten, deshlab muss es was anderes sein. 
Schau am besten mal ins Log

berndy2001, 2021-01-12:
Ich habe das Problem auch, siehe [IPPF](https://www.ip-phone-forum.de/threads/avm-watchdog-kann-man-den-guten-gewissen-deaktivieren.308425/page-6#post-2403226).

fda77, 2021-01-13:
disable-watchdog ist nicht mehr nur für developer, das war ne Zeit so, damit beim automatischen Update die .config angepasst wird da das remove-untrustedd für die 7590 ausgewählt wird.
Wie gesagt, bei mir läuft so, schau ob du eine Log von dem ausgelösten Reboot speichern kannst, dort sollte drin stehen weshalb der Watchdog rebootet

fda77, 2021-03-01:
Gleiche Ursache wie https://github.com/Freetz-NG/freetz-ng/issues/120

closed 2021-03-01 by fda77