Freetz-NG/freetz-ng #29: Makefile:275: recipe for target 'firmware-nocompile' failed

syncgw, 2020-07-13:
Hi,
I just cloned your fork and tried to compile on a hyper-v "Ubuntu 18.04.3 LTS" (Win10). Unfortunately compile broke :-(
```
make
fakeroot: FAKEROOTKEY set to 2020429160
fakeroot: nested operation not yet supported
Makefile:275: recipe for target 'firmware-nocompile' failed
make: *** [firmware-nocompile] Error 1
```
Any how to solve?

fda77, 2020-07-13:
Easiest to resolve is to use a real Linux...
I have no Windows with WSL support, so i don't know anything about it
Here's a link to a VM which works for freetz: https://github.com/Freetz-NG/freetz-ng/blob/master/README.md#readme

In case you want to make freetz work with WSL:
 * https://www.ip-phone-forum.de/297537/
 * https://trac.boxmatrix.info/freetz-ng/ticket/2854

fda77, 2020-07-13:
The patch from IPPF is added

fda77, 2020-07-13:
Sry, link to ippf was broken: https://www.ip-phone-forum.de/threads/297537/

closed 2020-07-13 by fda77