Freetz-NG/freetz-ng #30: WireGuard

syncgw, 2020-07-20:
Hi, 
ich versuche einen WireGuard Server auf einer FritzBox 7490 zu implementieren. Leider antwortet der Server einfach nicht (siehe Posting in [Freetz-NG // Wireguard](https://www.ip-phone-forum.de/threads/freetz-ng-wireguard.307403/#post-2378849)). Hier sind die Konfig-Dateien [wg-conf.zip](https://github.com/Freetz-NG/freetz-ng/files/4947454/wg-conf.zip). 

Alle Tests mit `freetz-ng 16957M-7eef0a3ac master 2020-07-19 ` gemacht

syncgw, 2020-07-20:
Sorry. Es war ein Konfigrationsproblem. Nachdem ich [dieses Tool](https://github.com/brittson/wireguard_config_maker) gefunden habe, hat es funktioniert.

closed 2020-07-20 by syncgw

fda77, 2020-07-20:
Bei Konfigurationsprobleme ist ein Issue hier schlecht, ich kenne mich nicht mit jedem Programm aus. Im IPPF gibt es mehr Leute die bei solchen Problemen helfen können