Freetz-NG/freetz-ng #31: compiler bei der 7590

MasterRoCcO, 2020-07-20:
Hi ich habe grade mal versucht mit den neuen Freetz-ng von heute ein image zu bauen was aber nicht geht. ich habe nur auf expert umgestellt mehr nicht. Denn fehler bekomme ich.

make -f ./scripts/Makefile.asm-generic \
            src=asm obj=arch/mips/include/generated/asm
Cannot use CONFIG_CC_STACKPROTECTOR_REGULAR: -fstack-protector not supported by compiler
make[2]: *** [Makefile:1106: prepare-compiler-check] Fehler 1
make[2]: Verzeichnis „/home/freetz/freetz-ng/test/75xx/source/kernel/ref-grx5-7590_07.20/linux-4.9“ wird verlassen
make[1]: *** [GNUmakefile:101: prepare] Fehler 2
make[1]: Verzeichnis „/home/freetz/freetz-ng/test/75xx/source/kernel/ref-grx5-7590_07.20/linux-4.9“ wird verlassen

ERROR: Build failed.
make: *** [make/linux/kernel.mk:139: source/kernel/ref-grx5-7590_07.20/.prepared] Fehler 1
freetz@freetz-debian:~/freetz-ng/test/75xx$

kamasudo, 2020-07-20:
Weshalb wird bei dir versucht den Kernel zu bauen?
Ausserdem fehlen elementare Infos wie die FOS Version, git-Revision, git-Branch usw. Am besten deine .config
Hast du an einer .in was verändert? Evtl nach einer Anleitung aus dem DEB?

fda77, 2020-07-20:
Weshalb wird bei dir versucht den Kernel zu bauen?
Ausserdem fehlen elementare Infos wie die FOS Version, git-Revision, git-Branch usw. Am besten deine .config
Hast du an einer .in was verändert? Evtl nach einer Anleitung aus dem DEB?

MasterRoCcO, 2020-07-20:
Sagte doch die neue Version von heute freetz-ng 16960-83d9ef86f master 2020-07-20
FW 7.20
[.config.txt](https://github.com/Freetz-NG/freetz-ng/files/4949224/default.config.txt)

MasterRoCcO, 2020-07-20:
vor den Test habe ich genau das gemacht
git clone https://github.com/Freetz-NG/freetz-ng ~/freetz-ng/test/75xx
dann 
cd /home/freetz/freetz-ng/test/75xx
dann
make menuconfig
und nur 2 sachen geändert expert modus und AVM tagging (none) 
save 
und make
dann kommt der fehler nach der zeit

fda77, 2020-07-20:
Dass du "master" verwendest stand da nicht, es gibt nämlich Unterschieden die den kernel 4.9 betreffen in einem branch.
Da in deiner Config steht "Kernel modules (not available, needs AVM sources)" sollte der kernel auch nicht gebaut werden.
Ich probier das mal aus

MasterRoCcO, 2020-07-20:
So ich habe mir mal kernel49 gesaugt und da mit versucht, damit geht es

fda77, 2020-07-20:
Blöd, es wird wohl doch versucht eine Kernel zu bauen, auch wenn es nicht möglich ist. Ich merge gleich den Branch in den master, dann brauche ich nicht zu suchen warum das so ist
Auslöser des Fehlers müsste  https://github.com/Freetz-NG/freetz-ng/commit/23b5254fcec2f99598ee289581280202cc72153a sein

closed 2020-07-20 by fda77