Freetz-NG/freetz-ng #32: 7590 7.20: AVM tagging verhindert schließen von Popups

berndy2001, 2020-07-23:
Der Patch Web Interface - AVM tagging verhindert das schließen von Overlay-Popups weil der Button dazu fehlt. Siehe Bilder.
Es gibt keine andere Möglichkeit das Popup zu schließen.

[image]
[image]

fda77, 2020-07-24:
Müsste jetzt funktionieren. Sieht nun etwas anders aus, da es SVG Dateien sind.
Vielleicht bist du Grafiker und kannst es verbessern

closed 2020-07-24 by fda77

berndy2001, 2020-07-24:
Bin ich tatsächlich ein bisschen.

Beim Patch ist mir noch aufgefallen, dass danach das svg-File mit `

  | <svg xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" viewBox="0 0 177.1 139.1" enable-background="new 0 0 177.1 139.1"><svg>
-- | --` und mit `</svg></svg>` endet, was für mein Verständis doppelt ist.

fda77, 2020-07-24:
Ist doppelt und so gewollt!
Original avm:
 ```<svg> avm-bild <svg/>```
Mit Tag:
```
<svg>
  <svg> avm-bild </svg>
  <svg> tag-overlay</svg>
</svg>
```
Die Dimensionen sind nur in den äusseren Container geschoben, damit die komplette grafik so ist wie AVM es erwartet.
Falls du auch einen besseren Weg kennst 2 svg (oder gar avm-svg + freetz-png) automatisiert zu verschmelzen, her damit!

berndy2001, 2020-07-24:
Ok, das macht Sinn. Der Patch macht aber derzeit `<svg><svg>avm.bild tag-overlay</svg></svg>`
Siehe [logo_fritzDiamond.svg.txt](https://github.com/Freetz-NG/freetz-ng/files/4972206/logo_fritzDiamond.svg.txt)

Besseren Weg zum verschmelzen kenne ich leider auch nicht. Man kann zwar Pixel-Grafiken in svg einbetten, aber Sinn macht es keinen und der Aufwand ist dafür auch zu hoch. Besser ist es die Tags neu zu erstellen, dass kann ich gerne noch machen.

fda77, 2020-07-24:
Das svg vom Freetz-MOD tag ist auch noch bisschen zu gross. Hab die mit Hilfe verschiedener Grafikprogramme "konvertiert"
Das Priskrat hab ich gar nicht hinbekommen, da das so pixelig und bunt ist. Macht aber nichts, da bei dem sogar das obere Ende vom "f" fehlt...
Kannst auch ein ganz neues zusätzlich machen, so wie es dir gefällt. Es muss auch nicht unbedingt "freetz" drin stehen, vielleicht modifiziert, gepimpt oder was auch immer... Sollte nur irgendwie auch für andere nutzbar sein, also nicht den Namen von deinem Goldfisch