Freetz-NG/freetz-ng #34: 7590 7.20: Busybox, Parameter beim ip Kommando für vpnd

cnbr, 2020-07-24:
Bei der 7.20 verwendet der vpnd wohl das Kommando ip, welches mit den Parametern` -b /var/tmp/xfrm_bootstrap` oder  `xfrm state flush` aufgerufen wird, 

In dem ip der busybox aus freetz werden diese Parameter nicht unterstützt,  evtl. ist das auch mit ein Grund für den Reboot beim Aufruf von vpn

Ausgabe beim Start von vpnd auf der Konsole 

```

root@fritz:/var/mod/root# vpnd -v -f -D all
vpnd: watchdog_csock_init() failed, running without watchdog
BusyBox v1.31.1 (2020-07-20 19:21:11 UTC) multi-call binary.

Usage: ip [OPTIONS] address|route|link|tunnel|neigh|rule [ARGS]

OPTIONS := -f[amily] inet|inet6|link | -o[neline]

ip addr add|del IFADDR dev IFACE | show|flush [dev IFACE] [to PREFIX]
ip route list|flush|add|del|change|append|replace|test ROUTE
ip link set IFACE [up|down] [arp on|off] [multicast on|off]
	[promisc on|off] [mtu NUM] [name NAME] [qlen NUM] [address MAC]
	[master IFACE | nomaster]
ip tunnel add|change|del|show [NAME]
	[mode ipip|gre|sit] [remote ADDR] [local ADDR] [ttl TTL]
ip neigh show|flush [to PREFIX] [dev DEV] [nud STATE]
ip rule [list] | add|del SELECTOR ACTION
BusyBox v1.31.1 (2020-07-20 19:21:11 UTC) multi-call binary.

Usage: ip [OPTIONS] address|route|link|tunnel|neigh|rule [ARGS]

OPTIONS := -f[amily] inet|inet6|link | -o[neline]

ip addr add|del IFADDR dev IFACE | show|flush [dev IFACE] [to PREFIX]
ip route list|flush|add|del|change|append|replace|test ROUTE
ip link set IFACE [up|down] [arp on|off] [multicast on|off]
	[promisc on|off] [mtu NUM] [name NAME] [qlen NUM] [address MAC]
	[master IFACE | nomaster]
ip tunnel add|change|del|show [NAME]
	[mode ipip|gre|sit] [remote ADDR] [local ADDR] [ttl TTL]
ip neigh show|flush [to PREFIX] [dev DEV] [nud STATE]
ip rule [list] | add|del SELECTOR ACTION
vpnd: [main]: startup ($Revision$$CompileDate: Jun 22 2020 15:27:30 $)
client_loop: send disconnect: Broken pipe

```
Die vermutlichen Parameter für den Aufruf habe ich per `strings vpnd` ausgefiltert

```
....
/var/tmp/xfrm_bootstrap
[%s]: fopen() failed, cannot bootstrap XFRM algorithms, kernel might BUG()
xfrm state add src 0.0.0.0 dst 0.0.0.0 proto esp spi 0x0
%s enc "cbc(des)" 0x%016X auth "hmac(md5)" 0x%032X
%s enc "cbc(aes)" 0x%032X auth "hmac(sha256)" 0x%064X
%s enc "cbc(aes)" 0x%032X auth "hmac(sha384)" 0x%096X
%s enc "cbc(aes)" 0x%032X auth "hmac(sha512)" 0x%0128X
%s enc "cbc(des3_ede)" 0x%048X auth "hmac(sha1)" 0x%040X
ip -b /var/tmp/xfrm_bootstrap
ip xfrm state flush
....
```

Inhalt einer /var/tmp/xfrm_bootstrap
`root@fritz:/var/mod/root# cat  /var/tmp/xfrm_bootstrap
xfrm state add src 0.0.0.0 dst 0.0.0.0 proto esp spi 0x0 enc "cbc(des)" 0x0000000000000001 auth "hmac(md5)" 0x00000000000000000000000000000001
xfrm state add src 0.0.0.0 dst 0.0.0.0 proto esp spi 0x0 enc "cbc(aes)" 0x00000000000000000000000000000001 auth "hmac(sha256)" 0x0000000000000000000000000000000000000000000000000000000000000001
xfrm state add src 0.0.0.0 dst 0.0.0.0 proto esp spi 0x0 enc "cbc(aes)" 0x00000000000000000000000000000001 auth "hmac(sha384)" 0x000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001
xfrm state add src 0.0.0.0 dst 0.0.0.0 proto esp spi 0x0 enc "cbc(aes)" 0x00000000000000000000000000000001 auth "hmac(sha512)" 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001
xfrm state add src 0.0.0.0 dst 0.0.0.0 proto esp spi 0x0 enc "cbc(des3_ede)" 0x000000000000000000000000000000000000000000000001 auth "hmac(sha1)" 0x0000000000000000000000000000000000000001
root@fritz:/var/mod/root# 
`

Vermutung von mir, 
Die Datei /var/tmp/xfrm_bootstrap wird vom vpnd erstellt und soll anschließend über `ip -b /var/tmp/xfrm_bootstrap` aufgerufen, was aber schief geht.
Da dort auch Bezug auf esp genommen wird, besteht vielleicht ein Zusammenhang mit [https://github.com/Freetz-NG/freetz-ng/issues/28#issuecomment-660510060](url)

Per `make menuconfig` habe leider keine Option für die busybox mit ip -b oder ip xfrm gefunden.

fda77, 2020-07-24:
Interessante Entdeckung! AVM nutzt bei manchen Boxen wohl eine ip-Binary, ab FOS 6.8. Die Versionen der busybox haben öfter nicht alle Optionen der normalen Tools. Zum testen: ```find build/ -name ip -exec ls -al {} ';'```
Dann in diese Zeile: https://github.com/Freetz-NG/freetz-ng/blob/master/make/busybox/busybox.mk#L64  'wget' mit 'wget|ip' ersetzen. evtl "make busybox-dirclean", "make" und nochmal mit find schauen und flashen

cnbr, 2020-07-24:
Vielen Dank für den schnellen Tip, das war es. 
vpnd läßt sich ohne Fehlermeldung aufrufen, vpn läuft auch, ohne Reboot der Box.
Vor der Änderung

```
freetz@freetz-linux:~/freetz-ng$ find build/ -name ip -exec ls -al {} ';'
-rwxrwxrwx 1 freetz freetz 482376 Jun 30 13:46 build/original/filesystem/sbin/ip
lrwxrwxrwx 1 freetz freetz 14 Jul 20 19:21 build/modified/filesystem/sbin/ip -> ../bin/busybox
freetz@freetz-linux:~/freetz-ng$

```

So sieht es nun aus, 
```
freetz@freetz-linux:~/freetz-ng$ find build/ -name ip -exec ls -al {} ';'
-rwxrwxrwx 1 freetz freetz 482376 Jun 30 13:46 build/original/filesystem/sbin/ip
-rwxrwxrwx 1 freetz freetz 482376 Jun 30 13:46 build/modified/filesystem/sbin/ip
freetz@freetz-linux:~/freetz-ng$ 
```

fda77, 2020-07-24:
Das verlinkte Problem sollte nichts damit zu tun haben, es sollte eher das gleiche wie https://www.ip-phone-forum.de/threads/.307326/post-2378631 und https://github.com/PeterPawn/YourFritz/issues/34 sein, also durch BUG_ON im Kernel die AVM da hineingemacht hat...

"ip" könnte aber der Auslöser für die "bootloops" sein, also die Reboots durch den avm-Watchdog.
Falls du Lust hast lösch diese Zeile wieder: https://github.com/Freetz-NG/freetz-ng/commit/cb329c7e4a9eae8c54d84e1067be2ef7794f9305 Danach unter other-pactches den DISABLE_AVM_WATCHDOG deaktivieren und schauen ob die Box 2 Minuten nach einem Reboot allein rebootet. Issue dazu: https://github.com/Freetz-NG/freetz-ng/issues/28

closed 2020-07-24 by fda77

cnbr, 2020-07-25:
Habe es wg. "bootloops" "bootloops" probiert, das binary von ip hat nun keine Auswirkungen auf den watchdog. Reboot erfolgte damit nach ca 2 Min nach dem Starten.

fda77, 2020-07-25:
Schade, aber danke fürs testen!
Es wird jetzt beim Installieren der Links angezeigt falls eine Datei überschrieben wird, bei der 7590 7.20 sind es bunzip2 bzcat bzip2 dmesg renice. Man kann das optional nicht ersetzen. Keine Ahnung welche Auswirkungen das Ersetzen bislang hatte