Freetz-NG/freetz-ng #37: libgpg-error Fehler mit gawk 5.0.1

forenuser, 2020-08-06:
Bei Verwendung von gawk 5.0.1 wird make aufgrund eines Fehlers abgebrochen. Wenn libgpg-error von 1.36 auf 1.37 hochgezogen wird, funktioniert läuft make wieder durch.

Siehe dazu: [https://www.digital-eliteboard.com/threads/libgpg-error-build-issue-error-errnos_msgstr-undeclared-first-use-in-this-function.491778/](url)

@fda77, leider bist Du im Digital Eliteboard nicht mehr als Benutzer registriert. Wie kann ich Dich alternativ erreichen?

fda77, 2020-08-06:
Hatte keine Lust mehr auf DEB, bin demnächst evtl in einem anderen Forum
Der Link funktioniert, ist aber nutzlos das SPOILER und CODE nur mit Anmeldung lesbar sind

Wenn es mit 1.37 funktioniert hast du es vermutlich ausprobiert. Dann erstell doch einen pull-request
Kurzanleitung: https://freetz-ng.github.io/freetz-ng/SUPPORT
Punkt 2+3 sind für eine Library meist nicht nötig

forenuser, 2020-08-06:
Im ip-phone-forum konnte ich an cuma keine PN schicken.

Inhalt des Posts im DEB ist folgender:

Hallo,

leider habe ich mein freetz-linux 1.5.2 zerschossen. Deshalb nutze ich nun freetz-linux 1.5.7 (ubuntu 20.04). Dadurch wird nun eine neue Version von gawk verwendet, welche Fehler beim Kompilieren von libgpg-error erzeugt. Der Fehler ähnelt dem bei GnuPG behandelten Fehler (https://dev.gnupg.org/T4469). Ich nehme an, dass das verwendete Paket aktualisiert oder gepatched werden muss. Kann ein Entwickler sich das bitte mal ansehen? Vielleicht könnte eine Aktualisierung von libgpg-error auf Version 1.37 den Fehler beheben.

```
make[3]: Entering directory '/home/freetz/7590/source/target-mips_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-3.10/libgpg-error-1.36/src'
/home/freetz/7590/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc -DHAVE_CONFIG_H -I. -I..  -DPKGDATADIR=\"/usr/share/libgpg-error\" -DLOCALEDIR=\"/usr/share/locale\"  -D_REENTRANT -D_GNU_SOURCE  -march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wall -Wpointer-arith -Wno-psabi -fvisibility=hidden -MT gpg_error-strerror-sym.o -MD -MP -MF .deps/gpg_error-strerror-sym.Tpo -c -o gpg_error-strerror-sym.o `test -f 'strerror-sym.c' || echo './'`strerror-sym.c
strerror-sym.c: In function 'gpg_strerror_sym':
strerror-sym.c:45:13: warning: implicit declaration of function 'errnos_msgidxof' [-Wimplicit-function-declaration]
       idx = errnos_msgidxof (code);
             ^
strerror-sym.c:47:9: error: 'errnos_msgstr' undeclared (first use in this function)
  return errnos_msgstr + errnos_msgidx[idx];
         ^
strerror-sym.c:47:9: note: each undeclared identifier is reported only once for each function it appears in
strerror-sym.c:47:25: error: 'errnos_msgidx' undeclared (first use in this function)
  return errnos_msgstr + errnos_msgidx[idx];
```
Der make läuft mit der Version 1.37 durch. Das erzeugte Image läuft seit gestern auf meiner Box. Da ich nicht weiß, von welcher Funktion libgpg-error verwendet wird, kann ich nicht genau sagen, ob alles noch so läuft, wie bisher. Trotzdem kann ich mal probieren, einen Pull Request zu erstellen.

Viele Grüße.

fda77, 2020-08-06:
Auf "cuma" im IPPF hab ich keinen Zugriff, wie gesagt "evtl demnächst"
Da die neue Version läuft, ist doch prima. Warum die alt nicht mehr geht? keine Ahnung!

forenuser, 2020-08-06:
Mit gawk 5.0.1 funktioniert die verwendete Syntax (/\#) wohl nicht mehr, wie mit gawk 4.x.

forenuser, 2020-08-06:
Den Pull Request habe ich nun erstellt.

fda77, 2020-08-06:
Danke! Hattest du https://freetz-ng.github.io/freetz-ng/SUPPORT (Package bump) angeschaut?
Hast nämlich die CHANGELOG vergessen

Und wenn LIB_VERSION verändert werden muss, betrifft das oft auch die external.* Dateien im Library/Package Verzeichnis da sich der Dateinamen ändert

forenuser, 2020-08-07:
Garnicht so einfach ;-)
Ich hoffe, dass es nun passt.

fda77, 2020-08-07:
Ich glaube schon. Da du die lib benutzt, wirst du es merken :D
Nur blöd dass der merge so viele commits sind und nicht nur einer. Man kann das auf github zwar als squash machen, nur ich glaube dann wird der author verändert

closed 2020-08-07 by fda77

forenuser, 2020-08-08:
Macht es was aus, dass die wiki Änderungen nicht mit übernommen wurden?
```
freetz@freetz-linux:~/freetz-ng$ git diff-tree --no-commit-id --name-only -r 49d2bb7427b64b3820720bc0ac5b30b6d2ed1861
docs/make/README.md
docs/patches/README.md
docs/wiki/00_FAQ/README.md
docs/wiki/10_Beginner/README.md
docs/wiki/20_Advanced/README.md
docs/wiki/30_Expert/README.md
docs/wiki/40_Troubleshooting/README.md
docs/wiki/50_Security/README.md
docs/wiki/60_Development/README.md
docs/wiki/70_Various/README.md
docs/wiki/README.md
make/README.md
```
Allerdings fehlen teilweise Umlaute, so dass es wohl besser ist, die Änderungen nicht zu übernehmen.

fda77, 2020-08-08:
Ich verstehe die Frage nicht so ganz. Was wurde von wem wobei wohin nicht übernommen?

forenuser, 2020-08-09:
Unter https://freetz-ng.github.io/freetz-ng/SUPPORT gibt es unter Package Bump den Punkt "Das Wiki aktualisieren durch ausführen von: docs/generate.sh". Ich nehme an, dass dabei die angegebenen Dateien geändert wurden, welche in meinem Commit zu libgpg-error nicht enthalten sind. Da dabei offensichtlich einige Umlaute verschluckt wurden, möchte ich diese Änderungen nicht in das GIT Repository übernehmen.

fda77, 2020-08-09:
Aschso. Wenn ich bei mir generate.sh ausführe, werden keine Datein geändert. Das liegt wohl an unterschiedlichen Spracheinstellungen. Ob die nun bei mir oder bei dir falsch sind weiss ich nicht.
Durch das Ausführen sollen die Packageversionen übernommen werden, wie man an anderen commits sieht