Freetz-NG/freetz-ng #38: Einige Kernelmodule werden nicht gefunden (7590 FW 7.12)

forenuser, 2020-08-06:
Mit der aktuellen freetz-ng trunk Version bricht der Make Prozess ab, weil einige Kernel Module nicht gefunden werden.

Fehlermeldungen:
WARNING: File of kernel module not found: exportfs.ko
WARNING: File of kernel module not found: lockd.ko
WARNING: File of kernel module not found: sunrpc.ko

Beiliegend die von mir verwendete Konfiguration:
[config.txt](https://github.com/Freetz-NG/freetz-ng/files/5033066/config.txt)

Mit einer anderen Konfiguration wird auch das folgende Modul nicht gefunden:
WARNING: File of kernel module not found: fuse.ko

fda77, 2020-08-06:
Hinter ERROR steht doch wie man das beheben kann. 
Vor kurzem wurden ~10 Kernel-Sourcecodes hinzugefügt, das wäre ein Riesenauswand alle Module für alle Lebenslagen auszuwählen. So wird nur die .config von AVM verwendet, und wenn was fehlt wird es angezeigt...
Zumal man ohne Hardware gar nicht prüfen kann ob die kernel-.config dann auch läuft. Manche Module funktionieren nur wenn der Kernel auch mir diesen Modulen gebaut wurde

forenuser, 2020-08-06:
Ich verwende noch die 7.12 FW für meine 7590, da ich im DEB die Antwort erhalten habe, dass zusätzliche Kernel Module für die 7.20 FW nicht gebaut werden können. Der Source Code für diese FW Version scheint jedoch seit Mitte Juli verfügbar zu sein.

fda77, 2020-08-06:
Im DEB wird viel Unsinn erzählt. Die Module die du brauchst musst du aktivieren, das gilt für alle Kernel(versionen). Wenn sie nicht aktiviert sind kommt die Meldung oben

forenuser, 2020-08-06:
Es fällt mir sehr schwer, unter kernel-menuconfig die zugehörigen Optionen zu finden.

fda77, 2020-08-06:
Deshalb hatte ich keine Lust das für die 10 neuen Kernel zusammenzusuchen
Im menuconfig sucht man mit "/". Dort steht dann auch warum einen Option evtl nicht sichtbar ist (depends)

Wenn was fest im kernel ist, braucht man im freetz-menuconfig das symbol nicht auszuwählen. Oder trotzdem als Modul bauen und schauen ob es ohne das Modul zu laden funktioniert.
Die .config die AVM beilegt muss nicht die sein die der Kernel auf der Box verwendet

forenuser, 2020-08-06:
Laut der beigelegten .config sollten auch GPT Partitionen unterstützt werden. Ob das funktioniert habe ich noch nicht getestet, da bisher die Module für nfs gefehlt haben, um eine 7.20er FW mit nfs zu bauen. Kann man .config irgendwie aus dem originalen AVM Kernel extrahieren?

fda77, 2020-08-06:
Ja, falls CONFIG_IKCONFIG_PROC im Kernel aktivert ist (Google hätte dir schneller geantwortet)

forenuser, 2020-08-06:
An der Kernel Konfiguration sollte es bei der 7590 7.12 eigentlich nicht liegen, da dies mit einem älteren Freetz-NG Stand funktioniert und die gleichen Kernel Quellen verwendet werden. Hat sich an der verwendeten .config vielleicht etwas geändert?

fda77, 2020-08-06:
Schau nach, bei git/svn hat jede Datei ne History
Was neu ist ist jedenfalls die Meldung "Enable missing with 'make kernel-menuconfig' as (M)odule (and push a pull request)."
Bevor es die gab konntest du auswählen was du wolltest. Ins Image kam allerdings nur was vorhanden war
Dh evtl brauchst du die Module nicht
Es steht nicht umsonst ganz am anfang bei knows problems: https://freetz-ng.github.io/freetz-ng/CHANGELOG.html

forenuser, 2020-08-06:
OK, ich habe gerade mal nachgesehen. Diese Module sind im alten Stand nicht vorhanden und werden vermutlich nicht benötigt. Nun wird deren Vorhandensein mit dem neuen Stand überprüft und make bricht ab. Wie komme ich nun weiter?

fda77, 2020-08-06:
Ich vermute du muss in https://github.com/Freetz-NG/freetz-ng/blob/master/config/pkg/nfs.in was ändern. Ich blicke da nicht ganz durch

forenuser, 2020-08-06:
Ich arbeite gerade an 7590 7.20. Dort sind die Module im kompilierten Kernel bereits verfügbar. Mal sehen, wie weit ich damit komme.

Sollte man für 7590 7.12 die entsprechenden Kernel Sourcen verwenden? Derzeit werden die 7583 7.10 Sourcen verwendet.

fda77, 2020-08-06:
Das stimmt nicht, es wird ein "delta"-patch angewendet

forenuser, 2020-08-06:
War nur eine Idee, da ich gesehen hatte, dass AVM die 7590 7.12 Sourcen mit Datum Mitte Juli 2020 bereitgestellt hat.

Bei 7590 7.20 fehlte nur pl2303.ko. Diese habe ich nun mit make kernel-menuconfig ausgewählt. Damit ist make nun durchgelaufen.

fda77, 2020-08-06:
Jetzt wirds schwierig.. du musst evtl noch irgend was anderes im kernel aktivieren...
Ich würde ich kernel source durchsuchen: grep locks_end_grace source/kernel/....... -r
Im schlimmsten Fall muss man irgendwas am Kernel herumpatchen!

forenuser, 2020-08-06:
Ich werde morgen nochmal probieren, ob ich weiter komme. Derzeit funktioniert nfsd nicht ansonsten läuft die Box schon mal.
`root@fritz:/var/mod/etc/init.d# rc.nfsd restart
Stopping nfsd ... not running.
Starting nfsd ... nfsd: Unable to access /proc/fs/nfsd errno 2 (No such file or directory).
Please try, as root, 'mount -t nfsd nfsd /proc/fs/nfsd' and then restart nfsd to correct the problem
failed.
root@fritz:/var/mod/etc/init.d# mount -t nfsd nfsd /proc/fs/nfsd
mount: mounting nfsd on /proc/fs/nfsd failed: No such device
root@fritz:/var/mod/etc/init.d# rc.nfsd restart
Stopping nfsd ... not running.
Starting nfsd ... nfsd: Unable to access /proc/fs/nfsd errno 2 (No such file or directory).
Please try, as root, 'mount -t nfsd nfsd /proc/fs/nfsd' and then restart nfsd to correct the problem
failed.
`

fda77, 2020-08-06:
Wenn unter /proc was fehlt heisst das oft dass ein Kernelmodul nicht geladen ist

forenuser, 2020-08-07:
Es werden offensichtlich nicht mehr alle compilierten Kernel Module in das Image übernommen, sondern nur die, die mit make menuconfig ausgewählt wurden bzw. auswählbar sind. Für nfs sind einige Kernel Module vorhanden, welche deshalb nicht in das Image übernommen werden. Wenn ich grace.ko im Image habe, funktioniert der nfs Server.

Allerdings gibt es folgende Probleme:
1. Die Festplattenpartition wird beim Neustart der Box zu spät gemountet, so dass das Verzeichnis data im Flash angelegt wird und transmission und pyload dort Verzeichnisse anlegen. Dadurch wird die Festplattenpartition als data-1 gemountet.
2. Der Neustart der Box läuft nicht durch, so dass ich immer die Stromversorgung trennen muss.

Hast Du eine Idee, wie den 1. Punkt lösen könnte?

fda77, 2020-08-07:
Ich glaub "grace.ko" gitb es im menuconfig von freetz noch gar nicht. Damit es ins Image kommt muss es dort vorhanden und ausgewählt sein. Dies hängt dann vermutlich noch von irgend einer kernel-version ab

Hast du udevmount im image? Das sollte alles beenden was eine umount verhindert, wenn die option im webif aktiviert ist.

Ich hab das bei mir so gemacht, dass external aktiviert ist, aber nichts zum auslagern ausgewählt.
Im webif dann bei den eigenen Diensten die eingetragen, die einen USB-Stick benötigen
external wird erst aktiv wenn
 - freetz gestartet ist
 - eine uhrzeit gesetzt ist
 - das angegebene Verzecihnis gemountet wurde

ausserdem beendet external die eingetragenen dienste bevor umount ausgeführt wird duch den udevmount-hook

forenuser, 2020-08-07:
"grace.ko" habe ich bei mir in modules.in eingetragen. "nfsv2/3" habe ich ebenfalls ergänzt, aber nfs derzeit nicht aktiviert.
```
config FREETZ_MODULE_grace
        bool "grace.ko"
        depends on FREETZ_NFS_SUPPORT_AVAILABLE_AS_MODULE
        default n

config FREETZ_MODULE_nfs
        bool "nfs.ko"
        depends on FREETZ_NFS_SUPPORT_AVAILABLE_AS_MODULE
        select FREETZ_MODULE_lockd
        select FREETZ_MODULE_nfsv2
        select FREETZ_MODULE_nfsv3
        default n

config FREETZ_MODULE_nfsv2
        bool "nfsv2.ko"
        depends on FREETZ_NFS_SUPPORT_AVAILABLE_AS_MODULE
        default n

config FREETZ_MODULE_nfsv3
        bool "nfsv3.ko"
        depends on FREETZ_NFS_SUPPORT_AVAILABLE_AS_MODULE
        default n
```
Ich habe im Freetz Menü unter freetz die Einträge udev_first und udev_final. Andere Optionen habe ich noch nicht gefunden.

Ich werde den Tipp mit external mal ausprobieren.

forenuser, 2020-08-07:
Python funktioniert leider nicht. Hast Du dazu eine Idee?
```
root@fritz:/usr/bin# ls -al py*
lrwxrwxrwx    1 root     root             7 Aug  7 19:09 python -> python2
lrwxrwxrwx    1 root     root             9 Aug  7 19:09 python2 -> python2.7
-rwxr-xr-x    1 root     root           401 Aug  5 19:56 python2.7
-rwxr-xr-x    1 root     root          3696 Aug  7 16:52 python2.7.bin
root@fritz:/usr/bin# ./python
./python: exec: line 9: /usr/bin/python2.7.bin: not found
root@fritz:/usr/bin# ./python2.7.bin
-sh: ./python2.7.bin: not found

```

fda77, 2020-08-07:
Was wird denn aus der grace.ko?
4kb ist bisschen wenig für python

forenuser, 2020-08-07:
Was wird denn aus der grace.ko?
--> Meinst Du, ob ich meine oben angegebene Änderung der Datei modules.in in Form eines Pull Request hochlade? Kann ich gerne machen.
Ich habe auch gedacht, dass die Größe des Executables zu klein sein könnte. Allerdings ist die compilierte Version mit Symbolen auch nur 6 kB groß. Dann müsste beim Compilieren etwas schief laufen.

fda77, 2020-08-07:
Wäre ganz gut, dann muss es nicht jeder der es braucht für sich zu machen.
Aber so dass die Module nur sichtbar sind wenn die Kernelversion es überhaupt unterstützt

forenuser, 2020-08-08:
Das ist dann etwas aufwendiger, als grace.ko für 7.20 verfügbar zu machen. Zunächst möchte ich meine Box wieder so einsetzen können, wie mit der bisherigen 7.12 FW.

Aktuelle Probleme:
1. Nfsd: Von Debian 6 aus voller Zugriff möglich, von Mediaplayer (CoreElec) sind die Nfs-Shares sichtbar, der Zugriff darauf funktioniert leider (noch) nicht
2. Python/PyLoad: Python funktioniert nicht
3. Users/Group-Konfiguration: Verschiedene Benutzer mit gleicher ID
```
root@fritz:/var/mod/root# cat /etc/passwd
asec::101:101::/nonexistent:/noshell
bittorrent:x:101:1:bittorrent:/home/bittorrent:/bin/false
ftp:x:102:1:ftp:/home/ftp:/bin/false
nobody:x:100:1000:nobody:/home/nobody:/bin/false
root:x:0:0:root:/mod/root:/bin/sh
smb::100:100::/nonexistent:/noshell
ftpuser:x:103:1:ftpuser:/var/media/ftp:/bin/false
```
```
root@fritz:/var/mod/root# cat /etc/group
asec::101:asec,smb
console::2:asec,smb
nas::3:asec
nobody:x:1000:nobody
root:x:0:
smb::100:smb
users:x:1:bittorrent,ftp,ftpuser
watchdog::1:smb,asec
```

forenuser, 2020-08-08:
"Aber so dass die Module nur sichtbar sind wenn die Kernelversion es überhaupt unterstützt."
--> Das scheint meine Kompetenzen zu übersteigen, weil fuse.ko über mehrere Pakete (fuse, davfs, ntfs) aktiviert wird, aber z.B. in FW 7.12 nicht nötig ist. Ich weiß auch nicht, ob eine Abhängigkeit von FREETZ_KERNEL_VERSION_4_9_MIN richtig wäre.

forenuser, 2020-08-08:
Mit einem neueren Freetz Stand läuft die Downgrade Installation von 7.20 auf 7.12 leider auch nicht durch.

[image]

fda77, 2020-08-08:
> bei FW 7.12 nicht nötig ist
Es hat sich einiges bei AVM geändert.
> Ich weiß auch nicht, ob eine Abhängigkeit von FREETZ_KERNEL_VERSION_4_9_MIN richtig wäre.
Ich auch nicht, ohne nachzuschauen

Kann sein dass das Downgrade nicht funktioniert da sich auch die Kernelversion von 7.1  zu 7.2 änderte. Das ist dann halt so.
Kann aber auch sein dass es an den DEB Scripten liegt (erkennt man am Skin). Die überschreiben Teile von Freetz und sind nicht auf das ganze neue Zeug von AVM vorbereitet, zb systemd

forenuser, 2020-08-08:
Ich habe die Anpassungen für die Konfiguration gemacht, kann aber nicht sagen, ob make für alle Boxen noch funktioniert. Andererseits funktioniert es ohnehin nicht, da Kernelmodule fehlen. 7590 7.20/7.12 make funktioniert nun. Insgesamt musste ich 5 Dateien anpassen.

Ich habe über den Bootloader wieder auf 7.12 geflasht. Python geht damit mit der gleichen Source Version. Jetzt baue ich 7.20 komplett neu. Mal sehen, ob Python dann funktioniert.

fda77, 2020-08-08:
Wäre schon gut wenn es für alle und nicht nur 1 Box passen würde.
Mit der gleichen Source Version? Obwohl 7590 7.1 Kernel v3 und 7.2 Kernel v4 nutzt? Dazu noch systemd, gcc, libc,...

forenuser, 2020-08-09:
> Mit der gleichen Source Version? Obwohl 7590 7.1 Kernel v3 und 7.2 Kernel v4 nutzt? Dazu noch systemd, gcc, libc,...
Ich meinte eher, dass die Python Sourcen die selben sind und konnte damit ausschließen, dass gawk5.0.1 die Funktion von Python beeinflussen könnte.

Den NFS-Server habe ich mit folgendem Ergebnis gestestet:
- NFS-v2 funktioniert, v3/v4 funktionieren nicht: Debian 6 verwendet diese Version, deshalb hat nfs-mount dort funktioniert.
```
root@freetz-linux:/tmp# mount -t nfs -o rw,nolock,vers=4 fritz.box:/var/media/ftp/data nfs
mount.nfs: Protocol not supported
root@freetz-linux:/tmp# mount -t nfs -o rw,nolock,vers=3 fritz.box:/var/media/ftp/data nfs
mount.nfs: Protocol not supported
root@freetz-linux:/tmp# mount -t nfs -o rw,nolock,vers=2 fritz.box:/var/media/ftp/data nfs
root@freetz-linux:/tmp#
```

fda77, 2020-08-09:
Könnte daran liegen dass die nfs-utils in freetz uralt sind. Verwendet wird 1.3.4, aktuell ist 2.5.1: http://www.linuxfromscratch.org/blfs/view/svn/basicnet/nfs-utils.html
Wenn man die Version aktualisiert, sollte man noch schauen dass für die alten Kernel v2 nichts kaputt geht
NFSv4 wird übrigens deaktiviert: https://github.com/Freetz-NG/freetz-ng/blob/master/make/nfs-utils/nfs-utils.mk#L27

forenuser, 2020-08-09:
NFSv4 habe ich im Kernel nicht aktiviert, weil es zu viele Optionen hat, mit denen ich mich nicht auskenne. Nach Aktivieren von NFSv3 im Kernel für den Server funktioniert NFS-Server bestens. Nur das nfs Module für die Clientfunktion wird nicht automatisch geladen. Nach insmod nfs.ko funktioniert auch der NFS Client. Eine Aktualisierung von nfs-utils ist aus derzeitiger Sicht nicht erforderlich.

Gibt es eigentlich Möglichkeiten zum Debuggen (Remote, Target oder Simulation)? Damit könnte man die Ursache des Python Problems sicher schneller finden.

fda77, 2020-08-09:
Weshalb sollte der NFS Server das NFS Client Module laden? 
Ich glaube nicht dass in der 4KB grossen Python "Binary" viel an Debugcode eingebaut ist. Würde auch keinen Sinn machen, da auf der Fritzbox nicht unendlich Platz ist!

forenuser, 2020-08-09:
Die Größe ist nicht das Problem, da die Größe im 7.12er Stand ähnlich ist. Ich hatte gehofft, Im Python Source Verzeichnis liegen die ungestrippten Python Dateien ;-) Wie würdest Du in diesem Fall vorgehen, um die Ursache des Fehlers zu finden?

Ich habe versucht, ein Oscam Binary zu bauen. Dieses beendet sich ohne jegliche Meldung gleich nach dem Start. Auch dafür wäre ein Debugging vielleicht hilfreich.

fda77, 2020-08-09:
Man kann mit "strace" bei jeder Binary schauen was diese so treibt. Nützlich kann auch "ldd" sein.
Was wird denn nun aus den Kernelmodulen?

forenuser, 2020-08-10:
python
```
root@fritz:/usr/bin# strace ./python2.7.bin
execve("./python2.7.bin", ["./python2.7.bin"], 0x7fb6d660 /* 302 vars */) = -1 ENOENT (No such file or directory)
strace: exec: No such file or directory
+++ exited with 1 +++
```
oscam
```
root@fritz:/var/media/ftp/data/addon/oscam# strace ./oscam
execve("./oscam", ["./oscam"], 0x7f8cd4c0 /* 302 vars */) = 0
open("/proc/self/exe", O_RDONLY)        = 3
mmap(NULL, 768852, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7745d000
mmap(0x7745d000, 768080, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x7745d000
cacheflush(0x775181a8, 0x9ac, 0x3)      = 0
mprotect(0x77517000, 6996, PROT_READ|PROT_EXEC) = 0
readlink("/proc/self/exe", "/var/media/ftp/data/addon/oscam/"..., 4095) = 37
cacheflush(0x7fa82620, 0x134, 0x3)      = 0
mmap(0x400000, 2121728, PROT_NONE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x400000
mmap(0x400000, 1720176, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x400000
cacheflush(0x400000, 0x134, 0x3)        = 0
cacheflush(0x400134, 0x1a3e3c, 0x3)     = 0
mprotect(0x400000, 1720176, PROT_READ|PROT_EXEC) = 0
mmap(0x5b3000, 14752, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x5b3000
cacheflush(0x5b3f70, 0x2a30, 0x3)       = 0
mprotect(0x5b3000, 14752, PROT_READ|PROT_WRITE) = 0
mmap(0x5b7000, 323496, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x5b7000
brk(0x606000)                           = 0xcd3000
open("/lib/ld-uClibc.so.0", O_RDONLY)   = -1 ENOENT (No such file or directory)
exit(127)                               = ?
```

closed 2020-08-10 by fda77

fda77, 2020-08-10:
Dein ursprüngliches Problem ```7590 FW 7.12: Make bricht ab, weil einige Kernelmodule nicht gefunden werden``` ist gelöst, für dich zumindest.