Freetz-NG/freetz-ng #40: 7580: BlueZ not woring

JakobTewes, 2020-08-06:
For bluetooth beaconing I need BlueZ. Seems that there are no kernel sources available for this box. But this should not be a problem as I need kernel modules only. Unfortunately the build fails with Error

`WARNING: File of kernel module not found: bnep.ko
WARNING: File of kernel module not found: l2cap.ko
ERROR: Enable missing with 'make kernel-menuconfig' as (M)odule (and push a pull request).
Makefile:275: recipe for target 'firmware-nocompile' failed
make: *** [firmware-nocompile] Error 1`

Can I fix this? Ideas?

fda77, 2020-08-06:
Yes, enable missing with 'make kernel-menuconfig' as (M)odule (and push a pull request). ;-)
Same as https://github.com/Freetz-NG/freetz-ng/issues/38

JakobTewes, 2020-08-07:
@fda77 : Thanks - found the bnep module and enabled it. Unfortunately I´m unable to locate "l2cap".
U got a hint on how to identify the corresponding kernel-config option?

fda77, 2020-08-07:
In menuconfig you could search with "/"

fda77, 2020-08-13:
-> https://freetz-ng.github.io/freetz-ng/SUPPORT

closed 2020-08-13 by fda77