Freetz-NG/freetz-ng #41: Update PCSC-Lite 1.8.24 zu PCSC-Lite 1.9.0 und libusb  zu libusb 1.0.23

MasterRoCcO, 2020-08-09:
Hi kannst du das PCSC-Lite 1.8.24 zu PCSC-Lite 1.9.0 Updaten
https://pcsclite.apdu.fr/files/pcsc-lite-1.9.0.tar.bz2

libusb 1.0.23
https://github.com/libusb/libusb/releases/tag/v1.0.23
Denn wenn man mit S3 nun ein oscam baut und das nun auf der fritzbox nutzen will geht dieses nicht weil was fehlt in denn alten sachen die in freetz sind

fda77, 2020-08-09:
Kann ich wahrscheinlich, mir ist es aber lieber wenn du es selbst versuchst!
Schau mal hier https://freetz-ng.github.io/freetz-ng/SUPPORT Dort "Package bump" und "Pull request". Nach Möglichkeit nur 1 Commit pro Packache bzw Library!
Ich glaub CCID wurde immer gleichzeitig mit PCSC erhöht, könnte deshlab auch nötig sein

MasterRoCcO, 2020-08-09:
das habe ich noch nie gemacht. Werde es aber mal versuchen die Tage

forenuser, 2020-08-10:
https://github.com/Freetz-NG/freetz-ng/issues/44 könnte vielleicht auch hierbei relevant sein.

forenuser, 2020-08-11:
Wer weiß, ob die Aktualisierung überhaupt nötig ist.

- **Ich sage Adios und viel Erfolg mit Deinem Project. Versuche mal runterzukommen von Deinem hohen Ross!**

fda77, 2020-08-13:
-> https://github.com/Freetz-NG/freetz-ng/pull/47

closed 2020-08-13 by fda77