Freetz-NG/freetz-ng #42: 7590 7.20: Python/Pyload funktioniert nicht

forenuser, 2020-08-10:
Python bricht beim Aufruf mit folgender Fehlermeldung ab:
```
root@fritz:/usr/bin# ./python
./python: exec: line 9: /usr/bin/python2.7.bin: not found
```
strace:
```
root@fritz:/usr/bin# strace ./python2.7.bin
execve("./python2.7.bin", ["./python2.7.bin"], 0x7fb6d660 /* 302 vars */) = -1 ENOENT (No such file or directory)
strace: exec: No such file or directory
+++ exited with 1 +++
```

fda77, 2020-08-10:
Ich sehe
```
/usr/bin/python2.7.bin: not found
 ENOENT (No such file or directory)
strace: exec: No such file or directory
```
Dh es gibt diese evtl nicht

fda77, 2020-08-10:
Aktuell ist Python 2.7.18. Bevor man einen Fehler sucht der schon behoben sein könnte, sollte man es mit der neueren Version probieren

forenuser, 2020-08-10:
Uns ist aufgefallen, dass im Terminal LD_LIBRARY_PATH wie folgt auf der Fritzbox gesetzt ist:
LD_LIBRARY_PATH='/mod/lib:/mod/usr/lib'
dort aber keine Dateien zu finden sind.

Könnte die Umgebungsvariable möglicherweise fehlerhaft sein? Wenn ja, wie kann man ändern, dass von freetz die richtigen Pfade eintragen werden?

forenuser, 2020-08-10:
> Ich sehe
> 
> ```
> /usr/bin/python2.7.bin: not found
>  ENOENT (No such file or directory)
> strace: exec: No such file or directory
> ```
> 
> Dh es gibt diese evtl nicht

Wird die nicht gerade ausgeführt? Ich war auch im Verzeichnis, wo sich diese Datei befindet. Damit erschließt sich mir die Meldung in zweifacher Hinsicht nicht, zumal es mit 7.12 läuft. Kann natürlich sein, dass eine neue Version hilft - who knows?

fda77, 2020-08-11:
Ich kenne die Ursache, ob ich es beheben kann weiss ich nicht.
Das schauen wir dann nach deinem Versionsbump an

forenuser, 2020-08-11:
Adios!

forenuser, 2020-08-11:
Obsolete