Freetz-NG/freetz-ng #43: 7590 7.20/7.12: NFS Server mit aktueller Trunk Version nicht verwendbar

forenuser, 2020-08-10:
Fehleranalyse wurde im geschlossenen Ticket gemacht
7.12: Make bricht ab, weil zu viele Kernelmodule im menuconfig enthalten sind
7.20: Make bricht ab, weil zu wenige Kernelmodule im menuconfig enthalten sind

Lokale Änderungen für 7590 7.12/7.20 kann ich bei Bedarf zur Verfügung stellen.

fda77, 2020-08-10:
Du meinst das Ticket mit Oscam?
Man kann da auch noch was schreiben wenn es zu ist.

```Enable missing with 'make kernel-menuconfig' as (M)odule (and push a pull request).```
Solange dies niemand macht, braucht man nicht weiterzuschauen
Sieh https://github.com/Freetz-NG/freetz-ng/issues/40

forenuser, 2020-08-11:
Die Voraussetzung, mich hier zu engagieren, hast Du wirksam verhindert.

- **Ich sage adios und viel Erfolg mit Deinem Project. Versuche mal runterzukommen von Deinem hohen Ross!**

forenuser, 2020-08-11:
Obsolete