Freetz-NG/freetz-ng #44: 7590 7.20: LD_LIBRARY_PATH

forenuser, 2020-08-10:
Uns ist aufgefallen, dass im Terminal LD_LIBRARY_PATH wie folgt auf der Fritzbox gesetzt ist:
LD_LIBRARY_PATH='/mod/lib:/mod/usr/lib'
dort aber keine Dateien zu finden sind.

Könnte die Umgebungsvariable möglicherweise fehlerhaft sein? Wenn ja, wie kann man ändern, dass von freetz die richtigen Pfade eintragen werden?

fda77, 2020-08-11:
Wie soll die Erkenntnis bei einem Bump helfen?
Das ist so okay und kann zum Testen verwendet werden.
Schau mal ins alte Wiki: Development, Compilierung, FAQ oder so, müsste dort beschrieben sein

closed 2020-08-11 by fda77

forenuser, 2020-08-11:
Du machst es Anwendern unnötig schwer, bzw. vergrault sie. Dein Kommentar ist für uns wenig hilfreich. Vielleicht gibt es einen Grund, warum Du beim IPPF und DEB rausgeflogen bist. Ich finde es sehr schade, weil wir in der Vergangenheit gut zusammengearbeitet haben.

fda77, 2020-08-11:
Was gefällt dir denn an der Erklärung auf die ich verwiesen hab nicht? Oder warst du nur zu faul zum suchen?
Du hast einen Bug gemeldet -> ist keiner -> gibt nichts zu tun.
Wenn du dein Oscam nicht compiliert bekommst, kann ich dir nicht helfen
Ich hatte keine Lust mehr aufs DEB, da dort so gut wie nur "Image"-Nutzer zu finden sind. Und selbst die Freetz-"Entwickler" dort können nur das menuconfig aufrufen sind sind nicht im geringsten im Stande eine Fehlermeldung zu beheben oder gar wiederzugeben

forenuser, 2020-08-11:
In diesem Fall wäre es vielleicht sinnvoll, wenn Du Dein Wissen weitergibst und Benutzer motivierst, sich mit freetz-NG zu beschäftigen und sie unterstützt um Wissen und Fähigkeiten aufzubauen und ggf. unterschiedliche Sichtweisen aufklärst. Jetzt soll ich wissen, was Du mit altem Wiki meinst und wo das zu finden ist? Ich habe 3 NAS Geräte mit Entware, mit denen ich meinen Anwendungsfall auf einfache Art und Weise in kurzer Zeit abdecken kann. Warum soll ich mich nun mit einer für mich neuen Materie beschäftigen?

Wie erfolgreich wäre ein Project, wenn der Projektleiter oder Senior Engineer so agiert, wie Du?

fda77, 2020-08-11:
Ich benutze LD_LIBRARY_PATH nicht, weshalb sollte ich für dich die Beschreibung aus dem Wiki heraussuchen?
Du hättest genau den gleichen Aufwand wie ich!

Ich hab deine Spam-Posts gelöschst, falls du das nochmal machst sperre ich dich hier

forenuser, 2020-08-11:
Hast doch einen Request von mir erhalten. Alles weitere Engagement hast Du wirksam zunichte gemacht. Du hast meine Erwartung total enttäuscht.

fda77, 2020-08-11:
> Ich hab deine Spam-Posts gelöscht, falls du das nochmal machst, sperre ich dich hier

Okay, du wolltest es so.
Ich habe deine weiteren Spam gelöscht
Viel Spass im DEB