Freetz-NG/freetz-ng #46: 7590 7.12: Doppelte Benutzer IDs in /etc/passwd

forenuser, 2020-08-11:
In /etc/passwd haben unterschiedliche Benutzer gleiche IDs.

/etc/passwd
```
root@fritz:/var/mod/root# cat /etc/passwd
asec::101:101::/nonexistent:/noshell
bittorrent:x:101:1:bittorrent:/home/bittorrent:/bin/false
ftp:x:102:1:ftp:/home/ftp:/bin/false
nobody:x:100:1000:nobody:/home/nobody:/bin/false
root:x:0:0:root:/mod/root:/bin/sh
smb::100:100::/nonexistent:/noshell
ftpuser:x:103:1:ftpuser:/var/media/ftp:/bin/false
```
100: smb und nobody
101: asec und bittorrent

Sieht möglicherweise danach aus, dass freetz eigene Benutzer mit gleichen IDs wie AVM Nutzer anlegt, bin mir aber nicht sicher. Könnte das Auswirkungen auf die Benutzersicherheit haben? Wie könnte man das beheben?

fda77, 2020-08-11:
Die Benutzer von AVM sind neu und werden mit festen IDs angelegt, egal was es schon gibt.
Um das Problem zu beheben mach einen Freetz-Werksreset und konfigurier alles neu...
Oder pass die IDs an wie es dir gefällt, danach auch noch die Rechte an den Verzeichnissen entsprechend anpassen!
Man könnte eine Warnmeldung bei gleichen IDs einbauen, die sind aber grundsätzlich erlaubt und evtl gewünscht.
Das ganze betrifft auch die GIDs: https://freetz-ng.github.io/freetz-ng/NEWS.html 2020-01-31
Ich wüsste nicht was man da noch machen sollte
PS: Im alten Freetz/ werden diese AVM-Benutzer komplett gelöscht, funktioniert auch. Hat sich noch niemand beschwert ^^

forenuser, 2020-08-11:
Die Benutzer IDs BitTorrent und Nobody hatte ich bereits manuell geändert und mit modsave Clash gespeichert. Nach einem Neustart oder Flaschen eines neuen Images waren diese Änderungen weg.

fda77, 2020-08-11:
Dann hast du falsch gespeichert!
Wie es richtig geht, steht in der FAQ

closed 2020-08-11 by fda77