Freetz-NG/freetz-ng #48: 1.32.0 - EXPERIMENTAL und 7590 fw 7.20

MasterRoCcO, 2020-08-13:
Hi ich habe gestern mal freetz neu aus gecheckt und habe BUSYBOX 1.32 benutzt. 

Da mit geht die :81 seite nicht mehr

fda77, 2020-08-13:
Bei mir läufts, ich hab die neue Busybox auf allen Geräten, auch 7590.
Jeder Busybox die auf .0 endet ist offiziell "unstable": https://www.busybox.net/

MasterRoCcO, 2020-08-13:
So wirklich nur 1.32.0 - EXPERIMENTAL im image und nein es läuft nicht.

fda77, 2020-08-13:
Keine Ahnung, bei mir rennts auf mehreren Geräten ohne Probleme
Vielleicht hast du nur die falsche IP eingegeben, oder ne alte config auf der Fritzbox oder Altlasten in der .config, oder oder...

MasterRoCcO, 2020-08-13:
so die avm seite lauft http://192.168.178.3/
und nein http://192.168.178.3:81 nicht. da kann ich doch nicht die falsche ip ein geben. mit der 1.31 geht es

MasterRoCcO, 2020-08-13:
mit der 7490 geht es auch. nur nicht mit der 7590 fw 7.20

fda77, 2020-08-13:
Komisch, hab nochmal probier. Lüppt:
```
$ busybox |head -n1
BusyBox v1.32.0 (2020-08-04 23:17:24 CEST) multi-call binary.

$ wget http://192.168.178.1:81/ -O -
--2020-08-13 17:42:25--  http://192.168.178.1:81/
Connecting to 192.168.178.1:81... connected.
HTTP request sent, awaiting response... 401 Unauthorized

Username/Password Authentication Failed.
```

berndy2001, 2020-08-13:
Bei mir passt auch alles.
`root@fritz:/var/mod/root# busybox |head -n1
BusyBox v1.32.0 (2020-08-13 23:58:47 CEST) multi-call binary.
root@fritz:/var/mod/root# curl http://192.168.1.1:81
<HTML><HEAD><TITLE>401 Unauthorized</TITLE></HEAD>
<BODY><H1>401 Unauthorized</H1>

</BODY></HTML>
`
Mach mal einen frischen Checkout mit minimal Image und lösche vorher alle Einstellungen auf der Box.

MasterRoCcO, 2020-08-14:
root@7590-client:/var/mod/root# busybox |head -n1
BusyBox v1.32.0 (2020-08-14 05:44:52 CEST) multi-call binary.
root@7590-client:/var/mod/root# wget http://192.168.178.1:81/ -O -
Connecting to 192.168.178.1:81 (192.168.178.1:81)
wget: server returned error: HTTP/1.0 401 Unauthorized
root@7590-client:/var/mod/root#

so ich lösche nun mal die Einstellungen in der box und flsche dann noch mal

MasterRoCcO, 2020-08-14:
es geht immer noch nicht nach denn löschen der Einstellungen und neuen flashen

root@fritz:/var/mod/root# busybox |head -n1
BusyBox v1.32.0 (2020-08-14 05:44:52 CEST) multi-call binary.
root@fritz:/var/mod/root# wget http://192.168.178.1:81/ -O -
Connecting to 192.168.178.1:81 (192.168.178.1:81)
wget: can't connect to remote host (192.168.178.1): Connection refused
root@fritz:/var/mod/root#

fda77, 2020-08-14:
Hab die Ursache für deine Problem gefunden: Die schlecht geschriebenen Addons der "Digitalen Elite"
Schau dir mal im DEB um, dort gibt es Berichte, dass
 - manche Seiten nicht gespeichert werden können
 - dass manche Einstellungen doppelt gespeichert werden

Bei der Installation der Addons werden Dateien von Freetz ohne jegliche Prüfung überschrieben. Dass das irgendwann Probleme verursacht war klar. Wahrscheinlich hast du damit deine config auf der Box so geschrottet dass die Probleme auch bleiben wenn du ein Image ohne Addons installierst.
Evtl gab es auch Experimente weil das mit dem compilieren nicht funktioniert
Mach am besten Recovery, lass die Addons weg und spiel kein Backup zurück!

Um ganz ehrlich zu sein hab ich von dem Unsinn aus dem DEB wirklich genug, nachdem @forenuser hier zuerst mehrfach herumgespamt, gesperrt wurde, und danach seine ganzen Postings löschte

Da ich mit den Addons nichts zu tun habe ->

closed 2020-08-14 by fda77

berndy2001, 2020-08-14:
Deshalb schrieb ich auch "minimalimage" - aber es ist ja einfacher andere damit zu beschäftigen als den Fehler bei sich zu suchen.

MasterRoCcO, 2020-08-14:
ok ich teste es mal. ich selber habe kein addon in meinen image. Aber ich habe mal ein image aus den deb getestet. und wenn das mein config geschossen hat ist das ja klar