Freetz-NG/freetz-ng #49: Dynamischen Linker zum Start der Programme fehlen auf der 7590 7.20

MasterRoCcO, 2020-08-17:
Da ja AVM für die 7590 FW 7.20 musl 1.1.23 benutzt und freetz uClibc 1.0.15 fehlen dynamischen Linker zum Start der Programme

AVM-Binaries verwenden /lib/ld-musl-mips-sf.so.1 
Funktionierende Freetz-Binaries verwenden /usr/lib/freetz/ld-uClibc.so.1
Nicht funktionierende Freetz-Binaries verwenden /lib/ld-uClibc.so.0 (und diese Datei gibt es nicht)

Daher laufen ganz viele Sachen nicht der Box.
Ich habe mir zwar meine dynamischen Linker selber gebaut und es lauft alles. Nur ich glaube nicht jeder ist dazu in der lage

Vielleicht kann ja einer musl 1.1.23 in freetz-ng einbauen.

fda77, 2020-08-17:
Nein, kann ich auch nicht, wäre auch überflüssig
Wenn eine von **dir** compilierte Binary Dinge auf der Box nutzt die es nicht gibt (siehe SB), hast du 2 Möglichkeiten:
 - Das was du nutzen willst zusätzlich einbauen
 - Das was vorhanden ist nutzen
Optimal wäre ein Package make/xxxx/ zu erstellen, dann wird die Toolchain von Freetz mit den passenden Einstellungen verwendet.

Diese Diskussion hatte ich übrigens mit rolu2 von den DEBen auch schon, es ging da um lcd4linux. Ich hatte es in Freetz eingebaut, damit ihr nicht immer mit diesen vorcompilierten Binaries herumfummelt. Es gibt verschieden Architekturen, Endianess, libcs und verschieden Versionen davon, und für jede Kombination müsste man eine Binary erstellen.
rolu2 fand trotzdem seine (von bpanther erstellen) binaries (ohne Argumente) "besser".

Aber wie schon gesagt habe ich nach der Aktion hier von @forenuser wirklich genug von den DEB-Problemen!
Ich hab meinen Account nicht ohne Grund löschen lassen

closed 2020-08-17 by fda77

WXbet, 2020-10-18:
Beim Kompilieren von Binaries folgendes LDFLAG setzen
`-Wl,-dynamic-linker,/usr/lib/freetz/ld-uClibc.so.0`
Dann lädt das Binary (s)einen Freetz-ulibc-Interpreter...

fda77, 2020-10-18:
Das steht doch schon in ```TARGET_CFLAGS_LD```, falls es nötig ist

WXbet, 2020-10-18:
Gut zu wissen. Wie kann man das nutzen, wenn man auf Binaries mit einem Freetz-NG Cross Toolchain kompiliert?

fda77, 2020-10-18:
Wie, gut zu wissen? Da hast du es doch bestimmt her, oder? Die Variable nutzt man mit $(TARGET_CFLAGS_LD), wird aber automatisch verwendet falls nötig

WXbet, 2020-10-18:
Nein, da hatte ich es nicht her, sondern mich bzgl. des oscam-Pakets damit beschäftigt, wie man ein lauffähiges Binary erzeugt, was auf der 7590 FOS 7.21 startet, ohne das Vorhandensein seines Interpreters im /lib-Verzeichnis... Dabei bin ich dann darauf gestoßen, das man den Interpreter via LDFLAG setzen kann.

fda77, 2020-10-18:
Ist halt blöd wenn man im DEB keine Quellen angibt...
Mach doch (für dich) ein eigenes Package make/PKG/, dann musst du dich um sowas nicht kümmern

WXbet, 2020-10-18:
<Ist halt blöd wenn man im DEB keine Quellen angibt...>
Verstehe ich nicht, wie meinst du das?

fda77, 2020-10-18:
Weil du dann gewusst hättest dass die Parameter aus "TARGET_CFLAGS_LD" stammen

WXbet, 2020-10-20:
Noch ein, zwei Fragen dazu, warum werden bei manchen Firmwareversion die ulibc-Dateien unter` /usr/lib/freetz` gespeichert (`FREETZ_AVM_PROP_UCLIBC_SEPARATE=y`) und bei machen nicht (`FREETZ_AVM_PROP_UCLIBC_XLOCALE=y`)?
 z.B.
https://github.com/Freetz-NG/freetz-ng/blob/master/config/.img/7590--DE--07_1X.in = FREETZ_AVM_PROP_UCLIBC_XLOCALE
https://github.com/Freetz-NG/freetz-ng/blob/master/config/.img/7590--07_2X.in = FREETZ_AVM_PROP_UCLIBC_SEPARATE

Könnte man die ulibc-Dateien nicht generell unter `/usr/lib/freetz` erzeugen oder alternativ Symlinks setzen, wenn AVM-libc und Freetz-Libc identisch sind? Das würde es erleichtern, universellere Binaries für freetz zu generieren, die immer ihren Interpreter unter `/usr/lib/freetz` benutzen.

fda77, 2020-10-20:
Ihr immer mit euren blöden Binaries... Mach doch einfach wie oben schon geschrieben ein eigenes Package make/irgendwas/. Dann wird immer automatisch die passende lib genommen und alles ist direkt im Image

Theoretisch möglich, praktisch blöd:
 - es gibt keine dl-Toolchains dafür, dh je box 5gb hdd Platz und ne Menge Zeit
 - ältere Boxen haben nur 8-16 mb flash, das wird mit 1mb zusätzlichen libs eng
 - für ganz alte mit uclib vor v1 fehlen möglicherweise Patches
 - bringt nichts bei Versionswechsel, dann fängt der ganze Spass von vorne an

WXbet, 2020-10-20:
Ich frage zum wiederholten mal nett und bekomme von dir Antworten mir dem Wort "blöd" darin....
Punkt 1 verstehe ich nicht.
Punkt 2 - 4 würde ja ein Symlink auf die ulibc-Files im /lib -Verzeichnis abdecken.

fda77, 2020-10-20:
Aber? Ist doch blöd. Weshalb steht ja danach in den 4 Zeilen.
Du kennst schon den Unterschied von "du bist blöd" zu "X ist blöd" (irgendwas ausser "du")?
1) There is no dl-toolchain available, so you need 5gb hdd space and much time
2) Nein, man kann nicht alles compilieren was davon abhängt