Freetz-NG/freetz-ng #58: 6590 7.20: Reboot durch AVM-Watchdog

MasterRoCcO, 2020-08-28:
Bei der 6590 sollte man auch erst mal 
FREETZ_PATCH_DISABLE_AVM_WATCHDOG=y
an machen

fda77, 2020-08-29:
Ich bin nicht sicher ob man das wirklich braucht. Ich warte mal noch auf weitere Meldungen
Es kann genau so gut sein, dass euer Addon die Ursache ist, da ist ua für spontane Reboots bekannt: https://freetz-ng.github.io/freetz-ng/ADDONS Zusammen mit falsch compilierten Binaries ist das gut möglich

MasterRoCcO, 2020-08-29:
No addon im image und ca nach 2min gibt es ein reboot. Wie bei der 7590. Super fehler werden gemeldet lol

⁣BlueMail for Android herunterladen ​

Conan179, 2020-08-30:
Ich bin gerade dabei ein minimal image für meine 6590 zu erstellen.
Das werde ich dann Flashen und beibachten.
(Die source Datein für die 7.20 FW der 6590 sind auf dem AVM server)

fda77, 2020-08-30:
@MasterRoCcO 
Super Lügner gefunden, doppel lol lol
Da stehts doch: https://www.digital-eliteboard.com/threads/492329/
Nur weil kein Addon im WEBIF zu sehen ist, heisst nicht dass es nicht im Image ist. Wahrscheinlich fehlen für das registrieren der Seiten irgendwelche Berechtigungen, das scheint ihr aktuell nicht im Griff zu haben... Der Watchdog kram kann dann trotzdem in der rccustom sein!
@Conan179 Danke! Schau am besten in Syslog (forward auf anderes Device) und die Crashlogs warum es crasht, falls es crasht

Conan179, 2020-08-30:
Ich (kudo_78) hatte das mit den source datein, auch im digital-eliteboard.com per PN geschickt, hab aber zuspät gesehen, das deinn Acc dort nicht mehr gibt.
Es dauert aber noch ein bischen, bin noch am bauen und dann muss ic hes noch auf die box bringen, sie ist nicht direkt am PC angeschlossen.

Prisrak1, 2020-08-30:
bei mir laufen 64/6590 sabil, erst aber mit Disable AVM watchdog für die 6490. Bei der 6590 muß es noch prüfen., meine ist nicht nötig.

fda77, 2020-08-30:
Man kann Syslog via normalem IP Netzwerk weiterleiten, das hat den Vorteil dass man so auch die letzten Nachrichten vor einem Reboot bekommt. Falls die Fritzbox USB hat und der Stick schon erkannt ist kann man auch dorthin loggen (könnte aber sein dass die letzten Zeilen nicht mehr geschrieben werden)
Das disable-watchdog ist nur ein Workaround! Jemand müsste noch untersuchen warum die AVM-Daemons meinen dass es ein Problem gibt und ein Reboot nötig ist...

Conan179, 2020-08-30:
Wie leitet man den log um?

fda77, 2020-08-30:
Checkbox "Network logger" anmachen, Hostname bei "Host" und Port bei "Port" eintragen: https://freetz-ng.github.io/freetz-ng/make/syslogd-cgi.html

Conan179, 2020-08-31:
So minimal Image, läuft seid 9h ohne neustarts.
[image]

Prisrak1, 2020-08-31:
geprüft: mit Disable AVM watchdog für die 6490. Bei der 6590 ist nicht nötig. 
so laufen beide.

fda77, 2020-08-31:
Kannst du noch logs (syslog) posten? AVM sollte beim herunterfahren zeigen weshalb das gemacht wird
Kannst ja zuerst auf USB oder den internen speicher loggen, vielleicht klappt das

Prisrak1, 2020-09-01:
meist von der 6490, wenn diese ständig neustartet? sonst sowas ..
[log.txt](https://github.com/Freetz-NG/freetz-ng/files/5154572/log.txt)

fda77, 2020-09-01:
In der Log ist leider gar nichts zum Reboot zu sehen. Hast du beim Syslog "Kernel-Log-Dämon aktivieren" aktviert? Kann aber sein dass das nichts bringt, da AVM auf den Docsis Geräten vieles deaktiviert hat.
Ich hab übrigens vor kurzem von einer 7580 mit freetz gelesen die problemlos lief. Die nutze die gleich CPU und Kernel wie die 7590. Der disable-watchdog wird aber nur für die 7590 automatisch aktiviert.
Die 6490 und 6590 haben beide einen Puma6

fda77, 2020-09-01:
Und bitte auch unbeidngt ohne "addons" ausprobieren. Es reicht nicht nur das Addon nicht ins image zu packen, sondenr man muss auch (mindestens) rc.custom und crontab ausmisten. Dort werden ungefragt Dinge hineingeschrieben, was auf dem Weg die das addon es macht nicht okay ist!

Conan179, 2020-09-03:
Das wäre die 2te bestätigte.

Prisrak1, 2020-09-03:
habs gelesen. Bei mir ,wie geschrieben:  mit und ohne "Disable AVM watchdog" für die 6590" läuft beides super.

Conan179, 2020-09-04:
meine 6590 Läuft jetzt seid 12h mit dem Replace Kernel, in der Zeit hat sie keine Neustarts gemacht.
OHNE
FREETZ_PATCH_DISABLE_AVM_WATCHDOG=y

fda77, 2020-09-04:
Danke! Es gibt wohl kein Problem mit dem Watchdog bei der 6590

closed 2020-09-04 by fda77