Freetz-NG/freetz-ng #60: bekomme freetz-ng nicht auf die 6590 geflasht

Conan179, 2020-08-30:
Hallo, ich bräuchte mal hilfe beim Flashen von freetz-ng auf meiner 6590.
Ich bekomme immer diese meldung.
[image]
Auf 2 pcs versucht.

Ich starte "make push-firmware" ziehe den stecker der box, warte kurz und stecke ihn dann wieder rein.
Auf der box ist die 7.20 drauf.

forenuser, 2020-08-31:
Ich nehme die flash Skripte vom DEB Teamserver: [Freetz-Teamserver](https://freetz.digital-eliteboard.com/?dir=Teamserver/Freetz/Freetz-VM/Flashen/scripte). Damit braucht man kein in-memory Image.

In der virtuellen Maschine:
ifconfig <network device> down
ifconfig <network device> 192.168.178.11 up
Fritzbox einstecken und ca. 8s warten
./start.sh <Image Name>

Conan179, 2020-08-31:
Ich hab es heute nacht noch hin bekommen, "make push-firmware" lief ohne fehler durch.
Auch "/home/freetz/6590/tools/push_firmware /home/freetz/6590/images/latest.image"
oder
"/home/freetz/6590/tools/push_firmware  --lfs 9 /home/freetz/6590/images/latest.image"
Damit die fw in die Aktive nand bank geflasht wird.
"/home/freetz/6590/tools/push_firmware  --lfs 1 /home/freetz/6590/images/latest.image"
brachte auch nichts.
Mein Letzer Versuch war:
"/home/freetz/6590/tools/push_firmware  /home/freetz/6590/images/latest.image -lfs 0" 
da sah ich schon eine Änderung:
 * Arm-System mtd: 0
 * Arm-Kernel mtd: 1
 * x86-System mtd: 6
 * x86-Kernel mtd: 7

und dann ging es.

closed 2020-08-31 by Conan179

fda77, 2020-08-31:
Man kann unter Linux 1 Netzwerkkarte mehrere IPs zuweisen. Falls die IP nucht für den AVM-Bootloader passt zeibt puash-firmware Befehle an mit denen man die IP setzen kann.
Und bitte immer die komplette Ausgabe posten, am besten mit dem Aufruf und dessen Parameter

Also grundsätzlich keine Screenshots!
 - die IP der VM in der VM mit "ifconfig" ermitteln
 - auf dem (Windows) PC mittels PUTTY zu der IP verbinden
 - im PUTTY Fenster kopiert man durch markieren, einfügen mit context-klick!

ich hoffe mal dass der SSH server in der VM aktiviert ist. Login geht mit gleinem Benutzer/Passwort wie auf der console

Conan179, 2020-08-31:
Es ist ein bau image speziell für freetz.
Die IP und der login waren NICHT das problem!

Das problem war was mit den 2 Nand Banken meiner 6590.

mich hatte die meldung:
ftp: setsockopt (ignored): Permission denied
auf die ihre geführt. 
Aber auch der flash vorgang mit:
"/home/freetz/6590/tools/push_firmware /home/freetz/6590/images/latest.image -lfs 0"
mit dem es schlussendlich ging, hatte das ausgespruckt.

fda77, 2020-08-31:
"ftp: setsockopt (ignored): Permission denied" ist "normal" und kann (ignoriert) werden.
Das hat was mit FTP zu tun, entweder aktivem/passivem oder dem ausgehenden Port, nur root darf Ports unter 1024 öffnen

funktioniert es auch mit "lfs 1"? Wenn man mit dem Parameter die bootbank/partition vorgibt, wird nicht versucht die aktuelle zu ermitteln ... das steht dann aber weiter oben was nicht mehr auf dem Screenshot ist ;-)

Conan179, 2020-08-31:
Die sache mit der Ftp meldung ist, aber wo steht das? 

Die aktive bootbank zu nutzen "lfs 9" und "lfs 1" haben bei mir nicht geholfen, erst mit "lfs 0" hat.
Auf dem screenshot oben sieht man, das er immer nur die mtd: 11 mtd: 12 mtd: 13 und mtd: 14 zum flashen nutzt, egal ob ich es mit make push_firmware oder per "lfs 9" und "lfs 1" direkt mit dem script.
Aber mit "lfs 0" hat er dann die anderen mtds genommen.
ich hab sicher gestern abend an die 20 mal versucht freetz zu flashen.

ich hab immer glück mit geräten und hab die merkwürdigsten Fehler.

Conan179, 2020-08-31:
Es war für mich so:
freetz per make push-firmware geflasht. 
make push-firmware schrieb, anlage gefunden und schreibt das was oben im screenshot zu sehen ist.
Nach dem Neustart kam ich aber nicht auf freetz webinterface und sah keine änderung beim normalen webinterface.
Ergo dachte ich die Meldung zeigt nen Fehler.

fda77, 2020-09-08:
Hier wird von einer 6591 berichtet bei der wohl die lfs=0 partitionen defekt sind: https://www.ip-phone-forum.de/threads/303332/post-2385260

Conan179, 2020-09-08:
Bei mir war es die umschaltung, die wollte nicht mehr von alleine. Ich hab letzte woche einmal manuel gewechselt weil ich was mit cifs ausprobiert hatte, jetzt wechselt sie wieder fröhlich hin und her.

fda77, 2020-09-08:
Bei den docsis wird das LFS automatisch umgeschaltet wenn die aktuelle nicht bootet. Bisher ist das aber nur bei 6591/6660 aufgefallen. Bei deiner 6590 evtl auch?

Conan179, 2020-09-08:
ICh hab es leider nicht per screenshot gesichert -.-
Aber es war so, das er immer die nandbank1 genommen hat, beim flaschen der labor fws, offiziellen fw.
Mit ist nemlich aufgefallen, das in der nandbank immer noch das freetz vom meinen letzten versuch, im feburar, freetz auf die 6590 zubringen.