Freetz-NG/freetz-ng #61: 7590 7.20: Cifsmount funktioniert möglicherweise nicht

forenuser, 2020-08-31:
Leider bekomme ich die Meldung "Host is down", wenn ich versuche per einen Samba Share mit cifsmount anzubinden.

root@fritz:~$ mount.cifs  //127.0.0.1/fritz.nas /tmp/CIFS -o user=admin
Password for admin@//127.0.0.1/fritz.nas:
mount error(147): Host is down
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)

Ich nutze den AVM SMB Server und kann auf den Share mit meinem Windows 10 Rechner zugreifen.

Die Verwendung von vers=1.0 hat bringt keine Verbesserung.

fda77, 2020-08-31:
Ich würde Version "2.1" nutzen, das ist die ab Windows 7. Evtl auch neuere
Kannst du auch andere Server zugreifen? Evtl liegt es auch am "NQCS" das AVM nun statt Samba verwendet
Und kannst du mit cifs auf einem Linux PC auf den AVM Server zugreifen?

forenuser, 2020-08-31:
Andere Versionen als 1.0 werden als Übergabeparameter nicht akzeptiert (Invalid argument).
Mit freetz-linux-1.5.7 (ubuntu 20.04) kann ich auf den Server zugreifen.
```
root@freetz-linux:~# mount.cifs //fritz.box/fritz.nas ~/myCIFS -o user=admin

root@freetz-linux:~#mount
//fritz.box/fritz.nas on /root/myCIFS type cifs (rw,relatime,vers=3.1.1,cache=strict,username=admin,uid=0,noforceuid,gid=0,noforcegid,addr=192.168.1.1,file_mode=0755,dir_mode=0755,soft,nounix,serverino,mapposix,rsize=65536,wsize=65536,bsize=1048576,echo_interval=60,actimeo=1)
```

Conan179, 2020-08-31:
ich hab eine frage, warum versuchst du den nas der fritzbox in der fritzbox zu mounten?

forenuser, 2020-08-31:
Ich nutze die FRITZ.NAS SMB Freigabe, um die Funktion von cifsmount zu testen.

fda77, 2020-08-31:
Für smb version 2.1 scheint in der makefile nichts deaktiviert zu sein: https://github.com/Freetz-NG/freetz-ng/blob/master/make/cifsmount/cifsmount.mk
Gibt es im kernel-menuconfig noch Optionen dazu? 
Aktuell scheint wohl version 6.10 zu sein, die 6.6 aus freetz ist wohl aus 2016:  https://wiki.samba.org/index.php/LinuxCIFS_utils

Conan179, 2020-09-01:
[image]

fda77, 2020-09-01:
> Support legacy servers which use weaker LANMAN security

scheint man für "1.0" zu brauchen

> SMB2 and SMB3 network file system support

gibt es beim kernel der 7590. das dürfte dann für smb2 und 3 sein

einfach mal aktivieren und ausprobieren

Conan179, 2020-09-01:
Ich hab bei mir alles im cifs bereich makiert, geändert hat sich nichts, mir hat es nicht geholfen.

fda77, 2020-09-01:
War das modul dann auch geladen?

Conan179, 2020-09-01:
Ehrlich gesagt, ich hab keine schimmer.
Wie schaue ich das nach?

forenuser, 2020-09-01:
Mit lsmod kannst Du sehen, ob das Modul geladen wird.

Conan179, 2020-09-01:
root@fritz:/var/mod/root# lsmod
Module                  Size  Used by    Tainted: P
cifs                  225783  0
krtp                  209212  0
ehci_hcd               34174  0
ismddemux_v3          468931  0
ismdfw_manager          9684  0
ismdclock_recovery     25067  1 ismddemux_v3
ismdclock             224987  0
ismdcore             3311007  4 ismddemux_v3,ismdfw_manager,ismdclock_recovery,ismdclock
ioctl_module            2376  7 ismddemux_v3,ismdfw_manager,ismdclock_recovery,ismdclock,ismdcore
sec_kernel            129429  2 ismddemux_v3,ismdfw_manager
qca_ol                885998  0
system_utils_linux      1655  3 ismddemux_v3,ismdfw_manager,sec_kernel
intel_ce_pm            38785  3 ismddemux_v3,ismdcore,sec_kernel
clock_control          17473  3 ismdclock,sec_kernel,intel_ce_pm
iosf_wrap               1270  3 ismdclock,sec_kernel,intel_ce_pm
sven_linux             26728  6 ismddemux_v3,ismdclock_recovery,ismdclock,ismdcore,sec_kernel,clock_control
platform_config        12762 10 ismddemux_v3,ismdfw_manager,ismdclock_recovery,ismdclock,ismdcore,sec_kernel,system_utils_linux,intel_ce_pm,clock_control,sven_linux
idl_i2c                 2114  1 clock_control
umac                 1455487  2 qca_ol
idl_gpio_shim           1295  0
pal_linux               9280  6 ismddemux_v3,ismdclock,ismdcore,intel_ce_pm,clock_control,sven_linux
osal_linux             12782 10 ismddemux_v3,ismdfw_manager,ismdclock_recovery,ismdclock,ismdcore,sec_kernel,system_utils_linux,intel_ce_pm,clock_control,sven_linux
ath_spectral           34140  2 qca_ol
ath_dfs                63910  2 qca_ol,umac
qdf                    23717  4 qca_ol,umac,ath_spectral,ath_dfs
asf                     5627  4 qca_ol,umac,ath_spectral,ath_dfs
mem_manager             6526  2 qca_ol,umac
aae                    93028  3 qca_ol,umac,ath_dfs
dect_io                18674  0
avm_dect              272250  1 dect_io
capi_codec            367837  0
zram                    7664  1
lzo_compress            1722  1 zram
kdsldmod             1400470 16
isdn_fbox_fon5       1258286  6 krtp
pcmlink               390668  4 avm_dect,capi_codec,isdn_fbox_fon5
cchardev               14314  1 kdsldmod
cprocfsmod              7767  2 kdsldmod,cchardev
Piglet_noemif          38367  0
led_module              2647  6

forenuser, 2020-09-01:
Ein cifs Modul sehe ich nicht. Beim nächsten Mal bitte das Ergebnis von lsmod|grep cifs in einem Codetag posten, damit man es besser lesen kann.
In Deiner Kernelkonfiguration ist die Option für smb2 und smb3 scheinbar nicht vorhanden.

Conan179, 2020-09-01:
Das reine cifs modul ist geladen.
cifs 225783 0
Steht in zeile 3.

forenuser, 2020-09-01:
Ok, habe ich übersehen.

forenuser, 2020-09-01:
Nach Aktivierung der beiden von @fda77 empfohlenen Optionen wird das Modul cifs.ko nicht mehr geladen. Beim manuellen Laden erhalte ich eine Fehlermeldung:

root@fritz:/var/mod/root# insmod /lib/modules/4.9.198/kernel/fs/cifs/cifs.ko
insmod: can't insert '/lib/modules/4.9.198/kernel/fs/cifs/cifs.ko': unknown symbol in module, or unknown parameter

dmesg
`[  348.368984][0][module-mem] module 'cifs' reserved size 0 too small for demand size 389120 - need 389120 more (module_alloc_waste=-958464)
[  348.369027][0][module-mem] module memory exhausted.
[  348.374074][0]cifs: Unknown symbol key_put (err 0)
[  348.374155][0]cifs: Unknown symbol key_type_logon (err 0)
[  348.374653][0]cifs: Unknown symbol request_key (err 0)
[  348.400231][0][module-mem] module 'cifs' reserved size 0 too small for demand size 389120 - need 389120 more (module_alloc_waste=-1347584`

fda77, 2020-09-01:
Das ist möglich.
Kam nicht schon beim "make" eine Warnung wegen der Symbole?

Man könnte noch schauen ob diese Symbole nicht auch in einem Modul sind die man nachladen kann
```
grep key_type_logon source/kernel/ref-grx5-7590_07.20/linux-4.9/ -r
source/kernel/ref-grx5-7590_07.20/linux-4.9/include/keys/user-type.h:extern struct key_type key_type_logon;
source/kernel/ref-grx5-7590_07.20/linux-4.9/fs/crypto/keyinfo.c:        keyring_key = request_key(&key_type_logon, full_key_descriptor, NULL);
source/kernel/ref-grx5-7590_07.20/linux-4.9/fs/crypto/keyinfo.c:        if (keyring_key->type != &key_type_logon) {
source/kernel/ref-grx5-7590_07.20/linux-4.9/fs/cifs/connect.c:  key = request_key(&key_type_logon, desc, "");
source/kernel/ref-grx5-7590_07.20/linux-4.9/fs/cifs/connect.c:          key = request_key(&key_type_logon, desc, "");
source/kernel/ref-grx5-7590_07.20/linux-4.9/security/keys/internal.h:extern struct key_type key_type_logon;
source/kernel/ref-grx5-7590_07.20/linux-4.9/security/keys/user_defined.c:struct key_type key_type_logon = {
source/kernel/ref-grx5-7590_07.20/linux-4.9/security/keys/user_defined.c:EXPORT_SYMBOL_GPL(key_type_logon);
source/kernel/ref-grx5-7590_07.20/linux-4.9/security/keys/key.c:        list_add_tail(&key_type_logon.link, &key_types_list);
```
Die Option CONFIG_CIFS_SMB2 mach auch 
```
Selects: NLS [=y] && KEYS [=y] && FSCACHE [=m] && DNS_RESOLVER [=n]    
```

Das CONFIG_KEYS gibt es nicht als Modul, sondern nur direkt im Kernel :( Damit sollte man diese Option wohl nicht nutzen

forenuser, 2020-09-02:
Ich möchte gerne ausschließen, dass es zu Inkompatibilitäten zwischen dem AVM Kernel und den hinzugefügten Modulen kommt. Gibt es eine Möglichkeit, festzustellen, dass der mit Freetz gebaute Kernel mit dem AVM Kernel kompatibel ist?

fda77, 2020-09-02:
Es gibt die GPL und laut dieser muss AVM den Kernelquellcode mit der genutzen .config herausgeben. Wie sich schon ab und an gezeigt hat passen die Einstellungen in dieser .config aber nicht immer zu den auf der Box wirklich genutzten...
Da es für die 7590 noch kein replace kernel gibt ist die Gefahr relativ klein dass Crashes auftreten. Wenn du ein Modul laden willst das etwas nutzt was es nicht im (avm-)Kernel gibt merkst du es, zb durch die "Unknown symbol" Fehlermeldung.
Und am besten immer syslog (mit kernellog) anschauen

forenuser, 2020-09-02:
Wenn CONFIG_IKCONFIG_PROC gesetzt ist. existiert die Datei /proc/config.gz. Diese ist identisch mit der AVM Konfiguration der Kernel Sourcen für 7590 7.20.
Mir geht es darum, mögliche Ursachen für die SMBv1.0 Fehlfunktion in der CIFS Kernel-Konfiguration auszuschließen, bevor Aufwand in die Aktualisierung von CIFSMOUNT gesteckt wird.

fda77, 2020-09-02:
Prima wenn die .config passt! Das hat sich letzte Zeit bei avm gebessert. Auf dem 1200 mit kernel 4.4 hat avm die config.gz leider nicht aktiviert
Warum cifs nicht will weiss ich nicht. Mir fallen nur diese 3 Dinge ein die es sein könnten: alter daemon, modul config oder im kernel fehlt was. Am letzten kann man nichts (ohne rk) ändern.  Es kann aber auch was ganz anderes sein"
Was manchmal hilft ist nach der Fehlermeldung zu suchen und zu schauen wie andere das behoben haben

fda77, 2020-09-02:
Ich habe mit der aktuellen labor der 7490 cifs erfolgreich mounten können (kernel 3.10)
Dazu habe ich einfach alle Module die im kernel-menoconfig von cifs ausgewählt werden ("help") ins image gepackt
```
installing and stripping modules
  nls_iso8859-15.ko .................    1.67 Kb
  nls_iso8859-1.ko ..................    1.59 Kb
  nls_cp437.ko ......................    1.92 Kb
  nls_cp850.ko ......................    1.86 Kb
  cifs.ko ...........................  131.76 Kb
  sha256_generic.ko .................    4.64 Kb
  md5.ko ............................    1.70 Kb
  md4.ko ............................    2.52 Kb
  ecb.ko ............................    1.86 Kb
  arc4.ko ...........................    1.49 Kb
  des_generic.ko ....................    6.60 Kb
  hmac.ko ...........................    2.48 Kb
```
Keine Ahnung ob man die alle braucht! Diese musst man auch einzeln auf der Box laden, sonst schlug der mount mit "mount error(2): No such file or directory" fehl

forenuser, 2020-09-02:
Danke, das werde ich mal bei der 7590 probieren.

forenuser, 2020-09-02:
Geht bei der 7590 7.20 ohne Patchen nicht, da einiges nicht als Modul auswählbar ist, sondern als builtin aktiviert wird.
` x Symbol: CIFS [=m]                                                                                                                   x
  x Type  : tristate                                                                                                                    x
  x Prompt: CIFS support (advanced network filesystem, SMBFS successor)                                                                 x
  x   Location:                                                                                                                         x
  x     -> File systems                                                                                                                 x
  x       -> Network File Systems (NETWORK_FILESYSTEMS [=y])                                                                            x
  x   Defined at fs/cifs/Kconfig:1                                                                                                      x
  x   Depends on: NETWORK_FILESYSTEMS [=y] && INET [=y]                                                                                 x
  x   Selects: NLS [=y] && CRYPTO [=y] && CRYPTO_MD4 [=m] && CRYPTO_MD5 [=y] && CRYPTO_HMAC [=y] && CRYPTO_ARC4 [=y] && \               x
  x CRYPTO_ECB [=y] && CRYPTO_DES [=y] && CRYPTO_SHA256 [=y] && CRYPTO_CMAC [=y]                                                        x
`

fda77, 2020-09-02:
Wenn Module auf Y sind sind die ja schon im Kernel - auch in dem von AVM.
Die cofnig die Freetz nutzt sollte da gleich der von avm sein
Nur das MD4 ist ein Modul.

fda77, 2020-09-03:
Funktioniert bei mir mit 7590 7.20
Ich hatte das md4.ko im image aber nicht geladen
config:
```
+FREETZ_MODULE_cifs=y
-FREETZ_MODULES_OWN=""
+FREETZ_MODULES_OWN="md4"
+FREETZ_PACKAGE_CIFSMOUNT_REMOVE_WEBIF=y
+FREETZ_PACKAGE_CIFSMOUNT=y
```
Vielleicht half der bump der cifstools

Conan179, 2020-09-03:
Bei mir gehts nicht. ich bekomme immer noch die selbe fehlermeldung.
das hinzufügen von md4 bricht er den make vorgang bei mir ab:
`WARNING: File of kernel module not found: exportfs.ko
ERROR: Enable missing with 'make kernel-menuconfig' as (M)odule (and push a pull request). Use '/' to search in menuconfig.
make: *** [Makefile:275: firmware-nocompile] Error 1`

fda77, 2020-09-03:
exportfs hat jetzt aber nichts mit cifs zu tun

Conan179, 2020-09-03:
-.- du hast recht, war mein fehler, ich hätte nicht gelichzeitig veruschen soll nfs reinzubringen.
Ohne nfs läuft der make wieder ohne probleme durch und 
+FREETZ_MODULES_OWN="md4"
hat meinen fehler behoben.
Aber beim start wird es nicht verbunden, die stelle wo er es versucht zu verbinden, ist bevor die WLAN MESH verbindung steht.

forenuser, 2020-09-04:
Bei mir hat sich nichts geändert. Beim Mounten der NAS Freigabe der Fritzbox erhalte ich "Host is down" und beim Mounten eines externen NAS stürzt die Box nach der Passworteingabe ab.

fda77, 2020-09-04:
Komisch. Gibt es noch Fehlermeldugnen vor dem Crash?
Wie mountest du denn?
```
mount.cifs //server/share $DIR -o user=u,pass=p,domain=d,ip=192.168.178.2,rw
```
Ich hab das mit Windows 7 und einem Fedora Freigaben probiert

[image]

Conan179, 2020-09-04:
@fda77 Kann den cifsmount verzögert starten lassen? das würde mien 2tes problem lösen.

forenuser, 2020-09-04:
Ich verwende:
root@fritz:/var/mod/root# mount -t cifs //192.168.1.28/homes test -o user=u,pass=p

Welche CIFS Kernelkonfiguration verwendest Du?

fda77, 2020-09-04:
Ich habe an der kernel-config nichts geändert, wie im repo. Ruf mal das mount mir "strace -fff ........." davor auf, vielleicht sieht man was
@Conan179: Ach, du nutzt das webinterface?. Ja, mit "eigenem service" in external

forenuser, 2020-09-04:
```
root@fritz:/var/mod/root# strace -fff mount.cifs //192.168.1.23/admin test -o user=admin
execve("/sbin/mount.cifs", ["mount.cifs", "//192.168.1.23/admin", "test", "-o", "user=admin"], 0x7fce9a54 /* 301 vars */) = 0
readlinkat(AT_FDCWD, "/proc/self/exe", "/sbin/mount.cifs", 4096) = 16
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x77a0a000
open("/usr/lib/freetz/libc.so.1", O_RDONLY) = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=471992, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x77a09000
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0\274\0\0\0\0004"..., 4096) = 4096
mmap(NULL, 557056, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x77970000
mmap(0x77970000, 463136, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x77970000
mmap(0x779f1000, 7956, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x71000) = 0x779f1000
mmap(0x779f3000, 17604, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x779f3000
close(3)                                = 0
munmap(0x77a09000, 4096)                = 0
stat("/usr/lib/freetz/ld-uClibc.so.1", {st_mode=S_IFREG|0755, st_size=29952, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x77a09000
set_thread_area(0x77a10460)             = 0
open("/dev/urandom", O_RDONLY)          = 3
read(3, "\6\5Ms", 4)                    = 4
close(3)                                = 0
mprotect(0x779f1000, 4096, PROT_READ)   = 0
mprotect(0x77a0e000, 4096, PROT_READ)   = 0
ioctl(0, TCGETS, {B38400 opost isig icanon echo ...}) = 0
ioctl(1, TCGETS, {B38400 opost isig icanon echo ...}) = 0
geteuid()                               = 0
mmap2(NULL, 11260, PROT_READ|PROT_WRITE, MAP_SHARED|MAP_ANONYMOUS, -1, 0) = 0x77a06000
brk(NULL)                               = 0x70c000
brk(0x70d000)                           = 0x70d000
getuid()                                = 0
chdir("test")                           = 0
brk(0x70e000)                           = 0x70e000
getcwd("/var/mod/root/test", 4095)      = 19
clone(strace: Process 22667 attached
child_stack=NULL, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x77a09068) = 22667
[pid 22666] wait4(-1,  <unfinished ...>
[pid 22667] getuid()                    = 0
[pid 22667] getgid()                    = 0
[pid 22667] getuid()                    = 0
[pid 22667] getuid()                    = 0
[pid 22667] getgid()                    = 0
[pid 22667] getuid()                    = 0
[pid 22667] open("/etc/gai.conf", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid 22667] brk(0x70f000)               = 0x70f000
[pid 22667] open("/dev/tty", O_RDWR)    = 3
[pid 22667] ioctl(3, TCGETS, {B38400 opost isig icanon echo ...}) = 0
[pid 22667] ioctl(3, TCGETS, {B38400 opost isig icanon echo ...}) = 0
[pid 22667] ioctl(3, TCSETSF, {B38400 opost -isig icanon -echo ...}) = 0
[pid 22667] write(3, "Password for admin@//192.168.1.2"..., 41Password for admin@//192.168.1.23/admin: ) = 41
[pid 22667] read(3, "x", 1)             = 1
[pid 22667] read(3, "x", 1)             = 1
[pid 22667] read(3, "x", 1)             = 1
[pid 22667] read(3, "x", 1)             = 1
[pid 22667] read(3, "x", 1)             = 1
[pid 22667] read(3, "x", 1)             = 1
[pid 22667] read(3, "\n", 1)            = 1
[pid 22667] write(3, "\n", 1
)           = 1
[pid 22667] ioctl(3, TCSETSF, {B38400 opost isig icanon echo ...}) = 0
[pid 22667] close(3)                    = 0
[pid 22667] exit_group(0)               = ?
[pid 22667] +++ exited with 0 +++
<... wait4 resumed>[{WIFEXITED(s) && WEXITSTATUS(s) == 0}], 0, NULL) = 22667
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=22667, si_uid=0, si_status=0, si_utime=0, si_stime=0} ---
brk(0x70f000)                           = 0x70f000
mount("//192.168.1.23/admin", ".", "cifs", 0, "ip=192.168.1.23,unc=\\\\192.168.1."...
```

forenuser, 2020-09-05:
Ich habe nun alles durch, was mir eingefallen ist (7590 recovern, Minimal Image nur mit CIFS, ...). Immer das gleiche Verhalten: Reset nach mount.cifs. Ich weiß nicht, wo ich noch suchen soll. Warum funktioniert es bei anderen und bei mir nicht?

fda77, 2020-09-07:
Ich hab jetzt mit einer 7590 und cifs auch einen Reboot hinbekommen, weiss nur nicht weshalb. Über Syslog kam gar keine Meldung, aber ein anderes direkt an die FB angeshclossenes Gerät hat sofort netlink-down gemeldet.
Evtl hängt es von der ausgehandelten Authentifizierung oder Protokoll ab. SMB3 gibts für die Fritzbox nicht, da dafür dieses KEY im Kernel fehlt und man diesen nicht ersetzen kann.

Vielleicht sollte man in den cifs-tools ein paar Dinge deaktivieren, wie dieses ntlmssp und smb3
Entpack mal die dl/cifs-utils-6.10.tar.bz2 irgendwo hin und ruf dann "./configure --help" auf. Parameter können dann in die make/cifsmount/cifsmount.mk eingetragen werden

forenuser, 2020-09-08:
Es gibt zwei Optionen:
  --disable-FEATURE       do not include FEATURE (same as --enable-FEATURE=no) und
  --without-PACKAGE       do not use PACKAGE (same as --with-PACKAGE=no)

Wo findet man, welche Features oder Packages abgeschaltet werden können?

fda77, 2020-09-08:
alle die da aufgelistet sind. jetzt sind aber eh alle aus..

fda77, 2020-09-11:
Auf der 7590 muss die md4.ko (und cifs.ko) geladen sein, sonst funktioniert mounten nicht. Mit "sec=ntlm" gibt es keine Crashes, "noserverino" hab ich nicht gebraucht.
@PeterPawn: Hast du eine Idee weshalb Die Option "sec=ntlmssp" die Box sofort und ohne jegliche Meldungen rebooten lässt? Ein an die Box angeschlossenes Gerät meldet in dem Moment in dem man enter drück "network inteface down". Ein Strace ist weiter oben

closed 2020-09-22 by fda77