Freetz-NG/freetz-ng #62: 6590 Cifsmount mount error(2): No such file or directory

Conan179, 2020-09-01:
hallo erst mal.
Ich bräuchte Hilfe beim cfismount.
Auf meiner NAS ist der Ordner
//*ipdernas*/6590/
freigeben für 2 User und Gäste.
cifsmount weigert sich sich den Ordner zu mounten.
Die fehlermeldung ist:
mount error(2): No such file or directory
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Starting cifs ... mount //*ipdernas*/6590/ ...
failed.
Auch per shell klappt das mounten nicht:
root@fritz:/var/mod/root# mount -t cifs //*ipdernas*/6590/ /var/media/ftp/qnap/
Password for root@//*ipdernas*/6590/: 
mount error(2): No such file or directory
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
mount: mounting //*ipdernas*/6590/ on /var/media/ftp/qnap/ failed: Invalid argument

Um auszuschliesen das die nas schult ist, habe ich den gleichen befehl bei einem Pi eingetippt, da klappt der mount.
Das Problem ist also nicht die NAS sondern der cifsmount von freetz.
Könnt ihr mir Helfen?

fda77, 2020-09-01:
Es gibt schon ein Issue zu cifsmount -> https://github.com/Freetz-NG/freetz-ng/issues/61

closed 2020-09-01 by fda77

Conan179, 2020-09-01:
Du hast aber gesehen, das es eine andere fehlermeldung ist?

fda77, 2020-09-01:
Ja, beides dürft aber die gleiche Ursache sein: Seit jahren hat niemand was an cifsmount gemacht, und die Software der Fritzbox wurde mittlerweile aktualisiert. Ich müsste also wiederholen was ich schon im anderen geschrieben habe: Optionen im kernel prüfen und eine neuere version von cifsmount nutzen

Conan179, 2020-09-01:
Endschuldigung, ich wollte blos, das man mir nicht vorwirft ein anderes Issue für mich zu benutzten.

fda77, 2020-09-01:
Viellicht könnt ihr euch dann gegenseitig helfen ;-)