Freetz-NG/freetz-ng #65: 6590 7.20 FW Sourcedatein

Conan179, 2020-09-02:
http://osp.avm.de/fritzbox/fritzbox-6590-cable/sources-PSQ19_Phase_1-6590-armp6.tar.gz
http://osp.avm.de/fritzbox/fritzbox-6590-cable/sources-PSQ19_Phase_1-6590-x86.tar.gz

neues dateinamensschema...
Zum min. endpacken lassen sich beide.

fda77, 2020-09-03:
Das PSQ19 ist der "release cycle", siehe FIRMWARES
Falls rk nicht funktioniert sage bescheid, dann deaktiviere ich es wieder

closed 2020-09-03 by fda77

Conan179, 2020-09-03:
@fda77 ähm ich eine kleine winzige frage, wie akt. den github clone auf meinem PC?

fda77, 2020-09-03:
Ich denke Google hätte dir das schneller gesagt: git pull
Falls du lokal Änderungen gemacht hast siehst du die mit git status und git diff

Conan179, 2020-09-03:
Ich hab git pull gemacht, 
make menuconfig
zeigt mir aber an, das Replace Kernel nicht verfügbar ist.
git diff zeigt nur das an, was ich für cifs aktivert habe.
`freetz@freetz-linux:~/6590$ git diff
diff --git a/make/linux/configs/freetz/config-puma6.x86-6490_07.19 b/make/linux/configs/freetz/config-puma6.x86-6490_07.19
index cee56dbbd..637971585 100644
--- a/make/linux/configs/freetz/config-puma6.x86-6490_07.19
+++ b/make/linux/configs/freetz/config-puma6.x86-6490_07.19
@@ -2464,13 +2464,16 @@ CONFIG_NFS_COMMON=y
 CONFIG_SUNRPC=m
 # CONFIG_CEPH_FS is not set
 CONFIG_CIFS=m
-# CONFIG_CIFS_STATS is not set
-# CONFIG_CIFS_WEAK_PW_HASH is not set
-# CONFIG_CIFS_UPCALL is not set
-# CONFIG_CIFS_XATTR is not set
-# CONFIG_CIFS_DEBUG2 is not set
-# CONFIG_CIFS_DFS_UPCALL is not set
-# CONFIG_CIFS_EXPERIMENTAL is not set
+CONFIG_CIFS_STATS=y
+CONFIG_CIFS_STATS2=y
+CONFIG_CIFS_WEAK_PW_HASH=y
+CONFIG_CIFS_UPCALL=y
+CONFIG_CIFS_XATTR=y
+CONFIG_CIFS_POSIX=y
+CONFIG_CIFS_DEBUG2=y
+CONFIG_CIFS_DFS_UPCALL=y
+CONFIG_CIFS_ACL=y
+CONFIG_CIFS_EXPERIMENTAL=y
 # CONFIG_NCP_FS is not set
 CONFIG_CODA_FS=m
 # CONFIG_AFS_FS is not set
freetz@freetz-linux:~/6590$
`
Muss ich noch was aktivieren das er es mir anzeigt?
[image]

fda77, 2020-09-03:
Der Screenshot zeigt dass du auf dem Stand 3430912b94872776d9bdae4a71202ba17ea9eccc vom 28.8. bist.
Vielleicht ging das "pull" nicht wegen den Änderungen? Diese Datei wird nämlich umbenannt
Vielleicht hilft "git reset --hard" ... ich hab keine Ahnung von git! Am besten selbst googeln

Conan179, 2020-09-03:
-.-
ICh danke dir für deine hilfe, meine änderung für den kernel war der übeltäter, änderung rückgäging gemacht git pull lief durch.
Endschuldigung für die doofen fragem.

fda77, 2020-09-03:
Ich hab ein paar git Befehle zusammengesucht und in die [README](https://github.com/Freetz-NG/freetz-ng) gepackt: f960d85df75eead8959f260707bc4f9818952aac
Damit sollte man auch alle lokalen Änderungen löschen können um zu updaten

Conan179, 2020-09-03:
@fda77 image mit replacment image ist geflsht und hat gebooted, ich werde es jetzt mal beobachten ob es zu neustart kommt.

Ich DANKE dir für deine Hilfe.

fda77, 2020-09-03:
Wenn der avm-watchdog einen reboot auslöst, geschieht dies maximal 5 minuten nach dem booten
berichte darüber am besten hier -> https://github.com/Freetz-NG/freetz-ng/issues/58