Freetz-NG/freetz-ng #67: 7590 7.20: Davfs2 Webkonfiguration funktioniert nicht

forenuser, 2020-09-05:
Die Webkonfiguration von Davfs2 scheint nicht zu funktionieren. Manuelles Mounten funktioniert mit dem folgenden Befehl problemlos:
`mount.davfs https://webdav.smartdrive.web.de /tmp/test/myDAV
`
Allerdings muss der Benutzer davfs2 vorher angelegt werden!

fda77, 2020-09-07:
Du könntest versuchen ob ein Aufruf von "config" in load) in https://github.com/Freetz-NG/freetz-ng/blob/master/make/davfs2/files/root/etc/init.d/rc.davfs2 bewirkt was du möchtest
Es bietet sich übrigens an autofs für cifs,davfs und samba zu verwenden. die 7590 hat das sogar scheinbar im kernel drin

forenuser, 2020-09-08:
In der Webkonfiguration gibt es 4 Konten. Bei Konto 0 wird versucht, die Daten zu übernehmen, welche in der AVM Konfiguration hinterlegt sind. Beim Benutzernamen und Passwort funktioniert dies nicht (mehr). Bei den folgenden Konten soll ein Serverkey hinterlegt werden - eigentlich der Key der Zertifizierungsstelle. Dieser wird scheinbar nicht akzeptiert, weil dieser Key nicht verifiziert werden kann.

fda77, 2020-09-08:
Entweder jemand fixt dieses avm-konto0 oder es wird einfach deaktiviert
Fehlt dir vielleicht einfach ein intermediate-Zertifikat? Das kommt öfter vor.
Wäre aber noch besser wenn man das ca-bundle package dazu verwenden könnte, dann muss man mit dem Zertifikat nur was machen wenn es ein selbst signiertes ist

forenuser, 2020-09-08:
Bei Konto 1 hat es nun mit dem T-TeleSec GlobalRoot Class 2 Zertifikat funktioniert. In /mod/etc/ssl/certs/ca-bundle.crt scheint das erforderliche Zertifikat nicht enthalten zu sein.

fda77, 2020-09-08:
Wenn du ohne Warnmeldung mit dem Browser auf die webdav Adresse zugreifen kannst, sollte es im ca-bundle drin sein

fda77, 2020-09-08:
Aber fang doch mit einer aktuellen Version an: http://mirror.netcologne.de/savannah/davfs2/
Die in freetz ist 6 Jahre älter als die neueste

forenuser, 2020-09-09:
Zuerst kommen die wichtigeren Themen ;-)

fda77, 2020-09-15:
0d3686e6ceedf5c296f3537129fbfce52a534129 bump lib-neon
e827e01e3dcde21bdfdd71c3e97dad24c8464ed3 bump davfs2
bf5b397e79c48763685bcc53ec499ce3eb3ee270 davfs2: add new patches
3fe177df563fc3dd3b9044d92456e6a7b706cf04 davfs2: move old patches

fda77, 2020-09-15:
geht nun mit autofs auf 7490

 847a823cf ca-bundle: fix davfs2
 d675b829d autofs: fix davfs2
 02cace68b davfs2: add ca-bundle suboption

closed 2020-09-15 by fda77