Freetz-NG/freetz-ng #70: Ist möglich collectd auf die Version 5.12.0 Aktualisieren?

Conan179, 2020-09-08:
Ich nutzte auf meiner box collectd im freetz ist die version 4.10.3, unter unstable packeges enthalten.
Es läuft bei mir, ich hab aber festgestellt das er mein syslog mit meldung wie:
Sep  8 21:28:51 fritz daemon.notice collectd[25310]: uc_update: Value too old: name = fritz.box/df/df-root; value time = 1599593331; last cache update = 1599593                                                                                                               331;
Sep  8 21:28:51 fritz daemon.info collectd[25310]: Filter subsystem: Built-in target `write': Dispatching value to all write plugins failed with status -1.
Sep  8 21:29:01 fritz daemon.notice collectd[25310]: uc_update: Value too old: name = fritz.box/df/df-root; value time = 1599593341; last cache update = 1599593341;
Sep  8 21:29:01 fritz daemon.info collectd[25310]: Filter subsystem: Built-in target `write': Dispatching value to all write plugins failed with status -1.
flutet.
Laut https://collectd.org/wiki/index.php/Plugin:LogFile gibt es ein plug in mit dem der log in ein seperates File geschrieben kann.

Meine frage ist also, wäre es möglich collectd zu aktullisieren und das plugin mit aufzu nehmen?

fda77, 2020-09-08:
Nur zu, siehe "Package bump" auf https://freetz-ng.github.io/freetz-ng/SUPPORT.html

Conan179, 2020-09-08:
Ok, danke ich lesen mich ein.

Conan179, 2020-09-08:
ICh bin bis:
`freetz@freetz-linux:~/free_up$ AUTO_FIX_PATCHES=y make collectd-unpacked
preparing ... mkdir -p source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/collectd-5.12.0; tools/bunzip2 -c dl/collectd-5.12.0.tar.bz2 | tools/tar-gnu -x -C source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/collectd-5.12.0 --transform='s|^./\+||' --strip-components=1
set -e; shopt -s nullglob; for i in make/collectd/patches/*.patch*; do case $i in *.patch|*.patch.gz|*.patch.bzip2|*.patch.bz2|*.patch.bz|*.patch.xz|*.patch.lz|*.patch.lzma|*.patch.Z) ;; *) continue ;; esac; tools/freetz_patch source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/collectd-5.12.0 $i; done;
    applying patch file make/collectd/patches/100-system_libltdl.patch
    patching file configure
    Hunk #1 FAILED at 2527.
    1 out of 1 hunk FAILED -- saving rejects to file configure.rej
    ----------------------------------------------------------------------
ERROR: modpatch: Error in patch-file make/collectd/patches/100-system_libltdl.patch
make: *** [make/collectd/collectd.mk:93: source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/collectd-5.12.0/.unpacked] Error 2
`

Morgen schaue ich weiter was da nicht will.

fda77, 2020-09-09:
Du könntest Code als solchen kennzeichnen

fda77, 2020-09-09:
Ein codeblock beginnt hier auf github mit je 3x ``` ` ```

Conan179, 2020-09-09:
Ok, das nexte mal weis ich das.
Der Knopf in der Leiste macht nur ein zeichen.

Conan179, 2020-09-09:
Falsch geklickt, sorry.

closed 2020-09-22 by fda77