Freetz-NG/freetz-ng #71: Problem with shell redirection

harryboo, 2020-09-11:
Hi,

i think there is a redirection problem when using dehydrated.

I got this error message when checking or updating my certificates:
/usr/bin/dehydrated: line 302: syntax error: unexpected redirection

In this line there is a redirector: <<<

Working version:
Freetz ng-16870-7cf432929 (03.07.2020 07:06:51)
BusyBox v1.32.0 (2020-07-03 06:47:13 CEST) built-in shell (ash)

Can you please check this. Thanx

fda77, 2020-09-11:
The "<<<" is a "here-string". Maybe something in the busybox needs to selected.
Or its a bug in the busybox. Does it work with the other busybox version?

harryboo, 2020-09-12:
I have not changed the busybox version and the busybox options are the same as before.
I have checked another version (Freetz ng-17010-416c8a85d (04.08.2020 08:19:59)). This version doesnt work.
I have tested also busybox v1.31.1 and it has the same problem.

harryboo, 2020-09-12:
NonWorking Version: Freetz ng-d119776f8-HEAD (12.09.2020 10:05:34)

harryboo, 2020-09-12:
It sems that this option is the culprint:

FREETZ_BUSYBOX___V132_BASH_IS_ASH=y

Where can i change this option?
If i set this option in the config, load and save the config the option is back.

Thanx

harryboo, 2020-09-12:
So i changed this option in the .config file. This results in a prompt during image creation. After pressing '3' for NONE the creation was succesfull and dehydrated works again with any error message.

fda77, 2020-09-12:
You could change make/busybox/generate.sh and run it. But i think it does not help.
Try: `cat <<< "xxx"` on your pc and fritzbopx
But selecting the bash PACKAGE should help to run dehydrated

harryboo, 2020-09-12:
Sorry, i think your fix doesnt work.
I creted a new image with the lastest version (freetz-ng-17145-a9988de92) and i got the same error message:
/usr/bin/dehydrated: line 302: syntax error: unexpected redirection

fda77, 2020-09-12:
Check it after "make" with "ls -al $(find build/mod* -name *ash)"
Bash should be a file and not replace with the link to busybox.
Maybe a "make busybox-dirclean" helps (before make)

harryboo, 2020-09-14:
bash is a binary and not a link to busybox:
jens@jens-virtual-machine:~/freetz-ng$ ls -al $(find build/mod* -name *ash)
lrwxrwxrwx 1 jens jens      7 Sep 14 15:57 build/modified/filesystem/bin/ash -> busybox
-rwxr-xr-x 1 jens jens 517496 Sep 14 15:44 build/modified/filesystem/bin/bash
lrwxrwxrwx 1 jens jens      7 Sep 14 15:57 build/modified/filesystem.outer/bin/ash -> busybox

but it doesnt work, same error message as before: /usr/bin/dehydrated: line 302: syntax error: unexpected redirection

fda77, 2020-09-14:
Then maybe the 1st line in dehydrated does not work as expected, try to run on the box:
`/usr/bin/env bash`
https://github.com/dehydrated-io/dehydrated/blob/master/dehydrated#L1
The internal busybox applets could be returned first
replacing it by "#!/bin/bash" should work.
Disabling bash in busybox as you wrote above could be the best, as bb's bash is not working...

fda77, 2020-09-14:
Btw, as you did not attach your .config ... have you enabled the experimental/buggy FREETZ_BUSYBOX__NOEXEC_NOFORK_OPTIMIZATIONS option in menuconfig?
this selects
FREETZ_BUSYBOX_FEATURE_PREFER_APPLETS
FREETZ_BUSYBOX_FEATURE_SH_NOFORK
FREETZ_BUSYBOX_FEATURE_SH_STANDALONE

harryboo, 2020-09-15:
FREETZ_BUSYBOX__NOEXEC_NOFORK_OPTIMIZATIONS=y

root@gateway:/var/mod/root# busybox bash --version
BusyBox v1.32.0 (2020-09-14 15:34:10 CEST) built-in shell (ash)

root@gateway:/var/mod/root# /usr/bin/env bash --version
BusyBox v1.32.0 (2020-09-14 15:34:10 CEST) built-in shell (ash)

fda77, 2020-09-15:
Then disable it :)
This option (the items selected by this) could cause other problems, eg internal applets are used even avm has external binaires.
I dont know why er13 enabled this by default, but it is no more default since some weeks. AVM does not use these options

harryboo, 2020-09-15:
Option disabled. ---->>> Works.

root@gateway:/var/mod/root# busybox bash --version
BusyBox v1.32.0 (2020-09-15 18:57:26 CEST) built-in shell (ash)

root@gateway:/var/mod/root# /usr/bin/env bash --version
GNU bash, version 3.2.57(2)-release (mips-unknown-linux-gnu)
Copyright (C) 2007 Free Software Foundation, Inc.

Thanx

fda77, 2020-09-15:
By enabling keep-binaries by default with https://github.com/Freetz-NG/freetz-ng/commit/fb1d740edf16e3a752ae1035655298f596008dc0 , the "optimizations" are force-disabled: https://github.com/Freetz-NG/freetz-ng/blob/master/make/busybox/Config.in#L527

closed 2020-09-15 by fda77