Freetz-NG/freetz-ng #75: Update package autofs 5.1.6

forenuser, 2020-09-14:
Ich habe mal versucht, autofs zu aktualisieren. Der Daemon läuft, das automatisch Mounten geht leider nicht. Es werden Mount Befehle ausgeführt. Diese funktionieren i.d.R., wenn ich diese in der Shell ausführe. Ich lege das Paket mal bei. Evtl. kann mal jemand nachsehen, wie das ganze lauffähig gemacht werden kann.
[make.autofs.tar.gz](https://github.com/Freetz-NG/freetz-ng/files/5220630/make.autofs.tar.gz)

fda77, 2020-09-14:
Man kann übrigens pull-requests auschecken: `git fetch origin pull/ID/head:BRANCHNAME`

forenuser, 2020-09-15:
Getestet habe ich mit 7590 7.20.

fda77, 2020-09-15:
Das Problem ist dass "mount" sich auf dem PC in einem anderen Verzeichnis als auf der Box befindet. Das neu eingebaute configure merkt sich diesen Pfad und der ist dann in der Binary

fda77, 2020-09-15:
0b9fab0f479b62f7fd9351bf6f7085075804cdfb autofs: suboptions need modules
86cdd811fbb9634771aeda61a9706e3af91dc869 autofs: no webifs for suboptions
b40ee9e4ef6c66a333094cd15af2339ebefc8213 autofs: add curlftpfs suboption
cbf7fc55a1ccf8983e507d66047b019d2673b543 autofs: pimp menuconfig
0161cdedc7479850a4e344074bc4d0c012bee4be autofs: add autofs.conf
8c00fa1e639c7db3d779a60a4a147d4fa5a0b6b3 autofs: add examples
4f30aa370f0c0c774dad9556e228cc8865ad3452 autofs: load some modules
c14eba1dd28563f269173782350d33e66f090b66 autofs: fix bump
6e972f5cf74624711cf03e970bcfb60ca0fe50dc autofs: update docs
17f2238432fc6550de7e306d96f1b709748e5de3 autofs: bump by forenuser
d1e0ae4d9ddc3d8fac29e9087c928094c619d02e autofs: change dependencies
9b539b9cab0d75cc201b5cef412c9229e77e4a37 autofs: add new patches, by forenuser
7646ac48b4de74f3ae00a0d08fc0c0015c8a9a1d autofs: move old patches

fda77, 2020-09-15:
Auf meiner 7490 mit kernel 3 funktioniert es nun.
sshfs könnte noch ein remove-webif vertragen. Pass aber auf die Tabs/Spaces auf, vermutlich zeigt dein Editior die nicht an

forenuser, 2020-09-15:
Ein kleines Problem scheint es noch zu geben: Die Webkonfiguration greift auf /tmp/flash/autofs/autofs.conf zu während der Daemon auf die Datei auto.conf im selben Verzeichnis zugreift. Ansonsten scheint es bei der 7590 7.20 zu funktionieren.

closed 2020-09-15 by fda77

forenuser, 2020-09-15:
OK, dann hole ich mir mal einen komplett neuen Stand.

forenuser, 2020-09-15:
Mir wäre es lieber, wenn es unter autofs eine Option zum Abschalten der Webinterfaces gäbe und dies nicht implizit vorgegeben wird.

fda77, 2020-09-15:
Man muss die "selects" nicht auswählen, man kann auch einfach die packages selbst auswählen. "selects" wählen anderes nur aus.
Dann blieben die Webifs. Nur was willst du damit?? Wenn man autofs verwendet, wird ausschliesslich damit konfiguriert.
Ausserdem: Die Webifs entladen Module, dann kann das eine oder andere nicht mehr gemountet werden

forenuser, 2020-09-15:
Mir geht es um die Verwaltung der zugehörigen Key Dateien. Damit ist man flexibel und kann im Falle, dass etwas nicht funktioniert über die einzelnen Funktionen testen. Wenn dann alles funktioniert, wie gewünscht, kann man die Webinterfaces über die Option entfernen.

fda77, 2020-09-15:
Das sind aber 2 verschiedene Wege. Wenn man testen möchte, kann man auch einfach den Mountbefehl im Terminal eingeben

fda77, 2020-09-18:
Hab autofs mit cifs+nfs auf Geräten mit Kernel 3.10, 4.4 und 4.9 getestet, geht überall!