Freetz-NG/freetz-ng #78: fehlende Module in Freetz der F.Ben 6591/6660

Prisrak1, 2020-09-16:
kann man evtl. die Einstellungen für den Bau der Freetz 6660/6591 Firmware 7.21 in der "make kernel-menuconfig" bereits richtig einstellen?
`WARNING: File of kernel module not found: crc16.ko
WARNING: File of kernel module not found: ext3.ko
WARNING: File of kernel module not found: ext4.ko
WARNING: File of kernel module not found: jbd2.ko
WARNING: File of kernel module not found: mbcache.ko
ERROR: Enable missing with 'make kernel-menuconfig' as (M)odule (and push a pull request). Use '/' to search in menuconfig.
Makefile:275: recipe for target 'firmware-nocompile' failed
make: *** [firmware-nocompile] Error 1
`

fda77, 2020-09-17:
Ist das eine rhetorische Frage? Du hast doch selbst gepostet wie es geht:
make kernel-menuconfig aufrufen und auswählen was du brauchst
mit make kernel-precompiled schauen ob der kernel noch compiliert
einen pull-request erstellen, siehe "Pull request" auf https://freetz-ng.github.io/freetz-ng/SUPPORT.html

closed 2020-09-17 by fda77

Prisrak1, 2020-09-17:
es freut mich sehr, dass du mitließt. Das immer wieder verändern des Quellcodes nimmt viel Zeit in Anspruch.  Denn auch das ".. FREETZ_STYLE_COLORED in Web Interface.." bei meinem und deinem Skin könnte man auch als grün benutzen. Ist noch nicht gemacht worden. Ich hätte beides gern automatisch vorangestellt gehabt. Das mit dem Pull request klappt bei mir noch nicht.

fda77, 2020-09-18:
Freut mich noch viel mehr :D
Geht wohl wegen den generischen Fehlerbehebungen an den Modulen

Wie, Skin ist noch nicht gemacht worden? Ich hab extra nochmal nachgeschaut und konnte keinen pull-request finden. Erinnerung: Du hast den Ursprung der Mauszeiger vergessen...

Prisrak1, 2020-09-20:
pull-request Funktioniert bei mir noch nicht, Dies hat mit der angefragten Skin-Umstellung von grau auf Grün und Grau, bzw. Benutzung dieser zwei verschiedenen Möglichkeiten, nichts zu tun. Die Mauszeiger (erinnere mich immer noch) werden von mir aktualisiert, bzw. neu erstellt.

fda77, 2020-09-20:
Wenn bei dir grunsätzlich pull-request "nicht geht", wende dich an support@github.com

Du scheinst aber etwas grundlegendes nicht verstanden zu haben
falsch: jemand will was, kommt hier her und schreit "ich will" und sonstwer kümmert sich darum
richtig: wenn du ein Update von deinem Skin willst und dass bei diesem grüne bars erlaubt sind, mach daraus einen pull request.

Prisrak1, 2020-09-20:
ich bemühe mich, danke für Hinweis.. benötigt "ich will" wird efinitiv nichts.

fda77, 2020-09-20:
Dann lösch die 1 Zeile die die grünen Bars für dein Skin verhindert