Freetz-NG/freetz-ng #79: 7490 crashes and reboots after starting openvpn

m4rtini2, 2020-09-24:
Hello,
 i have compiled a 7490_07.21.all_freetz-ng-17227-dfd1416_20200923-004128 image with openvpn and updated my 07.12 box. After starting openvpn the box crashes and reboots. This happens every time i start openvpn.
I suspect the missing yf_patchkernel and tun kernel modules as 'lsmod' does not list them. 
On 7.12 the yf_patchkernel module was selectable in the net section of the kernel modules. On 7.21 it is not there. Also i can not find the tun module.
Which options are needed to compile the modules and can them be added to the .config before executing 'make'?

fda77, 2020-09-24:
Theres no source available, ask avm to release it

closed 2020-09-24 by fda77

m4rtini2, 2020-09-24:
The FIRMWARES.md file lists 146.07.21 rev81809 {ALL} [PSQ19] as avaiable
see https://github.com/Freetz-NG/freetz-ng/blob/master/docs/FIRMWARES.md#fritzbox-fon-wlan-74xx
If this is not the correct one i will ask for the sources, but for which one?

Conan179, 2020-09-24:
Source fils are from
http://osp.avm.de/fritzbox/fritzbox-7490/
https://github.com/Freetz-NG/freetz-ng/blob/master/docs/FIRMWARES.md#fritzbox-fon-wlan-74xx
Does not say whether the source files are available. AVM does not release them on their own, you have to request them again and again by email.

fda77, 2020-09-24:
Yes .... for every device and for every version
See https://github.com/Freetz-NG/freetz-ng/commit/f8a7bbd1b22f681dafec8ba6ef9d55f4edf7d24e and https://freetz-ng.github.io/freetz-ng/SUPPORT.html

m4rtini2, 2020-09-24:
I don't understand the f8a7bbd file but i just asked AVM for the sources of FRITZ.Box_7490-07.21.image. I hope that will be enough to compile a working openvpn package. 
This issue can be closed.