Freetz-NG/freetz-ng #81: Alien Firmware für Fritzboc 7362?

forenuser, 2020-10-03:
Wäre es evtl. möglich eine Alien Firmware für die 7362 z.B. auf Basis der 7430 Firmware zu erzeugen? Die Hardware der beiden Boxen sollte sehr ähnlich sein.

fda77, 2020-10-03:
Puh, keine Ahnung. Falls du so eine Box hast, kannst du es mal analog dazu ausprobieren: https://github.com/Freetz-NG/freetz-ng/commit/65bc7ef13210838436b25fcd6f110a38cbee8371
Da man Kernelmodule wahrscheinlich nicht austauschen kann (vielleicht hilft vermagic?) ist eine serielle Console dafür sehr ratsam

forenuser, 2020-10-04:
Im Moment habe ich keine 7362. Mit einer Alien Firmware wäre sie wieder interessant. Der SoC entspricht der 7490, WLAN und DECT der 7430 (s. https://www.fritzbox-info.com/FritzBox_Uebersicht.php).

fda77, 2020-10-04:
Ob der Alien funktioniert bevor du dir so eine kaufst kann man nicht sagen. Wie du oben siehst ist der grundlegende Patch nicht sonderlich kompliziert

closed 2020-10-04 by fda77

sfritz2, 2020-10-18:
Wie schon im IP Forum geschrieben, es könnte klappen. Der Vorteil hier ist, dass sich bei der 74xx-Serie die Kernelversion nicht geändert hat, so dass es ein Versuch wert ist und man die fehlenden Module einfach nachladen kann.

Die 7490 wird (wahrscheinlich) wie die 3490 kein WLAN auf dem Lantiq-SoC haben, sondern dies würde einen separaten SoC nachgeladen, siehe auch: https://www.aboehler.at/doku/doku.php/projects:fritz3490 . 

Als ersten Versuch könnte man ein 7490 Image erstellen. evtl. mit Remove DECT und/oder Remove WLAN Patch. Klappt dies, könnte man als nächsten Schritt die 7430-Dateien für WLAN und DECT kopieren.

fda77, 2020-10-18:
Müsste jemand ausprobieren der es auch mit der Hardware testen kann.
Falls sich die Kernelversion nur "ein bisschen" geändert hat, könnte man versuchen mit "vermagic" die Version im Modul anzupassen.
Remove-wlan funktioniert bei mir mit der 7490, es soll aber noch Probleme auf der Loginseite geben