Freetz-NG/freetz-ng #83: 7170->7150 Alien schlägt fehl

sfritz2, 2020-10-06:
CC      networking/libiproute/ipneigh.o
  CC      networking/libiproute/iproute.o
networking/libiproute/ipneigh.c:14:29: fatal error: linux/neighbour.h: No such file or directory
compilation terminated.
scripts/Makefile.build:197: die Regel für Ziel „networking/libiproute/ipneigh.o“ scheiterte
make[2]: *** [networking/libiproute/ipneigh.o] Fehler 1
make[2]: *** Auf noch nicht beendete Prozesse wird gewartet …
  LD      runit/built-in.o
  AR      runit/lib.a
  LD      selinux/built-in.o
  AR      selinux/lib.a
  LD      shell/built-in.o
  CC      shell/ash.o
Makefile:742: die Regel für Ziel „networking/libiproute“ scheiterte
make[1]: *** [networking/libiproute] Fehler 2
make[1]: *** Auf noch nicht beendete Prozesse wird gewartet …
  CC      shell/ash_ptr_hack.o
  CC      shell/math.o
  CC      shell/random.o
  CC      shell/shell_common.o
  AR      shell/lib.a
make[1]: Verzeichnis „/freetz-ng/source/target-mipsel_gcc-4.6.4_uClibc-0.9.29/busybox-1.27.2“ wird verlassen

ERROR: Build failed.
make/busybox/busybox.mk:51: die Regel für Ziel „source/target-mipsel_gcc-4.6.4_uClibc-0.9.29/busybox-1.27.2/busybox“ scheiterte
make: *** [source/target-mipsel_gcc-4.6.4_uClibc-0.9.29/busybox-1.27.2/busybox] Fehler 1

Es findet neighbour.h in den Kernel-Sources nicht.

fda77, 2020-10-06:
Geht bei mir.
Auschecken, Alien einstellen "make busybox-precompiled":
```
  CC      util-linux/rdate.o
  CC      util-linux/umount.o
  AR      util-linux/lib.a
  LINK    busybox_unstripped
Trying libraries: crypt m
 Library crypt is not needed, excluding it
 Library m is not needed, excluding it
Final link with: <none>
make[1]: Verzeichnis „/home/user/freetz-ng/source/target-mipsel_gcc-4.6.4_uClibc-0.9.29/busybox-1.27.2“ wird verlassen
mkdir -p packages/target-mipsel_gcc-4.6.4_uClibc-0.9.29/busybox/; cp source/target-mipsel_gcc-4.6.4_uClibc-0.9.29/busybox-1.27.2/busybox packages/target-mipsel_gcc-4.6.4_uClibc-0.9.29/busybox/busybox; [ "" == "y" ] && tools/patchelf --set-interpreter /usr/lib/freetz/ld-uClibc.so.1 packages/target-mipsel_gcc-4.6.4_uClibc-0.9.29/busybox/busybox || true; /home/user/freetz-ng/toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.29/mipsel-linux-uclibc/bin/mipsel-linux-uclibc-strip --remove-section={.comment,.note,.pdr} packages/target-mipsel_gcc-4.6.4_uClibc-0.9.29/busybox/busybox;
cmd() { PATH="/home/user/freetz-ng/toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.29/mipsel-linux-uclibc/bin:/home/user/freetz-ng/toolchain/build/mipsel_gcc-3.4.6/mipsel-unknown-linux-gnu/bin:/home/user/.local/bin:/home/user/bin:/usr/lib64/ccache:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/home/user/.composer/vendor/bin" LD_RUN_PATH="/usr/lib/freetz" FREETZ_LIBRARY_DIR="/usr/lib/freetz" make -j2  "$@"  || { printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } };   if [ -e source/.echo_item_start -a ! -e source/.echo_item_build ]; then echo -ne "\e[48;5;56mbuilding\e[49m ... "; touch source/.echo_item_build; fi; cmd -C source/target-mipsel_gcc-4.6.4_uClibc-0.9.29/busybox-1.27.2 \
        CC="/home/user/freetz-ng/toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.29/mipsel-linux-uclibc/bin/mipsel-linux-uclibc-gcc" CROSS_COMPILE="/home/user/freetz-ng/toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.29/mipsel-linux-uclibc/bin/mipsel-linux-uclibc-" EXTRA_CFLAGS="-march=4kc -mtune=4kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64" ARCH="mips" \
        busybox.links
make[1]: Verzeichnis „/home/user/freetz-ng/source/target-mipsel_gcc-4.6.4_uClibc-0.9.29/busybox-1.27.2“ wird betreten
make[1]: Verzeichnis „/home/user/freetz-ng/source/target-mipsel_gcc-4.6.4_uClibc-0.9.29/busybox-1.27.2“ wird verlassen
touch -c source/target-mipsel_gcc-4.6.4_uClibc-0.9.29/busybox-1.27.2/busybox.links
cat source/target-mipsel_gcc-4.6.4_uClibc-0.9.29/busybox-1.27.2/busybox.links | sed -r -e 's,/(blkid|wget)$,/\1-busybox,g' > packages/target-mipsel_gcc-4.6.4_uClibc-0.9.29/busybox/busybox.links
done.
```

Am besten komplett neu auschecken und keine alte .config mitschleppen! Evtl hilft auch das neue "make reuseconfig"

closed 2020-10-06 by fda77

sfritz2, 2020-10-31:
Hallo,

es läuft zwar durch, aber das erstellte Image ist nicht lauffähig, es hängt beim Booten. Beim bauen mit freetz-org hingegen klappt es.

fda77, 2020-10-31:
Keine Ahnung, ich hab keine so alte Hardware mehr.
Vielleicht hast du im menuconfig irgendwas komisches eingestellt.
Warum compiliert BusyBox denn nun bei dir?

sfritz2, 2020-11-02:
Ich habe die alte .config einfach gelöscht, da zuviel überbleibsel enthalten war und dann auch freetz-ng neu gezogen. Dann ging auch die Kompilierung von BusyBox.

fda77, 2020-11-02:
Vielleicht hätte auch "make reuseconfig" oder die "make config-clean-deps" geholfen