Freetz-NG/freetz-ng #84: Alien Firmware für 3370 und 3490

sfritz2, 2020-10-06:
Eine Option sollte hinzugefügt werden, um aus der 3490 eine Alien-Firmware für 3370 und 3390 zu erstellen.

Bei 3390 muss zusätzlich WLAN entfernt werden, da es sonst um Reboot-Loop hängt.

Was nicht funktioniert: WLAN und USB.

ich werde später noch ein wenig testen und versuchen die USB-Treiber aus der 7430 in die 3370 und 3390 zu laden. Ebenso werd ich noch testen, ob zumindest das 2,4GHz in der 3390 funktionieren würde.

fda77, 2020-10-06:
Kannst du gerne machen.
Als Beispiel: https://github.com/Freetz-NG/freetz-ng/commit/65bc7ef13210838436b25fcd6f110a38cbee8371
Damit bleiben alle Kernelmodule erhalten. Wenn es läuft und du später aus der original Firmware Kernelmodule übernehmen willst, schau dir https://github.com/Freetz-NG/freetz-ng/blob/master/patches/scripts/100-7340_7390.sh an.
Wenn die Box nicht bootet sollte ist eine seriell Console nötig um zu erkennen wo das Problem liegt

closed 2020-10-06 by fda77

sfritz2, 2020-10-07:
Anbei der Patch für die 3390:  [100-3390_3490.zip](https://github.com/Freetz-NG/freetz-ng/files/5344374/100-3390_3490.zip)

Funktioniert aber nur bis 7.01 gut. Ab 7.10 verursacht das Deaktivieren des WLAN stets einen Login mit Benutzername und Kennwort auf der Fritzbox-Oberfläche

fda77, 2020-10-08:
Schade dass die wenigsten richtige pull-requests machen. Man kann mit git übrigens so einen Patch erstellen: `git diff > NAME.patch`. Aber nur bei einem PR steht der eigene Name als Autor im Commit
Für den Alien muss also Wlan entfernt werden? Statt CONFIG_WLAN=n im Alien-Script sollte der remove-wlan Patch ausgewählt werden https://github.com/Freetz-NG/freetz-ng/blob/master/patches/scripts/360-remove_wlan.sh der das auch macht. Nur gibt es diesen noch nicht für FOS 7.0+ https://github.com/Freetz-NG/freetz-ng/blob/master/config/ui/patches.in#L929

sfritz2, 2020-10-08:
Es sollte nur der WLAN in der Config deaktiviert werden und nicht die WLAN-Dateien entfernt werden. Bei mir ging der Ersteinrichtungs-Assistent sonst nicht mehr.

fda77, 2020-10-08:
Achso. Aber Wlan funktioniert dann trotzdem? Vielleicht muss irgend eine der .lua Webseiten verändert werden. Die "Loginseite" zeigt AVM immer wenn es irgend einen Fehler oder ungültigen Verweis gibt

sfritz2, 2020-10-08:
Wenn es in der Config deaktiviert ist, dann erscheint der WLAN-Abschnitt gar nicht mehr in der Oberfläche und WLAN selbst ist auch aus.

fda77, 2020-10-08:
Wie ein remove patch :D Nur dass mit einem solchendazu noch ein paar Dateien entfernt werden

fda77, 2020-10-08:
und https://github.com/Freetz-NG/freetz-ng/commit/5d165ea1a60dfa43677d557ee052532f73dc6812

Schau mal ob das so funktioniert. Für die neueren FOS muss man https://github.com/Freetz-NG/freetz-ng/blob/master/patches/scripts/360-remove_wlan.sh vermutlich noch anpassen

sfritz2, 2020-10-11:
ja, ich werd den "Remove WLAN" Patch anpassen, so dass es bis 7.0 geht. Der geht seit 6.5 nicht mehr.

fda77, 2020-10-11:
Aktuell ist er bis 6.9 sichtbar, und hat soweit auch auf meiner 7490 funktioniert

sfritz2, 2020-10-11:
ok, habs mit der 3490 alien auf 3390 mit 6.83 getestet, da geht die Seite Internetzugangsdaten einrichten beim Assistenten nicht mehr und man kann dann nur noch mit Parameter auf die Oberfläche kommen, falls die Box resettet wurde.

Unter https://github.com/Freetz/freetz/issues/161 beim freetz-org ist dasselbe Problem beschrieben.

fda77, 2020-10-11:
Ich hab die 7490 immer als ip-client im "internet-mitbenutzen" modus, wahrscheinlich fiel es mir deshalb nicht auf

sfritz2, 2020-10-11:
nein, hab ich nicht. Die Zugangsdaten-Seite funktioniert ja auch, aber nicht im Assistenten.

Mit der Konsole erhalte ich auch nicht wirklich hilfreiche Fehlermeldungen:
Uncaught TypeError: Node.appendChild: Argument 1 is not an object.
    paint /home/home.js?lang=de:734
    cbGetData /js/avmcontent.js?lang=de:137
    callback /js/avmcontent.js?lang=de:144
    callback /js/avmcore.js?lang=de:292
    onreadystatechange /js/avmcore.js?lang=de:288

Verwiesende Zeile: basediv.appendChild(create_FritzOs(data));

Habs aber herausgefunden: In Zeile 5 muss man /usr/lua hinzufügen, also so:

`rm_files $(find ${FILESYSTEM_MOD_DIR} ! -name '*.cfg' -a -name '*wlan*' | grep -Ev "^${FILESYSTEM_MOD_DIR}/(dev|oldroot|proc|sys|var|usr\/lua)/")`

sonst wird wlanflags.lua und wlanscan.lua entfernt und zumindest einer der beiden wird wohl benötigt.

sfritz2, 2020-10-11:
so unter https://github.com/Freetz-NG/freetz-ng/pull/90 Pull Request eingereicht und berücksichtigt auch noch eine Änderung für 7.0x.

fda77, 2020-10-12:
Cool :+1:
Sollte man nicht auch noch den Inhalt der Klammer hier löschen? https://github.com/Freetz-NG/freetz-ng/blob/master/config/ui/patches.in#L931
Und sollten danach nicht die Aliens den Patch auswählen?
Deine Daten auf dem Computer stimmen nicht mit denen von Github überein, deshlab wird "sfritz" nicht zu "sfritz2" verlinkt, siehe `"git config --global"`

sfritz2, 2020-10-12:
bis 7.0 kannst du freigeben, für 7.1 muss ich noch ein Patch für die Loginseite beisteuern.

fda77, 2020-10-12:
Ich hab es mit 7490 7.2x (ip-client) getestet, mir ist kein Fehler aufgefallen. Was ist denn mit der Loginseite? Evtl tritt es bei mir durch einen anderen remove-patch nicht auf?
Bei der Internetzugangsart war kein Wlan auswählbar. Bin mir aber nicht sicher ob nur die 7590 "Internet über Wlan Hotspot" kann

sfritz2, 2020-10-12:
Ab 7.1x kann ich mich nicht einloggen. Werkseinstellungen zurücksetzen funktioniert auf der Seite auch nicht. Die Loginseite kommt stets mit Username und Passwort, auch wenn man das gar nicht so konfiguriert hat. Ich habe gesehen, dass es Javascript fehler gibt und var config = leer ist, aber bisher hab ich noch keinen Weg gefunden, das zu beheben, da spielen andere Faktoren mit. Das alles wird nur verursacht,weil CONFIG_WLAN=n steht.

sfritz2, 2020-10-18:
wenn die Fritzbox 3390 zu oft neu gestartet wird, weil das WLAN nicht geladen werden konnte, gibt irgendwann die Firmware auf die fritzbox neu zu starten wegen wlan. Das heißt, es startet dann ohne Reboot-Loop durch.

/var/tmp/wland_support.log:

```
01:00:36: open
01:00.36/[0036.062]:[INF]derived config 'AP-only mode Dual', ID: 1 (0x00000000)
01:00.37/[0037.015]:[INF]nexus_init: Initialize Nexus API
01:00.37/[0037.015]:[ERR]nexus_comm_device_get_info: Cannot determine nexus right now - check again later(try again later)
01:00.37/[0037.027]:[INF]wland_hal_remote_create_service:255: Bind failed with EADDRNOTAVAIL, maybe vr9 datapath is currently in reinitialization, retrying
01:00.38/[0038.027]:[INF]wland_hal_remote_create_service:255: Bind failed with EADDRNOTAVAIL, maybe vr9 datapath is currently in reinitialization, retrying
01:00.39/[0039.028]:[INF]wland_hal_remote_create_service:255: Bind failed with EADDRNOTAVAIL, maybe vr9 datapath is currently in reinitialization, retrying
01:00.40/[0040.028]:[ERR]wland_hal_remote_create_service:251: Failed to bind hal callback server socket, reason: 126 (Cannot assign requested address)
Failed address was fe80::XXXXXXXX : 5053 on if 10
01:00.40/[0040.028]:[ERR]wland_hal_remote_proxy_init:42: Failed to create remote callback service
01:00.40/[0040.029]:[ERR]cfg_hal_hardware_init:239: Failed to connect to remote hal client
01:00.40/[0040.032]:[ERR]cfg_mgr_event_handler_cb:1973: Hardware initialization failed with status 0x80000001
01:00.41/[0041.015]:[INF]Sending crash report #1 with error 0101
01:00.41/[0041.032]:[ERR]cfg_hal_hardware_deinit:265: HAL proxy: client is NULL
01:00.41/[0041.048]:[WRN]cfg_mgr_event_handler_cb:1986, WARNING: ignoring return code 0xffffffea from hardware_deinit()
01:00.41/[0041.054]:[INF]signalled request to terminate
01:00.41/[0041.054]:[ERR]_add_job:370: Job 'internal event': epoll_ctl failed, reason: File exists
01:00.41/[0041.054]:[ERR]job_runner_add_write_job:241: Failed to add job 'internal event'
01:00.43/[0043.028]:[INF]Dumping -1934793074 regs from target
01:00.43/[0043.028]:[INF]target_reg_dump[0] = 8cad668e
01:00.43/[0043.028]:[INF]target_reg_dump[1] = 8cad668e
01:00.43/[0043.028]:[INF]target_reg_dump[2] = 8cad668e
01:00.43/[0043.029]:[INF]target_reg_dump[3] = 8cad668e
01:00.54/[0054.000]:[ERR]Timeout while waiting for MDIO lock (read 0x401).
01:00.54/[0054.000]:[ERR]Failed to get the MDIO lock.
01:00.54/[0054.000]:[ERR]Failed to read dump registers 4 from target.
01:00.54/[0054.000]:[ERR]Failed to load 1st stage firmware
01:00.54/[0054.000]:[ERR]ol_dl_main:113 OL-DL: Failed to download firmware to target
01:00.54/[0054.000]:[ERR]offload_mgr_start_system:105, Offload-Downloader exited unappropriately
01:00.54/[0054.000]:[INF]WLAN driver signalled crash
01:00.54/[0054.062]:[INF]wland_hal_remote_create_service:255: Bind failed with EADDRNOTAVAIL, maybe vr9 datapath is currently in reinitialization, retrying
01:00.55/[0055.062]:[INF]wland_hal_remote_create_service:255: Bind failed with EADDRNOTAVAIL, maybe vr9 datapath is currently in reinitialization, retrying
01:00.56/[0056.062]:[INF]wland_hal_remote_create_service:255: Bind failed with EADDRNOTAVAIL, maybe vr9 datapath is currently in reinitialization, retrying
01:00.57/[0057.062]:[ERR]wland_hal_remote_create_service:251: Failed to bind hal callback server socket, reason: 126 (Cannot assign requested address)
Failed address was fe80::2de:adff:febe:efca : 5053 on if 10
01:00.57/[0057.062]:[ERR]wland_hal_remote_proxy_init:42: Failed to create remote callback service
01:00.57/[0057.062]:[ERR]cfg_hal_hardware_init:239: Failed to connect to remote hal client
01:00.57/[0057.064]:[INF]config_exchange_master_init:900: ENTER
01:00.57/[0057.064]:[INF]nexus_init: Initialize Nexus API
01:00.57/[0057.088]:[INF]Sending crash report #1 with error 0400
01:00.57/[0057.091]:[ERR]cfg_hal_hardware_deinit:265: HAL proxy: client is NULL
01:00.57/[0057.094]:[WRN]cfg_mgr_event_handler_cb:1986, WARNING: ignoring return code 0xffffffea from hardware_deinit()
01:00.57/[0057.098]:[INF]signalled request to terminate
01:01.00/[0060.072]:[INF]Dumping -1934793074 regs from target
01:01.00/[0060.072]:[INF]target_reg_dump[0] = 8cad668e
01:01.00/[0060.072]:[INF]target_reg_dump[1] = 8cad668e
01:01.00/[0060.072]:[INF]target_reg_dump[2] = 8cad668e
01:01.00/[0060.073]:[INF]target_reg_dump[3] = 8cad668e
01:01.17/[0077.029]:[INF]wland_hal_remote_create_service:255: Bind failed with EADDRNOTAVAIL, maybe vr9 datapath is currently in reinitialization, retrying
01:01.18/[0078.029]:[INF]wland_hal_remote_create_service:255: Bind failed with EADDRNOTAVAIL, maybe vr9 datapath is currently in reinitialization, retrying
01:01.19/[0079.029]:[INF]wland_hal_remote_create_service:255: Bind failed with EADDRNOTAVAIL, maybe vr9 datapath is currently in reinitialization, retrying
01:01.20/[0080.029]:[ERR]wland_hal_remote_create_service:251: Failed to bind hal callback server socket, reason: 126 (Cannot assign requested address)
Failed address was fe80::XXXXXXXX : 5053 on if 10
01:01.20/[0080.029]:[ERR]wland_hal_remote_proxy_init:42: Failed to create remote callback service
01:01.20/[0080.029]:[ERR]cfg_hal_hardware_init:239: Failed to connect to remote hal client
01:01.20/[0080.031]:[ERR]job_runner_run:622: Try kickout for job: '11', events: 11
01:01.20/[0080.031]:[ERR]job_runner_run:623: Job description: (TIRPC)
01:01.20/[0080.031]:[ERR]cfg_mgr_event_handler_cb:1973: Hardware initialization failed with status 0x80000001
[... cut ...]
https://www.aboehler.at/doku/doku.php/projects:fritz3390
https://www.aboehler.at/doku/doku.php/projects:fritz3490

Also beim 3390 kriegt man 5GHz direkt auf dem Lantiq-SoC, während das 2,4GHz über einen separaten SoC nachgeladen muss. Bei der 3490 gibt es auf dem Lantiq-SoC kein WLAN, sondern nur über den separaten SoC. Bei der 7490 müsste das wohl wie der 3490 sein.

Die Dateien /lib/firmware/ath_tgt_fw1.fw und ath_tgt_fw2.fw existieren auch in der 3390, dort gibt es aber zusätzlich auch Kernelmodule fürs WLAN, vermutlich für das 5GHz.

sfritz2, 2020-10-18:
Nachtrag:

mit dem rm-Befehl aus dem Remove WLAN Patch:

```
rm_files \
  ${FILESYSTEM_MOD_DIR}/sbin/hostapd \
  ${FILESYSTEM_MOD_DIR}/sbin/wstart \
  ${FILESYSTEM_MOD_DIR}/sbin/wpa_supplicant \
  ${FILESYSTEM_MOD_DIR}/usr/bin/wpa_authenticator \
  ${FILESYSTEM_MOD_DIR}/sbin/avmstickandsurf \
  ${FILESYSTEM_MOD_DIR}/usr/bin/iwpriv \
  ${FILESYSTEM_MOD_DIR}/usr/bin/iwconfig \
  ${FILESYSTEM_MOD_DIR}/usr/share/ctlmgr/libwlan.so \
  ${FILESYSTEM_MOD_DIR}/etc/hotplug/udev-avmwlan-usb
```

wird verhindert, dass das WLAN sich startet. Somit wäre die Version ab 7.10 auf 3390 nutzbar.

fda77, 2020-10-18:
Dh also dass der wlan-remove Patch für 7.2 angepasst werden müsste. Das war auch schon das Ergebnis bei einem anderen Alien.
Es gibt durch den patch mifr FOS 7.1+ Probleme beim Login (bei mir nicht)

sfritz2, 2020-10-18:
genau, am einfachsten wäre es, wenn nur diese o.g. Dateien entfernt werden. Dann ist es nur im Webinterface sichtbar, aber es ist nicht aktivierbar und im Ereignislog verschwindet auch die Meldung, dass das WLAN nicht initialisiert wurde.

AVM hat CONFIG_WLAN=n wohl nur für die 7412 gepflegt, da 1&1 fürs WLAN einschalten extra bezahlt werden möchte. Jedoch hat die 7412 keine Version höher als 6.8 bekommen. Das dürfte wohl der Grund sein, warum es in höheren Versionen nicht richtig funktioniert.

fda77, 2020-10-18:
Ich seh keinen Vorteil die Datein fürs Wlan zu entfernen, aber alle Optionen im Webif zu lassen. Man kann damit ja eh nichts konfigurieren