Freetz-NG/freetz-ng #86: NZBGet unvollständig?

forenuser, 2020-10-10:
Ich habe den Eindruck, dass beim Paket NZBGet nur das Binary im Image ist. Eigentlich sollte mindestens die Datei nzbget.conf vorhanden sein. Allerdings finde ich diese Datei nicht.

fda77, 2020-10-10:
Stimmt, da sollte "binary only" dazu

closed 2020-10-10 by fda77

forenuser, 2020-10-10:
Ich versuche gerade, die Version 21.0 zu compilieren. Leider werden dabei C++ Header nicht gefunden. Welche Option muss ich in der .mk Datei setzen, damit compiliert werden kann?

`In file included from daemon/connect/TlsSocket.cpp:21:
./daemon/main/nzbget.h:220:10: fatal error: unordered_map: No such file or directory
 #include <unordered_map>
          ^~~~~~~~~~~~~~~
compilation terminated.
`

fda77, 2020-10-10:
binary-only heisst bei Freetz es gibt nur ne Binary, also keine Webinterface usw Man muss seine config also selbst zusammenbasteln, falls eine nötig ist. Ich kenne nzbget gar nicht, weiss noch nicht mal was man damit machen kann!
Anhand vom Dateinamen "TlsSocket" würde ich mal *vermuten* dass geforderte ssl-libs nicht gefunden werden. Die kann man beim configure evtl definieren oder muss was patchen

Conan179, 2020-10-10:
"NZB"" Files, sind, wie .torrent datein, die auf einen Inhalt zeigen, um sie dann aus dem Usenet, genau genommen aus einem https://de.wikipedia.org/wiki/Usenet#Newsgroups_mit_Bin%C3%A4rdateien_(Binary_Usenet) runter zuladen.

forenuser, 2020-10-10:
NZBGet ist ein Usenet Downloader. Dieser enthält, ähnlich wie PyLoad ein eigenen Web Server. Dafür müssen  die entsprechenden Java Skript Dateien und die Konfiguration in das Image übernommen werden. Normal würde man das laut README mit make inst-conf machen.

Die fehlende Include Datei unordered_map habe ich im c++ Verzeichnis gefunden. Wie kann ich dieses in der nzbget.mk Datei hinzufügen, damit dieser Header gefunden wird?

fda77, 2020-10-10:
Keine Ahnung! Ich würde mir die ganzen anderen make/*/*mk Dateien anschauen, vielleicht gibts da was ähnliches.

fda77, 2020-10-10:
> make inst-conf

Schau was dadurch angelegt wird, und sorg dafür dass es diese Dateien auch auf der Fritzbox gibt. Dateien die man bearbeiten sollte müssen dann ins Flash umgebogen werden

forenuser, 2020-10-11:
Transmission verwendet unordered_map. Allerdings wird mit gcc compiliert. Die NZBGet Compilierung verwendet g++. Der Include Pfad für die c++ Header ist vorhanden. Trotzdem findet g++ die Header nicht.
```
/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-g++-wrapper -DHAVE_CONFIG_H -I.  -I./daemon/connect -I./daemon/extension -I./daemon/feed -I./daemon/frontend -I./daemon/main -I./daemon/nntp -I./daemon/postprocess -I./daemon/queue -I./daemon/remote -I./daemon/util -I./daemon/nserv -I./lib/par2 -I./lib/yencode  -D_REENTRANT -I/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/usr/include/libxml2  -I/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/usr/include -I/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/usr/include  -march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -fpermissive -MT daemon/extension/CommandScript.o -MD -MP -MF $depbase.Tpo -c -o daemon/extension/CommandScript.o daemon/extension/CommandScript.cpp &&\
mv -f $depbase.Tpo $depbase.Po
In file included from daemon/extension/FeedScript.cpp:21:
./daemon/main/nzbget.h:220:10: fatal error: unordered_map: No such file or directory
 #include <unordered_map>
          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [Makefile:1656: daemon/extension/FeedScript.o] Error 1
```

forenuser, 2020-10-11:
Die Version 16.4 konnte ich kompilieren. Bei den neueren Versionen habe ich es aufgrund der c++ Header nicht geschafft.
[image]

fda77, 2020-10-11:
Mal geraten: Vielleicht kannst du die (neuen?) ssl-sachen mit configure deaktivieren? Evtl ist es nur ein https-webinterface?

forenuser, 2020-10-11:
Ich denke eher, dass ssl für die Verbindung zum Newsserver verwendet wird. Ich hatte auch probiert, dass Verzeichnis c++/8.3.0 als Include-Verzeichnis anzugeben. Dann wurden andere c++ Header nicht gefunden. Nach einem weiteren Include gab es dann Fehler bei Compilieren. Ich bin deshalb nicht weitergekommen und deshalb die 16.4 verwendet. Nun kümmere ich mich nun erst um diese Version und das Web Interface.

fda77, 2020-10-12:
Durch https://github.com/nzbget/nzbget/issues/88 brauchen neue Versionen wohl libstdc++ https://github.com/Freetz-NG/freetz-ng/commit/ab6c5f3f4d7b16cd0f8eac713624b207eb8a58d4

forenuser, 2020-10-12:
Wie kann ich das nun mit der 7590 nutzen?

fda77, 2020-10-12:
Genau. Es gibt noch andere Packages die das genau so machen, zb emailrelay, tresseract

forenuser, 2020-10-16:
Tesseract make läuft nicht durch.