Freetz-NG/freetz-ng #104: 7590 7.21: unrar - Probleme mit Umlauten oder Akzentzeichen

forenuser, 2020-10-27:
Beim Entpacken gibt es Probleme mit Dateinamen, welche z.B. deutsche Umlaute oder Akzente enthält. Der Dateiname wird entweder abgeschnitten oder es folgen einige der vorherigen Zeichen. Das Ziel-Filesystem ist ext4. Dateinamen mit "besonderen Zeichen" sind möglich, wenn Sie per samba geschrieben werden.

fda77, 2020-10-27:
Kann alles sein. Ob das Programm es unterstützt, ob es mit compiliert wurde, ob es mit konfiguriert wurde, ob das Dateisystem es kann, wie es gemountet wurde, ob das System es kann, wie es konfiguriert wurde und vielleicht auch ob der Computer mit dem es compiliert wurde es kann.

forenuser, 2020-10-27:
Mit unrar 6.0 beta 1 geht es. Ich werde ein Pull Request dazu erstellen.

forenuser, 2020-10-27:
Geht leider nur, wenn man unrar manuell ausführt. Wenn unrar von nzbget ausgeführt wird, gibt es weiterhin die beschriebenen Probleme.

fda77, 2020-10-27:
Hm, dann stimmt wohl was mit nzbget nicht. Vergleich mal diese Ausgaben auf einem Linux das deutsch installiert hat
```
ls --help
LANG=C ls --help
```

forenuser, 2020-10-27:
Im verlinkten Thread wurde aber keine Lösung gefunden. Die Änderung auf en_US.UTF-8 per shell Skript hat nicht funktioniert. Lediglich eine alte 5.21er unrar Version hat geholfen. Ich kann ja mal ausprobieren, ob eine Änderung der Funktion ResetEnv() was bringt.

fda77, 2020-10-27:
Ja, weil nbz das ENV leert!

forenuser, 2020-10-28:
Ich habe nun ein Shell Skript ausprobiert, mit welchem en_US.utf-8 gesetzt wird. Dies ist beim Aufruf von unrar mit nzbget nicht gesetzt und verursacht den Fehler.

fda77, 2020-10-28:
Hab ich doch oben schon geschrieben: Die Funktion von nzbget ResetEnv() macht was der Name sagt. In diese nimm man dann am Ende das setzen von LANG auf. Wie das geht steht direkt unter dieser Funktion (setenv oder so)

forenuser, 2020-10-28:
Mir ist trotzdem nicht klar, warum die Funktion ResetEnv() LANG nicht zurücksetzt, wenn nzbget manuell aus der ssh-Shell aufgerufen wird. Ich habe nun versucht, LANG in rc.nzbget zu setzen, aber LANG ist beim Aufruf von unrar trotzdem nicht gesetzt. Nun stellt sich mir die Frage, ob ich nzbget patchen sollte oder ein Wrapper Skript verwende, welches den gleichen Zweck erfüllt.

fda77, 2020-10-28:
Probier das was ich oben schon 2x geschrieben hab!
nzb setzt das env zurück und startet dann unrar

forenuser, 2020-10-28:
Einen Fehler habe ich gefunden und den von Dir vorgeschlagenen Workaround eingebaut. Trotzdem hätte ich gerne verstanden, warum es in einem Fall funktioniert und in einem anderen nicht. Nur so kann ich bewerten, ob ein Vorgehen sinnvoll ist oder nicht.

fda77, 2020-10-28:
Gib mal "echo $LANG" auf der Fritzbox und deinem PC ein. Das ist deine Spracheinstellung. nzb löscht scheinbar alle, bevor externe Programme aufgerufen werden. Warum man sowas machen sollte weiss ich nicht! Wenn man manche nicht offenlegen möchte, könnte man die zb mit "unset" löschen. Aber wie dem auch sei
Auf der Fritzbox ausprobieren mit einer "Umlaute"-rar:
"rar <entpackbefehle>"
"LANG='' rar <entpackbefehle>"
"LANG='C' rar <entpackbefehle>"
"LANG='<utf-dingens>' rar <entpackbefehle>"
C ist das minimale englische (ohne umlauts), das jedes System haben sollte, siehe auch 5. Kommentar hier. Das nutzt man manchmal wenn ein Programm parsbare englische ausgaben haben soll

closed 2020-10-28 by fda77

forenuser, 2020-10-28:
Ich habe mit rc.nzbget und dem unrar wrapper experimentiert. Dabei habe ich festgestellt, dass LANG beim Ausführen von rc.nzbget nicht gesetzt ist. Wenn ich LANG in rc.nzbget gesetzt habe, hat das keine Auswirkungen auf das Ausführen von unrar gehabt. In der ssh Shell ist LANG=en_US.UTF-8 gesetzt. Beim Ausführen von nzbget ist diese Umgebungsvariable auch beim Ausführen von unrar vorhanden. Wo wird LANG für die ssh Shell gesetzt? Eine .profile Datei ist in /var/mod/root nicht vorhanden bzw. im Zielverzeichnis des enthaltenen Links nicht vorhanden.

fda77, 2020-10-28:
> Wenn ich LANG in rc.nzbget gesetzt habe, hat das keine Auswirkungen auf das Ausführen von unrar gehabt

nzbget -> clearenv -> unrar
Daher ists egal was vorher gesetzt war. Das Problem kommt ja nicht von Freetz, das im nbz Forum auch andere sich drüber beschweren. Sondern wohl dieses clearenv
Keine Ahnung wo das gesetzt wird, es ist immer gesetzt ausser wenn man es löscht :)

forenuser, 2020-10-30:
Meine Beobachtung hat sich bestätigt. Deshalb habe ich LANG=en_US.UTF-8 vom Patch in rc.nzbget verschoben und einen neuen Pull Request erstellt.

fda77, 2020-10-30:
```
grep LANG= /var/env.cache
export LANG='en_US.UTF-8'
```
Wenn die Variable in dem einen Script fehlt, fehlt sie bei allen. Falls da so ist würde man das besser allgemein löscen

forenuser, 2020-10-30:
Auf diesen Punkt hatte ich schon hingewiesen. Während LANG in der putty shell gesetzt ist, ist das bei der Ausführung von rc.nzbget nicht der Fall. Wodurch wird LANG in der Putty Shell gesetzt (/etc/profile, /var/mod/root/.profile, ...) aber beim automatischen Ausführen von rc.??? Skripten nicht? Bei pyLoad war mir ein Problem mit Dateinamen nicht aufgefallen, obwohl ebenfalls unrar verwendet wird. Allerdings nutze ich pyLoad schon länger nicht.

fda77, 2020-10-30:
Ich weiss noch immer nicht wie putty das macht. Aber ich weiss wo Freetz es löscht: https://github.com/Freetz-NG/freetz-ng/blob/master/make/mod/files/root/etc/init.d/modlibrc#L198
Bei Aufrufen übers webif ist es aber eh leer

forenuser, 2020-10-30:
Wird das webif nicht mit rc.webcfg gestartet? Dann würde modlibrc LANG beim Start von webcfg löschen.

fda77, 2020-10-30:
Das soll ja so sein damit nichts ausgelesen werden kann

fda77, 2020-10-30:
Ja, da der Hunk fehlt(e) noch

forenuser, 2020-10-30:
Ist in meinem Pull Request mit drin.

fda77, 2020-10-30:
Ich hoffe das macht in sone keine Package Probleme