Freetz-NG/freetz-ng #111: 7590 7.21: make läuft nicht durch

forenuser, 2020-11-02:
Aktueller git Stand:
```
make[2]: Entering directory '/home/freetz/7590/source/target-mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/ncursesw-6.2/misc'
make[2]: warning: -j5 forced in submake: resetting jobserver mode.
/usr/bin/install -c ncurses-config /home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/usr/bin/ncursesw6-config
/bin/sh ./gen-pkgconfig
** creating ncursesw.pc
** creating panelw.pc
** creating menuw.pc
** creating formw.pc
touch pc-files
installing ncursesw.pc
installing ncursesw.pc
installing panelw.pc
installing menuw.pc
installing formw.pc
make[2]: Leaving directory '/home/freetz/7590/source/target-mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/ncursesw-6.2/misc'
make[1]: Leaving directory '/home/freetz/7590/source/target-mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/ncursesw-6.2'
sed -i -r -e "s,^(libdir=)(['"'"'"]?)([^'"'"'"]*)(\2)$,\1\2/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc\3\4,g" -e "s,^(includedir=)(['"'"'"]?)([^'"'"'"]*)(\2)$,\1\2/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc\3\4,g" -e "s,^(prefix=)(['"'"'"]?)([^'"'"'"]*)(\2)$,\1\2/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc\3\4,g" -e "s,^(exec_prefix=)(['"'"'"]?)([^'"'"'"]*)(\2)$,\1\2/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc\3\4,g" -e "/^dependency_libs/s,[ \t],  ,g;s,([ '])((/usr)?/lib/[^ /]+[.]la)([ ']),\1/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc\2\4,g;s, +, ,g" /home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/usr/bin/ncursesw6-config
sed -i -r -e "s,^(bindir=)(['"'"'"]?)([^'"'"'"]*)(\2)$,\1\2/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc\3\4,g" -e "s,^(datadir=)(['"'"'"]?)([^'"'"'"]*)(\2)$,\1\2/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc\3\4,g" -e "s,^(mandir=)(['"'"'"]?)([^'"'"'"]*)(\2)$,\1\2/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc\3\4,g"  /home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/usr/bin/ncursesw6-config
sed -i -r -e "s,^(libdir=)(['"'"'"]?)([^'"'"'"]*)(\2)$,\1\2/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc\3\4,g" -e "s,^(includedir=)(['"'"'"]?)([^'"'"'"]*)(\2)$,\1\2/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc\3\4,g" -e "s,^(prefix=)(['"'"'"]?)([^'"'"'"]*)(\2)$,\1\2/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc\3\4,g" -e "s,^(exec_prefix=)(['"'"'"]?)([^'"'"'"]*)(\2)$,\1\2/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc\3\4,g" -e "/^dependency_libs/s,[ \t],  ,g;s,([ '])((/usr)?/lib/[^ /]+[.]la)([ ']),\1/home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc\2\4,g;s, +, ,g" /home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/usr/share/pkgconfig/ncursesw.pc /home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/usr/share/pkgconfig/formw.pc /home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/usr/share/pkgconfig/menuw.pc /home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/usr/share/pkgconfig/panelw.pc
sed: can't read /home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/usr/share/pkgconfig/ncursesw.pc: No such file or directory
sed: can't read /home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/usr/share/pkgconfig/formw.pc: No such file or directory
sed: can't read /home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/usr/share/pkgconfig/menuw.pc: No such file or directory
sed: can't read /home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/usr/share/pkgconfig/panelw.pc: No such file or directory
make: *** [make/libs/ncursesw/ncursesw.mk:66: /home/freetz/7590/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/usr/lib/libncursesw.so.6.2] Error 2
```

fda77, 2020-11-02:
"make dirclean" könnte helfen

forenuser, 2020-11-02:
Log zeigt die Situation nach bereits durchgeführtem dirclean!

fda77, 2020-11-02:
> checking for mips-linux-ncursesw5-config... no
> checking for ncursesw5-config... ncursesw5-config

forenuser, 2020-11-02:
freetz@freetz-linux:~/7590$ git pull
Already up to date.

fda77, 2020-11-02:
Wegen dem dem share / lib Verzeichnis, siehe Zeilr 64/65
https://github.com/Freetz-NG/freetz-ng/commit/825600d4c37e5784eeb650fc997447ffe882cb72#diff-f5b361b5162600e4b96900cae25b8917149ffc82ec9ae94867a5f933c5f6681bL64

forenuser, 2020-11-02:
Und was muss ich nun tun?

fda77, 2020-11-02:
Beschreiben was ich sehen würde wenn ich durch deine Augen schaue
> UTF-8 support was requested, but insufficient support was
>  *** detected in your curses and/or C libraries. 
Ist wieder was ganz anderes. wie das im OP beschrieben

forenuser, 2020-11-02:
Wegen dieser Meldung habe ich make dirclean gemacht. Danach bekam ich den o.g. Fehler. Durch Umstellen auf 1 parallelen Job bin ich an der Stelle weitergekommen, bis ich wieder an dieser Stelle bin, wo ich nicht weiterkomme.

fda77, 2020-11-02:
Ich finds blöd wenn ich dir erklären muss was du gemacht hast!
Vermutlich hast du irgend einem package gesagt dass es unicode nutzen soll. Dies nutze aber die falsche ncurses, sogar die falsche version davon.

fda77, 2020-11-02:
PS:
> 7590 7.21: make läuft nicht durch 

proviziert ein schliessen wegen "hat keine Beine"!

forenuser, 2020-11-02:
Die Fehlermeldung kommt beim Configure von nano.

fda77, 2020-11-02:
Ah, nano. Bei mir gabs da kein Problem beim Umschalten der Option. Sicher dass dirclean in dem Verzeichnis ausgeführt wurde? Evtl muss es auch ein "make distclean" sein. Die ncursesw5-config gitb es eigentlich nur in der alten Version 5

forenuser, 2020-11-02:
Ich habe 2x geschrieben, dass ich make dirclean ausgeführt hatte.

kamasudo, 2020-11-02:
Und deshlab soll ich auch 2x schreiben dass vielleicht nur ein distclean hilft?

fda77, 2020-11-02:
Huch, sorry!
distclean <> dirclean

forenuser, 2020-11-02:
Beide Fehler treten nach neuem git clone immer noch auf. Vielleicht liegt es an der Konfiguration.

fda77, 2020-11-02:
Komisch. Vielleicht muss zu nano irgendwo ein "-lncursesw" hin

forenuser, 2020-11-02:
Ich probiere nun dirclean && distclean && rm -rf ~/.freetz-ccache && rm .config.

fda77, 2020-11-03:
Wie, an beiden Problemen?? Die hängen voneinander ab!
Es kann nur 1 auftreten: https://github.com/Freetz-NG/freetz-ng/blob/master/make/nano/nano.mk#L12

forenuser, 2020-11-03:
Ich habe ja einen neuen Git Stand verwendet und nur mc, nano und nzbget aktiviert. Folgende Beobachtungen sind reproduzierbar:

1. MC kann mit ncursesw oder slang kompiliert werden
2. Nano Option UTF-8 kann ich nicht aktivieren, sonst bricht make ab
3. NZBGet ist nur kompilierbar, wenn ich alle Optionen deaktiviere. Wenn ich With curses user interface oder With SSL/TLS encryption support aktiviere, bricht make ab.

Deine letzte Änderung (ncursesw) ist mit zwei Paketen nicht kompatibel!

MasterRoCcO, 2020-11-03:
mal grade freetz ganz neu per git clone https://github.com/Freetz-NG/freetz-ng ausgelesen
`gcc -m32  avm_kernel_config.extract.o lib_avm_kernel_config.o memory_mapped_file.o  -L/home/bot/server1/freetz-ng/75xx/source/host-tools/dtc-1.6.0/libfdt -lfdt -o avm_kernel_config.extract
/usr/bin/ld: inkompatibles /usr/lib/gcc/x86_64-linux-gnu/10/libgcc.a wird bei der Suche nach -lgcc übersprungen
/usr/bin/ld: -lgcc kann nicht gefunden werden
/usr/bin/ld: inkompatibles /usr/lib/gcc/x86_64-linux-gnu/10/libgcc.a wird bei der Suche nach -lgcc übersprungen
/usr/bin/ld: -lgcc kann nicht gefunden werden
collect2: error: ld returned 1 exit status
make[1]: *** [<eingebaut>: avm_kernel_config.extract] Fehler 1
make[1]: *** Auf noch nicht beendete Prozesse wird gewartet …
make[1]: Verzeichnis „/home/bot/server1/freetz-ng/75xx/source/host-tools/yourfritz-akc-host“ wird verlassen
make: *** [tools/make/yourfritz-akc-host/yourfritz-akc-host.mk:17: /home/bot/server1/freetz-ng/75xx/source/host-tools/yourfritz-akc-host/avm_kernel_config.extract] Fehler 2
bot@s10925715:~/server1/freetz-ng/75xx$
`
der nächste fehler

forenuser, 2020-11-03:
rien ne va plus - Nichts geht mehr ;-)

feedzapper, 2020-11-03:
Der Fehler tritt auch bei einem frisch ausgecheckten freetz-ng auf.
Dabei liegt das Problem beim Erstellen von "ncursesw" selbst.  Man kann dies leicht überprüfen, indem man einfach eine Box im menuconfig z.b. (7590) anwählt und unter "Shared Libraries"->ncurses die ncursesw libs bauen lässt ohne irgendeine "package" mit auszuwählen (was vermutlich dann irgendwann beim Compilieren mit dem bekannten error fehlschlägt.) ...
Dies bezieht sich auf die in  https://github.com/Freetz-NG/freetz-ng/commit/f27db937e974571a09ff11d3dbbb78eb85d132f0#commitcomment-43806675 gemachte Feststellung.

fda77, 2020-11-03:
Prima, schon 3 Probleme im gleichen Ticket
Wenn ihr euch geeinigt habt, könnte mal ein passender TITEL gesetzt werden!

@MasterRoCcO
Dein Problem mit "yourfritz-akc-host" hat hier gar nichts mit zu tun, daran wurde auch nichts geändert. 

> Reproduzierbar mit make ncursesw-dirclean && make ncursesw-uninstall && make
> Workaround: make -j1 nach make Abbruch
Geht bei mir, ich nutze immer "-j9"

Warum jetzt auch noch ssl wegen ncurses nciht funktionieren soll versteh ich nicht
Ich muss das ganze bei Gelegenheit wohl mal auf einem Ubuntu ausprobieren

MasterRoCcO, 2020-11-03:
Der Titel heist doch 7590 7.21: make läuft nicht durch also past er.
und warum geht dann nichts mehr. ganz neu ausgelesenes freetz nichts geändert. und dann der fehler

fda77, 2020-11-03:
ES LÄUFT NICHT WEIL ES KEINE BEINE HAT. OHNE IST AUCH SCHLECHT GEHEN

MasterRoCcO, 2020-11-03:
Super tipp 
gut das damit die fehler behoben werden können!!

closed 2020-11-04 by fda77

forenuser, 2020-11-04:
installing panelw.pc
+ /usr/bin/install -c -m 644 panelw.pc /home/freetz/7590-neu/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.15-nptl_kernel-4.9/mips-linux-uclibc/usr/lib/**x86_64-linux-gnu**/pkgconfig/panelw.pc

Sieht irgendwie komisch aus...

forenuser, 2020-11-04:
nano-5.3/configure.ac -> sucht nach ncursesw5-config!
Hast Du das bei Dir gepatched und  vergessen, den Patch einzuspielen?
A-ha, invalid? Sind schon mal zwei definitive Fehler!