Freetz-NG/freetz-ng #113: Emergency-Off for Freetz-Mod?

PeterPawn, 2020-11-14:
Maybe I'm too blind ... is the emergency switch with "ds_off=y" in "kernel_args" really still working?

I've found it in a line in "/etc/boot.d/core/99-zzz-rcmod":

```
#!/bin/sh
[ "$ds_off" == "y" ] || . /etc/init.d/rc.mod 2>&1 | tee /var/log/mod.log | sed "s/^/[FREETZ] RCMOD: /g"
```

but I can't find the place, where the "ds_off" value is exported to the shell environment somewhere before.

I've thought, it has to be read first (e.g. by "ka_getVars") and afterwards be parsed, to get exported for later use.

But it looks like I'm (very) wrong - I can't even find the place where this was done in the earliest Freetz versions (tag freetz-1.0).

How the f*ck was this set ever and how is it set nowadays, where the init process isn't a single shell instance anymore and a variable "ds_off" only can been set in the shell environment (if it's not set in the script itself)?

PeterPawn, 2020-11-14:
OK, it's parsed by the kernel and put into environment of "init" process: https://www.kernel.org/doc/html/v4.14/admin-guide/kernel-parameters.html

 As long as this setting is passed along to `supervisor` (seems so according to `cat /proc/$(pidof supervisor)/environ`) and further to the started service script, it should be functioning.

closed 2020-11-14 by PeterPawn

fda77, 2020-11-15:
Ich schreib mal auf deutsch
Diese Variable hab ich noch nie benutzt. Wenn ein Image nicht startet schau ich auf die Console oder flashe einfach ein anderes. Wie man am Namen erkennt ist die noch aus der Zeit vor Freetz

PeterPawn, 2020-11-15:
Das ist halt eine einfache Methode, die Abarbeitung des gesamten Mod zu verhindern (und ja, ich weiß, daß das noch aus "ds-mod"-Zeiten stammt) und zu testen, ob Freetz oder die Box das Problem ist (statisch geänderte Dateien bleiben natürlich trotzdem ein Unterschied zur originalen Firmware) - auch für die ohne bestückte Serielle und sonst kommt man nur schlecht an die Console-Ausgaben.

Ich war auch irgendwie "vernagelt" bei den Überlegungen, wo das wohl herkommen mag - vermutlich weil ich einmal zu oft die "kernel_args" so ausgewertet habe wie hier: https://github.com/PeterPawn/modfs/blob/master/modscripts/mod_rc_tail_sh#L40 - obwohl ich das mit dem Kernel ja durchaus aus kenne/kannte - mir war halt so, daß diese KV-Sets dann ausschließlich Ziffern und Buchstaben enthalten dürfen im Namen, damit der Kernel sie an "init" weiterreicht - aber der Unterstrich ist da offenbar auch erlaubt. 

Außerdem hat mich die Tatsache irritiert, daß ein "audit=1" eben NICHT im Environment landet - aber das gilt für alle vom Kernel "konsumierten" Variablen und das "audit=" wird dann für "audit enable" verwendet, was ich eher der Variablen "audit.enable=1" zugeordnet hätte, aber offenbar soll das "audit" nicht als Module angesehen werden.

Es fiel mir erst wie Schuppen aus den Haaren, als ich das hier schon aufgemacht hatte - auch weil das Environment heutzutage ja i.d.R. erst einmal "saniert" wird, bevor es an andere Prozesse weitergereicht wird. Das "init" der BusyBox saniert da auch seinen Aufruf, aber nur seinen eigenen unter "/proc/1/cmdline" und "/proc/cmdline" enthält immer noch alles aus dem Kernel-Start, ebenso wie das weitervererbte Environment diese übrig gebliebenen KV-Sets vom Kernel-Start behält. 

Das ist aber auch eine Gefahr beim "supervisor" ... wenn man dem Service mit "EnvironmentFile=..." ein eigenes "spendiert", fehlen da die "ererbten" Werte, wenn man sie nicht explizit dort einträgt.

AVM macht das bei einigen Services auch (Freetz präpariert sich ja auch eine entsprechende Datei (env.daemon), aber erst in der "rc.mod") und das wird damit auch unterstützt - da muß man dann halt mittlerweile aufpassen, wo die Vererbung beschränkt wird durch solche "Unterbrechungen" und solche Stellen noch einmal einer Prüfung unterziehen und ggf. ändern.

fda77, 2020-11-15:
Man könnte auch ein avm image flashen, wenn man freetz kurz mal loswerdne möchet. Ich hatte nue einen Nutzen an der Variablen.
env.daemon ist relativ neu, es nur dazu da LANG auch an die daemons weiterzugeben. Wenn daemons mit komplettem env laufen,  gab es irgend wann mal Probleme mit dem webif, da wurde manches geleakt

PeterPawn, 2020-11-15:
> hatte nue einen Nutzen

Na ja ... überlege Dir das mal bei einer "non-mirrored"-Box und mit dem Gedanken im Hinterkopf, daß danach wieder mit Freetz weitergearbeitet werden soll. Das ist schon ein sehr deutlicher Unterschied, was den Aufwand angeht. Sooo sinnlos ist die Variable nicht - selbst auf "mirrored"-Devices - und auch neuere Modelle mit einem einzelnen System gibt es ja noch.

Das wäre also sehr subjektiv, wenn man das jetzt einfach auslassen würde - daher würde ich die auch wieder in die "115-..." einbauen, wenn ich Du wäre - es gibt ja keinen objektiven Grund, warum man das entfernen müßte.

fda77, 2020-11-15:
Um es anders auszudrücken: Ich glaube nicht dass es mehr als  1 Person nutzt, nämlich die die sich beschweren würde wenn es entfernt wäre

PeterPawn, 2020-11-15:
Dann mal meinerseits anders formuliert: Wenn jemand im IPPF ein Problem hat und man ihm den Vorschlag machen will, doch erst einmal zu prüfen, ob das ohne Freetz auch auftritt, ist es schon hilfreich, wenn man dem einfach nur schreiben muß, daß er doch "ds_off" benutzen könnte. 

Sagt Dir das hier was? https://freetz-ng.github.io/freetz-ng/wiki/00_FAQ/FAQ.de.html

Mir wäre es ja schon zu viel Aufwand, da erst noch im FAQ erklären zu müssen, daß das alles für 07.19+ nicht mehr gelten soll - aber das mußt Du letztlich selbst wissen.

fda77, 2020-11-15:
Es gibt da überall Anker: https://freetz-ng.github.io/freetz-ng/wiki/00_FAQ/FAQ.de.html#hilfe-die-box-ist-total-verkonfiguriert--freetz-not-aus
Ich würd das nicht erklären sondern ein recovey empfehlen

PeterPawn, 2020-11-16:
Interessant - nur funktionieren die (zumindest zum Teil) für mich gar nicht - u.a. auch der von hier: https://github.com/Freetz-NG/freetz-ng/tree/master/docs/wiki/00_FAQ in der Auflistung am Beginn, der auf diesen "ds_off"-Punkt verweisen sollte ... mein Browser landet da am Anfang der Page. Nun ist der sicherlich aber auch eher exotisch - ist halt ein Feuerfuchs (letztes Modell).

Conan179, 2020-11-16:
> Es gibt da überall Anker: https://freetz-ng.github.io/freetz-ng/wiki/00_FAQ/FAQ.de.html#hilfe-die-box-ist-total-verkonfiguriert--freetz-not-aus
Im Google Chrome geht der Link, genau an die Richtige Stelle.
> Ich würd das nicht erklären sondern ein recovey empfehlen
Im 99% der Fälle geht das, für den 1% Teil (oder eher noch weniger) haben keine möglichkeit ein Recovery zu nutzten, weil AVM keins raus gibt.

PeterPawn, 2020-11-16:
> Im Google Chrome geht der Link

Soo unklar hatte ich mich hoffentlich nicht ausgedrückt - der Link von oben auf "freetz.github.io" funktioniert auch bei mir, obwohl das mit einem "Anker" überhaupt nichts zu tun hat. 

Unter einem solchen Anker verstünde man ein <a>-Element in HTML - der Link zu dem entsprechenden <h3>-Element in der Seite funktioniert nur deshalb, weil heutzutage die meisten Browser (selbst ein Lynx unter Linux) nach der Raute als "fragment identifier" (https://tools.ietf.org/html/rfc3986#section-3.5) auch die ID eines HTML-Elements in der Seite akzeptieren, das sie dann in den Viewport scrollen, nachdem sie das Dokument (vollständig) von der Quelle geladen haben. Damit das klappt, muß die ID eindeutig sein - das ist aber KEINE Forderung im HTML-Standard.

Aber auch mit Chrome wird der Link an der von mir beschriebenen Stelle in der "docs/wiki/00_FAQ/README.md" nicht funktionieren ... da fehlt nämlich im Markdown für den Link ein (doppelter) Bindestrich zwischen "verkonfiguriert" und "freetz". Wenn sich jetzt Chrome nicht aufs Raten verlegt (was ich mal nicht hoffe, aber auch nicht gänzlich ausschließen will), findet der das Element mit der angegebenen ID auch nicht.

Und da in der Seite auch nur mit den IDs "gearbeitet" wird und eben nicht mit expliziten "Ankern" (was in Markdown auch nicht anders geht, zumindest nicht in dem bei GitHub verwendeten Umfang, afaik), gibt es eben auch kein "Inhaltsverzeichnis" in der Jekyll-Form der Dokumentation, die bei der Google-Suche zuerst gefunden wurde - daher ist es auch (ohne einen Blick in den HTML-Quelltext der Seite zu werfen, damit man die ID des H3-Tags findet - was wohl nur wenige machen würden) nicht ohne weiteres möglich, da auf einzelne Seitenelemente zu verlinken. Da fällt es dann auch kaum noch auf, daß der Link zu "voipfans.de" am Beginn des Abschnitts nicht funktioniert.

Aber ich wollte ja eigentlich über "ds_off" reden und die Dokumentation habe ich nur angeführt, weil es mir persönlich tatsächlich zu anstrengend wäre, die Beschreibungen zu ändern ... ich würde dann einfach wieder diese Option einbauen. Immer unter der Annahme, daß das bei den Änderungen für den Start unter 07.2x, wenn sie im Repo landen, nicht mehr vorhanden sein sollte - im Moment sehe ich in der "...patch_ds_off..." (oder wie die auch immer genau heißt) noch keine Änderung dahingehend, habe mir das aber auch in den letzten drei Tagen nicht im Detail angesehen, was da an Hilfsfunktionen dazukam.

Warum man zum Test, ob ein unter Freetz(-NG) entdecktes Problem auch mit der originalen Firmware auftauchen würde (zumindest für einen Anhaltspunkt dafür oder dagegen), jetzt ein Recovery-Programm verwenden sollte und damit u.U. sogar die "Besonderheiten" in der Konfiguration vernichtet, die zu diesem Problem geführt haben, verstehe ich ohnehin nicht.

Es sei denn, man will das Problem nur "weg haben" und gar nicht genau verstehen, worin es besteht und was man dagegen (außer dem Laden der Werkseinstellungen) noch unternehmen könnte. 

Das "schnellstmögliche Verstecken" eines Problems ist halt nicht meine Lösungsstrategie - vielleicht finde ich es auch genau deshalb so wichtig, daß man weiterhin eine (temporäre) Möglichkeit hat, die ganzen Modifikationen auf einen Schlag abzuschalten ... und sei es nur für eine "Zuordnung", in welchem Teil der Firmware (AVM oder Freetz(-NG)) das zu suchen wäre.

Aber es ist halt wie immer ... mir ist's eigentlich (für mich selbst) egal, weil ich selbst ja gar kein Freetz benutze. Ich fände es nur praktisch, wenn man bei Versuchen, anderen zu helfen (das soll im IPPF ja ab und an auch mal vorkommen), auf eine solche Möglichkeit verweisen könnte. 

Ich bin ja mal gespannt, wie die Reaktionen im IPPF ausfallen, wenn @fda77 / cuma / fda89 in der Zukunft bei Problemen mit Freetz, wo nicht klar ist, ob die von Freetz herrühren oder ohne "Freetz-Mod" auch auftreten würden, erst einmal die Verwendung des Recovery-Programms und eine komplette Neukonfiguration vorschlägt - aber ich meine zu wissen, für wie "hilfreich" das gehalten würde.

fda77, 2020-11-16:
Ich war mir sicher dass github in den Ankern aus "--" ein "-" macht, und hatte das so umgesetzt. Möglich ist auch dass github dieses Verhalten geändert hat. Sollte jetzt funktionieren

fda77, 2020-11-21:
Das Markdown Addon für Firefox "Markdown Viewer Webext"ersetzt "--" durch "-". Vielleicht hat es auf Github nie funktioniert

PeterPawn, 2020-11-21:
Wenn es Dir jetzt noch gelingt, aus dem oben schon angesprochenen Link auf "voipfans.de" einen auf das IPPF zu machen und dann ggf. auch noch so, daß man am Beginn des Threads landet und nicht am letzten Beitrag darin (das wäre dann: https://www.ip-phone-forum.de/threads/not-ausschalter-f%C3%BCr-ds-mod-und-debug-cfg.155770/post-1000573), dann wäre auch das (aus dem Titel nicht ersichtliche, weil erst später aufgekommene) Thema mit der Dokumentation "buried".

fda77, 2020-11-21:
Ich habs mal ohne der Threadüberschrift gemacht, dann funktioniert der Link auch bei "erledigt" Ergänzungen usw