Freetz-NG/freetz-ng #114: 7590 7.21: mc ncursesw UTF-8

forenuser, 2020-11-15:
Mit ncursesw werden bei mc nun wieder Buchstaben anstatt der Rahmen dargestellt. Wenn ich mit der Maus klicke, werden irgendwelche Zeichen in der Kommandozeile eingefügt, anstatt die geklickte Aktion auszuführen. Die .config Datei kann ich bei Bedarf hochladen.

fda77, 2020-11-15:
Vermutlich passt die Einstellung deines Terminalclients nicht

forenuser, 2020-11-16:
Ist die gleiche Einstellung von Putty. Wenn ich slang wähle und ein Image baue wird mc mit der gleichen Einstellung korrekt dargestellt.

fda77, 2020-11-16:
Wenn du mit einem Franzosen, Deutschen und Polen sprichst, meinst du jeder versteht dich, nur weil du immer den gleichen Mund verwendest?

forenuser, 2020-11-16:
Im alten Issue hatte ich geschrieben, dass MC sowohl mit slang, als auch mit ncursesw funktioniert. Dabei hatte ich die gleiche putty Einstellung verwendet. Nach der letzten Änderung funktioniert das nun nicht mehr. OK, dann muss ich halt für MC nun explizit slang verwenden.

fda77, 2020-11-16:
Das wäre dann das einfachste

feedzapper, 2020-11-16:
> Mit ncursesw werden bei mc nun wieder Buchstaben anstatt der Rahmen dargestellt. Wenn ich mit der Maus klicke, werden irgendwelche Zeichen in der Kommandozeile eingefügt, anstatt die geklickte Aktion auszuführen. Die .config Datei kann ich bei Bedarf hochladen.

putty -> Window -> Translation -> Enable VT100 line drawing even in UTF-8 Mode

forenuser, 2020-11-16:
Die Line Drawings gehen mit dieser Einstellung. Bei mir stand Terminal-Type auf "xterm". Das habe ich auch angepasst. Bekommst Du auch cryptische Zeichen in der Kommandozeile unten, wenn Du mit der linken Maus Taste ins Fenster klickst oder kannst Du MC mit der Maus bedienen?

feedzapper, 2020-11-16:
Yep, Maus geht bei mir auch nicht - es kommen irgendwelche ESC sequenzen. 
Evtl. muss man noch was beim Compilieren von dem MC Package dafür berücksichtigen ?
auch start mit "mc -x" ergibts das selbe Ergebnis ..

forenuser, 2020-11-16:
Ich habe da was gefunden. Das sollte eigentlich in unserer Version schon gelöst sein: [https://midnight-commander.org/ticket/3954](url)

fda77, 2020-11-16:
Solche Dinge könnte man hier hinterlegen:
 - im Repo: https://github.com/Freetz-NG/freetz-ng/blob/master/docs/make/mc.md
 - Ausgabe: https://freetz-ng.github.io/freetz-ng/make/mc

feedzapper, 2020-11-16:
Yep nun gehts.
In Putty kann man dann auch noch justieren - wie mit der Maus bei drag und drop verfahren werden soll.
-> Selections -> options controlling copy an paste 
wenn man mal lange weile hat ;-)

feedzapper, 2020-11-16:
nano sollte auch mit der Maus funktionieren.
Es ist aber in der nano.mk explizit deaktiviert (--disable-mouse). Dabei laufen alle Versuche zur späteren Aktivierung über die nanorc mit "set mouse" oder beim Start von nano über "nano -m" im Shell der Box ins leere ...

fda77, 2020-11-17:
Aktiviert es wenn ihr es braucht. Falls die Binary dadurch viel größer wird oder eine zusätzliche Libs benötigt wird sollte es eine eigene Option sein. 
Da das ursprüngliche Problem keines ist...

closed 2020-11-17 by fda77

forenuser, 2020-11-17:
Bei nano funktioniert bei mir Copy und Paste mit der Maus. Welche Aktionen kann man sonst noch ausführen? Ohne dem Menü am  unteren Rand kann ich nano jedoch nicht bedienen, da ich die Ctrl Tastenkombinationen nicht auswendig kann.

forenuser, 2020-11-17:
--disable-mouse scheint nur die folgende Funktion abzuschalten:

Nano has its own built-in mouse services, but they are limited. They provide only the ability to move the cursor to the point where you click, and to mark the beginning and end of a section of text. Use the Alt-M key combination to toggle between using the terminal's mouse services and nano's built-in mouse services.

Viel mehr, als den Cursor an eine bestimmte Position zu setzen, kann man scheinbar nicht machen. Für Copy und Paste muss man wieder in den Terminal Modus gehen, der auch ohne Aktivierung der Nano internen Mouse Services funktionieren sollte.