Freetz-NG/freetz-ng #115: [BUILD] build error for 7390 & 7490: ncurses

harryboo, 2020-11-16:
build failed

configs attached

[image]
[g.config.txt](https://github.com/Freetz-NG/freetz-ng/files/5548549/g.config.txt)
[q.config.txt](https://github.com/Freetz-NG/freetz-ng/files/5548550/q.config.txt)

fda77, 2020-11-16:
6 weeks old version
``´
#  freetz-ng 17273M-5672c2cee master 2020-10-06
```

closed 2020-11-16 by fda77

harryboo, 2020-11-16:
Hi,

sorry wrong config files.
Here are the files for the latest trunk.

Thanx
[7490.config.txt](https://github.com/Freetz-NG/freetz-ng/files/5549045/7490.config.txt)
[7390.config.txt](https://github.com/Freetz-NG/freetz-ng/files/5549046/7390.config.txt)

fda77, 2020-11-16:
you noticed NEWS after upate?

harryboo, 2020-11-16:
Yes, but 'make distclean' is not the solution. thats why i opened this issue.

fda77, 2020-11-16:
Solution for which error?
You "screenshot" shows only "Error 1"

fda77, 2020-11-16:
I downloaded your config and checked it against an unmodified clone.

```
--- 7490.config.txt     2020-11-16 19:55:10.000000000 +0100
+++ .config     2020-11-16 22:34:48.885149253 +0100
@@ -721,10 +721,9 @@
 # FREETZ_PACKAGE_TIFF is not set
 # FREETZ_PACKAGE_TINC is not set
 FREETZ_PACKAGE_TINYPROXY=y
-FREETZ_PACKAGE_TINYPROXY_WITH_XTINYPROXY=y
+FREETZ_PACKAGE_TINYPROXY_WITH_TRANSPARENT_PROXY=y
 FREETZ_PACKAGE_TINYPROXY_WITH_FILTER=y
 FREETZ_PACKAGE_TINYPROXY_WITH_UPSTREAM=y
-FREETZ_PACKAGE_TINYPROXY_WITH_TRANSPARENT_PROXY=y
 FREETZ_PACKAGE_TINYPROXY_WITH_REVERSE=y
 # FREETZ_PACKAGE_TINYPROXY_STATIC is not set
 # FREETZ_PACKAGE_TMUX is not set
```

It seems you there are little modifications... XTINYPROXY is not available with the NG shipped version. So the author of it should help you
I'm really pissed that here are all the time issues with unmentioned modification

harryboo, 2020-11-17:
Hi,
 i am the author of this package. It works with the new pre-version of tinyproxy.
But this is not the problem.
The problem is that ncurses 6.2 doesnt compile. 'make distclean' as stated in the news is not the solution.
I will try a new checkout in the evenning.

harryboo, 2020-11-18:
The same error after a clean checkout and NO changes.

Here the log for the package ncurses: