Freetz-NG/freetz-ng #118: [build] 7520/7530: wireguard-linux-compat build failed

wall007, 2020-11-20:
Somehow skb_put_data is already declarated when building wireguard-linux-compat. Same applies for the near identical 7530.
As a workaround i removed the skb_put_data declaration from compat.h. This build runs now since 2 weeks stable on my FritzBox 7520.

```
---> package/wireguard-linux-compat ... preparing ... mkdir -p source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729; tools/unxz -c dl/wireguard-linux-compat-1.0.20200729.tar.xz | tools/tar-gnu -x -C source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729 --transform='s|^./\+||' --strip-components=1 
set -e; shopt -s nullglob; for i in make/wireguard-linux-compat/patches/*.patch*; do case $i in *.patch|*.patch.gz|*.patch.bzip2|*.patch.bz2|*.patch.bz|*.patch.xz|*.patch.lz|*.patch.lzma|*.patch.Z) ;; *) continue ;; esac; tools/freetz_patch source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729 $i; done; 
    applying patch file make/wireguard-linux-compat/patches/010-kernel_cross_compile.patch
    patching file src/Makefile
    ----------------------------------------------------------------------
cmd() { PATH="/tmp/freetz-ng/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/arm-linux-uclibcgnueabi/bin:/tmp/freetz-ng/toolchain/build/arm_gcc-5.5.0/arm-unknown-linux-gnueabi/bin:/bin:/usr/bin:/usr/local/bin:/usr/local/sbin:/opt/mentor/questa-vip/2019.2/bin:/opt/mentor/questasim/2019.2/linux_x86_64:/opt/mentor/questasim/2019.2/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/opt/intel_fpga_lite/19.1/quartus/bin:/opt/intel_fpga_lite/19.1/quartus/sopc_builder/bin" LD_RUN_PATH="/usr/lib/freetz" FREETZ_LIBRARY_DIR="/usr/lib/freetz" make -j2  "$@"  || { printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } };       if [ -e source/.echo_item_start -a ! -e source/.echo_item_build ]; then echo -ne "\e[48;5;56mbuilding\e[49m ... "; touch source/.echo_item_build; fi; cmd -C source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729/src \
        SUBDIRS="/tmp/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729/src" \
        KERNELDIR="/tmp/freetz-ng/source/kernel/ref-ipq40xx-7530_07.20/linux-4.4" \
        ARCH="arm" \
        CROSS_COMPILE="arm-unknown-linux-gnueabi-"
building ... make[1]: Verzeichnis „/tmp/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729/src“ wird betreten
  CC [M]  /tmp/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729/src/main.o
  CC [M]  /tmp/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729/src/noise.o
In file included from <command-line>:0:0:
/tmp/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729/src/compat/compat.h:674:21: error: redefinition of 'skb_put_data'
 static inline void *skb_put_data(struct sk_buff *skb, const void *data, unsigned int len)
                     ^
In file included from include/linux/ip.h:20:0,
                 from include/linux/if_tunnel.h:4,
                 from include/net/ip_tunnels.h:4,
                 from include/net/udp_tunnel.h:4,
                 from /tmp/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729/src/compat/udp_tunnel/udp_tunnel_partial_compat.h:143,
                 from /tmp/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729/src/compat/compat.h:79,
                 from <command-line>:0:
include/linux/skbuff.h:1894:21: note: previous definition of 'skb_put_data' was here
 static inline void *skb_put_data(struct sk_buff *skb, const void *data,
                     ^
In file included from <command-line>:0:0:
/tmp/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729/src/compat/compat.h:674:21: error: redefinition of 'skb_put_data'
 static inline void *skb_put_data(struct sk_buff *skb, const void *data, unsigned int len)
                     ^
In file included from include/linux/ip.h:20:0,
                 from include/linux/if_tunnel.h:4,
                 from include/net/ip_tunnels.h:4,
                 from include/net/udp_tunnel.h:4,
                 from /tmp/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729/src/compat/udp_tunnel/udp_tunnel_partial_compat.h:143,
                 from /tmp/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729/src/compat/compat.h:79,
                 from <command-line>:0:
include/linux/skbuff.h:1894:21: note: previous definition of 'skb_put_data' was here
 static inline void *skb_put_data(struct sk_buff *skb, const void *data,
                     ^
make[4]: *** [scripts/Makefile.build:258: /tmp/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729/src/main.o] Fehler 1
make[4]: *** Es wird auf noch nicht beendete Prozesse gewartet....
make[4]: *** [scripts/Makefile.build:258: /tmp/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729/src/noise.o] Fehler 1
make[3]: *** [Makefile:1408: _module_/tmp/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729/src] Fehler 2
make[2]: *** [GNUmakefile:101: modules] Fehler 2
make[1]: *** [Makefile:26: module] Fehler 2
make[1]: Verzeichnis „/tmp/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729/src“ wird verlassen

ERROR: Build failed.
make: *** [make/wireguard-linux-compat/wireguard-linux-compat.mk:18: source/target-arm_gcc-5.5.0_uClibc-1.0.15-nptl_kernel-4.4/wireguard-linux-compat-1.0.20200729/src/wireguard.ko] Fehler 1
```
[7520.config.txt](https://github.com/Freetz-NG/freetz-ng/files/5574025/7520.config.txt)

fda77, 2020-11-20:
7520 and 7530 use the same kernel sources.
Please test the lateste revision, maybe you need to run  ```make wireguard-linux-compat-dirclean``` to apply the patch

aus-sua, 2020-11-21:
Bei einer 4040 tritt der "wireguard-linux-compat build failed" trotz der letzten Revision auch/immer noch auf.
Da hatte es in Deinem Patch:
make/wireguard-linux-compat/patches/100-avm_has-skb_put_data.patch
geholfen, die Kernelversion auf <4.5.0 zu setzen, siehst Du bitte da noch einmal "drüber",
ob das so dann auch noch für alle anderen Boxen gilt:
`+#if LINUX_VERSION_CODE < KERNEL_VERSION(4, 5, 0) && !defined(ISRHEL7)`
 ?

fda77, 2020-11-21:
Wie, immer noch? Diese 11 benutzen doch die gleiche Kernelversion: https://github.com/Freetz-NG/freetz-ng/tree/master/make/linux/patches/4.4.60
Wenn man auf 4.5 ändert geht es für die 7530/7520 nicht mehr. Für den 1200 wird de Patch so grbaucht
Vielleicht hast du einen anderen Fehler?
Es müsste sich mal jemand die 11 Kernel anschauen

aus-sua, 2020-11-21:
gerade noch einmal neu ausgecheckt, nur 4040 und expert und wireguard ausgewählt:
Doch, 
Fehler ist bei der 4040 bei dem Übersetzen des wireguard-linux-compat-1.0.20201112 
doch (noch/wieder) vorhanden:
`set -e; shopt -s nullglob; for i in make/wireguard-linux-compat/patches/*.patch*; do case $i in *.patch|*.patch.gz|*.patch.bzip2|*.patch.bz2|*.patch.bz|*.patch.xz|*.patch.lz|*.patch.lzma|*.patch.Z) ;; *) continue ;; esac; tools/freetz_patch source/target-arm_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-4.4/wireguard-linux-compat-1.0.20201112 $i; done; 
    applying patch file make/wireguard-linux-compat/patches/010-kernel_cross_compile.patch
    patching file src/Makefile
    ----------------------------------------------------------------------
    applying patch file make/wireguard-linux-compat/patches/100-avm_has-skb_put_data.patch
    patching file src/compat/compat.h
    ----------------------------------------------------------------------
...
...
/home/freetz/4040g2-07.14-wg-test/source/target-arm_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-4.4/wireguard-linux-compat-1.0.20201112/src/peerlookup.o
/home/freetz/4040g2-07.14-wg-test/source/target-arm_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-4.4/wireguard-linux-compat-1.0.20201112/src/socket.c: In function 'wg_socket_send_buffer_to_peer':
/home/freetz/4040g2-07.14-wg-test/source/target-arm_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-4.4/wireguard-linux-compat-1.0.20201112/src/socket.c:198:2: error: implicit declaration of function 'skb_put_data' [-Werror=implicit-function-declaration]
  skb_put_data(skb, buffer, len);
  ^
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:258: /home/freetz/4040g2-07.14-wg-test/source/target-arm_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-4.4/wireguard-linux-compat-1.0.20201112/src/socket.o] Fehler 1
make[4]: *** Es wird auf noch nicht beendete Prozesse gewartet....
make[3]: *** [Makefile:1406:
...`

> Es müsste sich mal jemand die 11 Kernel anschauen

Was und wo genau?

fda77, 2020-11-21:
Ich beginne und beende Codeblöcke immer mit 3 ``` ` ``` in einer einzigen Zeile, dann bleiben die Umbrüche.
Hab mir die Kernel schon angeschaut, nur die 3 4.4er für FOS 7.2 brauchen den Patch, die älteren nicht.

aus-sua, 2020-11-21:
ok, danke
:-)

fda77, 2020-11-21:
Müsste jetzt passen...
Evtl ist ein "make wireguard-linux-compat-dirclean" nötig.
@aus-sua & @wall007 How fast is wireguard on your devices?

closed 2020-11-21 by fda77

aus-sua, 2020-11-22:
Danke für die Anpassungen.
Ich habe mangels Zeit noch keine lauffähige Konfiguration, die VPN-Einbindung
muß aber innerhalb der nächsten drei Wochen laufen...
Dann kann ich auch etwas zu den Datenraten sagen.
z.Z. behelfe ich mich dort mit SSH-Verbindungen  ;-)

wall007, 2020-11-22:
Thanks for the fast fix!

I have connected two networks over 100/40 internet lines. And use a x86 Server as wireguard remote for my 7520. With iperf3 i messearued a throughput of **~34Mbits/s**. When I connect my PC directly to my Server I get the same result. It seems that my Internet is the bottleneck. Therefore I tested bidirectional and got ~44MBits/s as combined throughput.

aus-sua, 2020-11-22:
Thanks for the feedback.
These are the data rates I expect :-)

fda77, 2020-11-22:
@aus-sua: Your device could have different throughput!
 Im using a Raspberry which can route more than 100mbit: https://github.com/Freetz-NG/freetz-ng/pull/101#issuecomment-716029888